robotology/icub-main #253: xml module descriptor for old modules

tanismar, 2015-11-27:
Hi!
Most "old" modules do not have xml descriptors, and although most of the documentation can be found on the .h files, in order to use them properly on yarpbuilder the proper xml file is needed. We should make them for those old files, which were written before the convention was decided.

gsaponaro, 2015-11-27:
Related question: where can I find an up-to-date example template of these yarpbuilder xml descriptors?

pattacini, 2015-11-27:
@gsaponaro did you check out our tutorial page at http://wiki.icub.org/iCub/main/dox/html/xml_module_documentation.html?

gsaponaro, 2015-11-27:
Thank you, I remembered there was one page about it but was not sure which one. Between that page and http://www.yarp.it/yarpmanager.html I have plenty of examples.

pattacini, 2015-11-27:
`yarpmanager` does not cover (yet) the `xml` modules you're seeking for, hence stick to the tutorial.

closed 2017-03-28 by pattacini