robotology/icub-main #255: Check how xml parameters are assigned by motionControl to the robot joints, taking in account the axisMap

randaz81, 2015-11-27:
See #229.

traversaro, 2017-09-01:
cc @nunoguedelha

nunoguedelha, 2017-09-02:
Ciao @randaz81, @barbalberto , is there any update on this issue? This topic has been brought to our attention (estimation guys in DIC lab) as we require accurate information on the joint/motor coupling when identifying motor and joint friction parameters: https://github.com/robotology/robots-configuration/issues/39.

Our main concern is w.r.t. the couping matrix **matrixM2J**, and how it is mapped to the list of axis returned by  `yarp::dev::IAxisInfo::getAxisName(int j, yarp::os::ConstString & name)`.

### Example on the torso
`getAxisName()` returns the list of axes: torso_yaw, torso_roll, torso_pitch
Which matches the `AxisName` order in:
https://github.com/robotology/robots-configuration/blob/0ff6cee7bd4dee7811811e4d2eee29b16407234b/iCubGenova04/hardware/mechanicals/torso-eb5-j0_2-mec.xml#L8-L9
But the matrix lines follow the order R-P-Y which is the one in the `_AxisName` variable. This variable gets the axis names from the config files, remapped by `AxisMap`.
I was expecting to find the same remapped order in the axes names returned by `getAxisName()`, is this normal?

barbalberto, 2017-10-03:
My two cents.
From the YARP point of view, the matrices are not mapped in any way. This is because the matrix is a _**single**_ data entry (big as much as it is required, but one piece of information). 

A matrix is referred to the whole controller (e.g. EMS), I don't know how it is filled. My guess is that the matrix is used by the firmware, so it is in the order preferred by the firmware, probably in the motor order.

closed 2021-03-27 by pattacini