robotology/icub-main #257: Cannot open /icubSim/cam/left/fovea and /icubSim/cam/right/fovea

wainting, 2015-12-06:
Hi to all,

when i run iCub_SIM during the configuration stage I see that for the initialization of camera it found the Sim_camera.ini here:
||| finding file [cameras]
||| checking [/home/daniele/Sim_camera.ini] (pwd)
||| checking [/usr/local/share/iCub/contexts/simConfig/Sim_camera.ini] (context)
||| found /usr/local/share/iCub/contexts/simConfig/Sim_camera.ini
yarp: Port /icubSim/cam/left active at tcp://192.168.0.19:10019
yarp: Port /icubSim/cam/right active at tcp://192.168.0.19:10020
yarp: Port /icubSim/cam active at tcp://192.168.0.19:10021

but when i open the Sim_camera.ini in  /usr/local/share/iCub/contexts/simConfig the content is : 

name_left /cam/left
name_right /cam/right
name_wide /cam
name_leftFov /cam/left/fovea
name_rightFov /cam/right/fovea
name_leftLog /cam/left/logpolar
name_rightLog /cam/right/logpolar

Why my simulator don't open the port for fovea???

Here all ports opened after starting iCub_SIM (i run yarp name list)

registration name /icubSim/cam ip 192.168.0.19 port 10021 type tcp
registration name /icubSim/cam/left ip 192.168.0.19 port 10019 type tcp
registration name /icubSim/cam/right ip 192.168.0.19 port 10020 type tcp
registration name /icubSim/face/eyelids ip 192.168.0.19 port 10049 type tcp
registration name /icubSim/head/command:i ip 192.168.0.19 port 10031 type tcp
registration name /icubSim/head/rpc:i ip 192.168.0.19 port 10030 type tcp
registration name /icubSim/head/state:o ip 192.168.0.19 port 10032 type tcp
registration name /icubSim/head/stateExt:o ip 192.168.0.19 port 10033 type tcp
registration name /icubSim/inertial ip 192.168.0.19 port 10007 type tcp
registration name /icubSim/joint_vsens/left_arm:i ip 192.168.0.19 port 10012 type tcp
registration name /icubSim/joint_vsens/left_leg:i ip 192.168.0.19 port 10008 type tcp
registration name /icubSim/joint_vsens/right_arm:i ip 192.168.0.19 port 10011 type tcp
registration name /icubSim/joint_vsens/right_leg:i ip 192.168.0.19 port 10009 type tcp
registration name /icubSim/joint_vsens/torso:i ip 192.168.0.19 port 10010 type tcp
registration name /icubSim/left_arm/command:i ip 192.168.0.19 port 10023 type tcp
registration name /icubSim/left_arm/rpc:i ip 192.168.0.19 port 10022 type tcp
registration name /icubSim/left_arm/state:o ip 192.168.0.19 port 10024 type tcp
registration name /icubSim/left_arm/stateExt:o ip 192.168.0.19 port 10025 type tcp
registration name /icubSim/left_leg/command:i ip 192.168.0.19 port 10035 type tcp
registration name /icubSim/left_leg/rpc:i ip 192.168.0.19 port 10034 type tcp
registration name /icubSim/left_leg/state:o ip 192.168.0.19 port 10036 type tcp
registration name /icubSim/left_leg/stateExt:o ip 192.168.0.19 port 10037 type tcp
registration name /icubSim/right_arm/command:i ip 192.168.0.19 port 10027 type tcp
registration name /icubSim/right_arm/rpc:i ip 192.168.0.19 port 10026 type tcp
registration name /icubSim/right_arm/state:o ip 192.168.0.19 port 10028 type tcp
registration name /icubSim/right_arm/stateExt:o ip 192.168.0.19 port 10029 type tcp
registration name /icubSim/right_leg/command:i ip 192.168.0.19 port 10039 type tcp
registration name /icubSim/right_leg/rpc:i ip 192.168.0.19 port 10038 type tcp
registration name /icubSim/right_leg/state:o ip 192.168.0.19 port 10040 type tcp
registration name /icubSim/right_leg/stateExt:o ip 192.168.0.19 port 10041 type tcp
registration name /icubSim/skin/left_arm_comp ip 192.168.0.19 port 10014 type tcp
registration name /icubSim/skin/left_forearm_comp ip 192.168.0.19 port 10016 type tcp
registration name /icubSim/skin/left_hand_comp ip 192.168.0.19 port 10003 type tcp
registration name /icubSim/skin/left_hand_comp/rpc:i ip 192.168.0.19 port 10004 type tcp
registration name /icubSim/skin/right_arm_comp ip 192.168.0.19 port 10015 type tcp
registration name /icubSim/skin/right_forearm_comp ip 192.168.0.19 port 10017 type tcp
registration name /icubSim/skin/right_hand_comp ip 192.168.0.19 port 10005 type tcp
registration name /icubSim/skin/right_hand_comp/rpc:i ip 192.168.0.19 port 10006 type tcp
registration name /icubSim/skin/torso_comp ip 192.168.0.19 port 10018 type tcp
registration name /icubSim/skinManager/skin_events:o ip 192.168.0.19 port 10013 type tcp
registration name /icubSim/texture/face ip 192.168.0.19 port 10046 type tcp
registration name /icubSim/texture/screen ip 192.168.0.19 port 10047 type tcp
registration name /icubSim/texture/sky ip 192.168.0.19 port 10048 type tcp
registration name /icubSim/torso/command:i ip 192.168.0.19 port 10043 type tcp
registration name /icubSim/torso/rpc:i ip 192.168.0.19 port 10042 type tcp
registration name /icubSim/torso/state:o ip 192.168.0.19 port 10044 type tcp
registration name /icubSim/torso/stateExt:o ip 192.168.0.19 port 10045 type tcp
registration name /icubSim/world ip 192.168.0.19 port 10002 type tcp
registration name /root ip 192.168.0.19 port 10000 type tcp
registration name fallback ip 224.2.1.1 port 10000 type mcast
*** end of message

Best

Daniele

pattacini, 2015-12-18:
Did you compile the [**`logpolar`**](https://github.com/robotology/logpolar) component? Without it, the simulator won't open up any `logpolar` related ports.

wainting, 2015-12-19:
Ok. I ve installed ICUBcontrib and logpolar from the directory you linked. I run make in my icub-main setting logpolar_DIR and logpolar_INCLUDE_DIR, but when i run iCub_SIM 
||| finding file [cameras]
||| checking [/home/daniele/robsoft/icub-main/build/Sim_camera.ini] (pwd)
||| checking [/usr/local/share/iCub/contexts/simConfig/Sim_camera.ini] (context)
||| found /usr/local/share/iCub/contexts/simConfig/Sim_camera.ini
yarp: Port /icubSim/cam/left active at tcp://192.168.1.128:10019
yarp: Port /icubSim/cam/right active at tcp://192.168.1.128:10020
yarp: Port /icubSim/cam active at tcp://192.168.1.128:10021

there is no fovea ports..

vtikha, 2015-12-19:
@wainting once you successfully compiled the [**`logpolar`**](https://github.com/robotology/logpolar), did you run cmake again for the icub-main before building it again? With ``logpolar_DIR`` and ``logpolar_INCLUDE_DIR`` set up correctly, cmake-ing should detect them and define the correct definitions thus allowing the creation of the ``fovea`` and ``logpolar`` ports.

wainting, 2015-12-22:
Hi vtikha,
yes i run cmake on my icub-main and i can see

`-- iCub Simulator: logpolar library detected and used`

at the end of output.
I make an empty file named Sim_camera.ini in my pwd when i run iCub_SIM, I would expected that the ports cam/right and cam/left will not configure but i get this:

```
||| checking [/home/daniele/Sim_camera.ini] (pwd)
||| found /home/daniele/Sim_camera.ini
yarp: Port /icubSim/cam/left active at tcp://192.168.1.128:10019
yarp: Port /icubSim/cam/right active at tcp://192.168.1.128:10020
yarp: Port /icubSim/cam active at tcp://192.168.1.128:10021
```
If I write in my Sim_camera.ini this code:
```
name_left /cam/left
name_right /cam/left
name_wide /cam
name_leftFov /cam/left/fovea
name_rightFov /cam/right/fovea
name_leftLog /cam/left/logpolar
name_rightLog /cam/right/logpolar
```
The outuput of iCub_SIM is 
```
||| finding file [cameras]
||| checking [/home/daniele/Sim_camera.ini] (pwd)
||| found /home/daniele/Sim_camera.ini
yarp: Port /icubSim/cam/left active at tcp://192.168.1.128:10019
yarp: Port /icubSim/cam/right active at tcp://192.168.1.128:10020
yarp: Port /icubSim/cam active at tcp://192.168.1.128:10021
[DEBUG]Creating interface for body part left_arm
||| finding file [left_arm]

```
Why it is bypassing the other ports?

closed 2016-05-10 by pattacini