robotology/icub-main #262: icub_SIM it does not work anymore

misaki43, 2015-12-22:
Hi, 
I have followed the following instruction "http://wiki.icub.org/wiki/Linux:Installation_from_binaries" and I have done the "check your installation". It was working.
After that, I have installed a lot of software (as codyco_superbuild) and now, the icub_SIM doesn't work anymore: When I follow the "check your installation">> "icub_Sim", i have a segmentation fault. In the DEBUG I can see it doesn't find "simulator.ini". I have sought it in all folders ("grep" research), but I didn't found it.
Then, i tried to install it again (sudo apt-get install icub) but I have already the newest version. I don't know what to do know. Could you help me please?

traversaro, 2015-12-24:
Hi @misaki43 , 
  I am afraid there could be some kind of conflict between the binary installation of the iCub software and the `codyco-superbuild`. To help us debug the issue, could you paste the output that you get in the terminal for the following commands: 
~~~
which iCub_SIM
~~~
~~~
ldd `which iCub_SIM`
~~~
~~~
echo $YARP_DATA_DIRS
~~~
Thanks and happy holidays!

misaki43, 2015-12-28:
Hi @traversaro, sorry I was in holidays. Indeed it is most likely a conflict problem because I have following many tutorials to install softwares, in a bad order. I may have install the same software many times (because with "superbuild" installation, I don't know exactly what I have already install: I should have read what I was installing sorry).

which iCub_SIM
>>/usr/bin/iCub_SIM

ldd `which iCub_SIM`
>>	linux-vdso.so.1 =>  (0x00007ffeaf3d9000)
	libYARP_OS.so.1 => /usr/local/lib/libYARP_OS.so.1 (0x00007f2772afd000)
	libYARP_sig.so.1 => /usr/local/lib/libYARP_sig.so.1 (0x00007f27728c4000)
	libYARP_math.so.1 => /usr/local/lib/libYARP_math.so.1 (0x00007f27726aa000)
	libYARP_dev.so.1 => /usr/local/lib/libYARP_dev.so.1 (0x00007f277230a000)
	libYARP_init.so.1 => /usr/local/lib/libYARP_init.so.1 (0x00007f2772107000)
	libSDL-1.2.so.0 => /usr/lib/x86_64-linux-gnu/libSDL-1.2.so.0 (0x00007f2771e71000)
	libode.so.1 => /usr/lib/libode.so.1 (0x00007f2771b7d000)
	libGLU.so.1 => /usr/lib/x86_64-linux-gnu/libGLU.so.1 (0x00007f277190f000)
	libGL.so.1 => /usr/lib/nvidia-340/libGL.so.1 (0x00007f27715c3000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f27712bf000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f2770fb9000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f2770da3000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f27709de000)
	libACE-6.0.3.so => /usr/lib/libACE-6.0.3.so (0x00007f277066b000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f2770463000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f2770245000)
	libreadline.so.6 => /lib/x86_64-linux-gnu/libreadline.so.6 (0x00007f276ffff000)
	libgsl.so.0 => /usr/lib/libgsl.so.0 (0x00007f276fb9f000)
	libgslcblas.so.0 => /usr/lib/libgslcblas.so.0 (0x00007f276f953000)
	libasound.so.2 => /usr/lib/x86_64-linux-gnu/libasound.so.2 (0x00007f276f663000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f276f45f000)
	libpulse-simple.so.0 => /usr/lib/x86_64-linux-gnu/libpulse-simple.so.0 (0x00007f276f25b000)
	libpulse.so.0 => /usr/lib/x86_64-linux-gnu/libpulse.so.0 (0x00007f276f012000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f276ecdd000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f276eacb000)
	libcaca.so.0 => /usr/lib/x86_64-linux-gnu/libcaca.so.0 (0x00007f276e7fe000)
	libnvidia-tls.so.340.96 => /usr/lib/nvidia-340/tls/libnvidia-tls.so.340.96 (0x00007f276e5fb000)
	libnvidia-glcore.so.340.96 => /usr/lib/nvidia-340/libnvidia-glcore.so.340.96 (0x00007f276b9e7000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f2772ef7000)
	libtinfo.so.5 => /lib/x86_64-linux-gnu/libtinfo.so.5 (0x00007f276b7be000)
	libpulsecommon-4.0.so => /usr/lib/x86_64-linux-gnu/pulseaudio/libpulsecommon-4.0.so (0x00007f276b557000)
	libjson-c.so.2 => /lib/x86_64-linux-gnu/libjson-c.so.2 (0x00007f276b34c000)
	libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007f276b107000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f276aee8000)
	libslang.so.2 => /lib/x86_64-linux-gnu/libslang.so.2 (0x00007f276ab58000)
	libncursesw.so.5 => /lib/x86_64-linux-gnu/libncursesw.so.5 (0x00007f276a924000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f276a70b000)
	libwrap.so.0 => /lib/x86_64-linux-gnu/libwrap.so.0 (0x00007f276a501000)
	libsndfile.so.1 => /usr/lib/x86_64-linux-gnu/libsndfile.so.1 (0x00007f276a299000)
	libasyncns.so.0 => /usr/lib/x86_64-linux-gnu/libasyncns.so.0 (0x00007f276a093000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f2769e8f000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f2769c89000)
	libnsl.so.1 => /lib/x86_64-linux-gnu/libnsl.so.1 (0x00007f2769a6f000)
	libFLAC.so.8 => /usr/lib/x86_64-linux-gnu/libFLAC.so.8 (0x00007f276983e000)
	libvorbisenc.so.2 => /usr/lib/x86_64-linux-gnu/libvorbisenc.so.2 (0x00007f276936f000)
	libvorbis.so.0 => /usr/lib/x86_64-linux-gnu/libvorbis.so.0 (0x00007f2769142000)
	libogg.so.0 => /usr/lib/x86_64-linux-gnu/libogg.so.0 (0x00007f2768f39000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007f2768d1e000)

echo $YARP_DATA_DIRS
>> :/home/[myName]/codyco-superbuild/build/install/share/codyco:/home/[myName]/codyco-superbuild/build/install/share/codyco

Thanks for the help and happy holidays too !

traversaro, 2015-12-29:
It seems like this behavior is caused by this bug: https://github.com/robotology/yarp/issues/336 . 
Can you try to comment out the `YARP_DATA_DIRS` line from your .bashrc and see if `iCub_SIM` is working?

traversaro, 2015-12-29:
Furthermore it seems that you are using an iCub_SIM installed from binaries packages (because executable from binary packages is installed in `/usr/bin`) that is linking yarp installed from source (because the default installation directory for library installed from source is `/usr/local/lib`).

misaki43, 2015-12-29:
Hi,
It doesn't work better with the bashrc modification.
Concerning "binaries" and "sources" what does it change? I'm a beginner, I only know that binaries are pre-compiled and sources require to be compiled... But know I have install it, what is the problem? (And what is the better installation?)

traversaro, 2015-12-29:
The main difference between the binary version of YARP (that you install following the instructions at http://wiki.icub.org/wiki/Linux:Installation_from_binaries) and the source installation (that you install following the instructions at http://wiki.icub.org/wiki/Linux:Installation_from_binaries) is that the binaries are made out of a **released** version of `yarp` or `icub-main`, while by installing from source you got the latest version, the one in the `master` branch of the repo. 

This means  that if the [`Application Binary Interface`](https://en.wikipedia.org/wiki/Application_binary_interface) (usually called ABI) of YARP has changed between the released version (the one against the iCUB binary has been built) and the latest version, the ICUB binary will not work anymore if is linking the latest version of YARP insted of the binary of YARP of the right version. 

By the way, after the change in the `.bashrc` have you opened a new terminal to make sure that the `YARP_DATA_DIRS` enviromental variable is not defined?

misaki43, 2015-12-29:
Yes, I have verified the variable wasn't defined. 
After that, I saw that the icub-main has disappeared, so I want to see why I can't install it again. Then,  I read the part "follow the section Install from sources -- Compiling YARP (you an skip installation of packages)." Because of that, I tried to install it again, but this time from sources.
I think I have made things worse, sorry I thought I knew what I was doing...
Now, when I try to use yarp alone it doesn't work anymore... (when I launch yarpmanager, it is written: >>did not find ymanager.ini

To resume, i think it is the mess on my computer.
Should I try to uninstall all things and reinstall all things? Or did you think it is cleanable?

traversaro, 2015-12-29:
I think it should be cleanable. : ] 

As a first step, I added installation for uninstalling the binary packages at http://wiki.icub.org/wiki/Linux:Installation_from_binaries#Uninstall . Can you try to follow these instruction and check that `which iCub_SIM` does not return a binary in the `/usr/bin` directory? Thanks!

misaki43, 2015-12-30:
Yes is back in this directory but it doesn't work better: 
>>iCub_SIM 
[DEBUG]
ODE configuration: ODE ODE_EXT_trimesh ODE_EXT_opcode ODE_OPC_new_collider ODE_double_precision

>>[DEBUG]SimConfig::configure
||| clearing context
||| adding context [simConfig]
||| configuring
||| no policy found
||| default config file specified as simulator.ini
||| checking [/home/user/simulator.ini] (pwd)
||| checking [/home/user/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/user/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [robots/default] (robot YARP_DATA_DIRS)
||| checking [/home/user/codyco-superbuild/install/share/codyco/robots/default] (robot YARP_DATA_DIRS)
||| checking [config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/user/codyco-superbuild/install/share/codyco/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/user/.config/yarp/contexts/simConfig] (context YARP_CONFIG_HOME)
||| checking [/home/user/.local/share/yarp/contexts/simConfig] (context YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/contexts/simConfig] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/contexts/simConfig] (context YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/contexts/simConfig] (context YARP_CONFIG_DIRS)
||| checking [contexts/simConfig] (context YARP_DATA_DIRS)
||| checking [/home/user/codyco-superbuild/install/share/codyco/contexts/simConfig] (context YARP_DATA_DIRS)
||| checking [/home/user/.config/yarp/simulator.ini] (YARP_CONFIG_HOME)
||| checking [/home/user/.local/share/yarp/simulator.ini] (YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/simulator.ini] (YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/simulator.ini] (YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/simulator.ini] (YARP_CONFIG_DIRS)
||| checking [simulator.ini] (YARP_DATA_DIRS)
||| checking [/home/user/codyco-superbuild/install/share/codyco/simulator.ini] (YARP_DATA_DIRS)
||| did not find simulator.ini
yarp: cannot read from simulator.ini
[INFO]default module name:  /icubSim 
[INFO]default verbosity level:  0 
||| finding file []
||| checking [/home/user/] (pwd)
||| found /home/user/
Segmentation fault (core dumped)

Anyway, I'm afraid because I have installed other things (for example codyco-superbuild) and I don't know how works dependencies, so the other things might doesn't work anymore, does it ?

traversaro, 2015-12-30:
Even by cleaning the yarp/icub binaries installation, you could still have problems in running `iCub_SIM` if you defined the YARP_DATA_DIRS variable (see https://github.com/robotology/yarp/issues/336#issuecomment-167793620). 

As a temporary workaround, when you define the `YARP_DATA_DIRS` variable you can explicitly add the yarp and icub-main configuration directories. Assuming that you install yarp and icub in the default installation directories, the line in your `.bashrc` file should be something like:
~~~
YARP_DATA_DIRS=/usr/local/share/yarp:/usr/local/share/iCub:/home/user/codyco-superbuild/install/share/codyco/
~~~

misaki43, 2015-12-30:
But i don't understand: i don't find the icub-main directory, where is it supposed to be?
The "iCub" directory is not in "/usr/local/share" but i find it here: "/usr/share". I change it in the bashrc and now the segmentation fault is not located in the same area:
>>iCub_SIM 
[DEBUG]
ODE configuration: ODE ODE_EXT_trimesh ODE_EXT_opcode ODE_OPC_new_collider ODE_double_precision

>>[DEBUG]SimConfig::configure
||| clearing context
||| adding context [simConfig]
||| configuring
||| no policy found
||| default config file specified as simulator.ini
||| checking [/home/user/simulator.ini] (pwd)
||| checking [/home/user/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/user/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/iCub/robots/default] (robot YARP_DATA_DIRS)
||| checking [/home/user/codyco-superbuild/install/share/codyco/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/user/codyco-superbuild/install/share/codyco/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/user/.config/yarp/contexts/simConfig] (context YARP_CONFIG_HOME)
||| checking [/home/user/.local/share/yarp/contexts/simConfig] (context YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/contexts/simConfig] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/contexts/simConfig] (context YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/contexts/simConfig] (context YARP_CONFIG_DIRS)
||| checking [/usr/local/share/yarp/contexts/simConfig] (context YARP_DATA_DIRS)
||| checking [/usr/share/iCub/contexts/simConfig] (context YARP_DATA_DIRS)
||| found /usr/share/iCub/contexts/simConfig
||| checking [/home/user/codyco-superbuild/install/share/codyco/contexts/simConfig] (context YARP_DATA_DIRS)
||| checking [/usr/share/iCub/contexts/simConfig/simulator.ini] (context)
||| found /usr/share/iCub/contexts/simConfig/simulator.ini
[INFO]default module name:  /icubSim 
[INFO]default verbosity level:  0 
||| finding file [ode_params.ini]
||| checking [/home/user/ode_params.ini] (pwd)
||| checking [/usr/share/iCub/contexts/simConfig/ode_params.ini] (context)
||| found /usr/share/iCub/contexts/simConfig/ode_params.ini
Segmentation fault (core dumped

traversaro, 2015-12-30:
Mhh, if you removed the icub binary packages you should not have anything related to icub neither in `/usr/bin` nor in `/usr/share`. 
Can you check again that `which iCub_SIM` **does not** return a binary in the `/usr/bin` directory? Otherwise that means that the `icub` deb packages has not correctly uninstalled.

traversaro, 2015-12-30:
Sorry if I did not see the question before, but what you mean by 
> NOTE: Sorry I forgot to say: why the icub-main is not created when i follow the installation there: http://wiki.icub.org/wiki/Linux:Installation_from_sources?
?

misaki43, 2015-12-30:
Sorry I have misunderstood: I thought you wanted me to reinstall iCub after uninstalling. (so it appear again in the /usr/bin).
Now I have uninstalled it again, it is no longr in the "/usr/bin" but because of that I have no iCub_SIM at all. 
So, what I am supposed to do now?
Concerning icub-main, before all these problems, I had a icub-main folder that have disappeared. I thought this folder was installed with "sudo apt-get install icub" but even when I launch it, it is not install.

traversaro, 2015-12-30:
I think that the most tested option is to install `icub-main` following the instruction here: http://wiki.icub.org/wiki/Linux:Installation_from_sources . Another possible option is to let the superbuild download and install `icub-main`: in this way you should not have any problem releated to a confict between the superbuild and icub-main .

misaki43, 2015-12-30:
It is what I have done before, but when I do it, the "icub-main" folder is not where it is supposed to be (and again the iCub_SIM is in /usr/bin etc).

traversaro, 2015-12-30:
That is strange. Can you paste the content of the `CMakeCache.txt` file in the `icub-main/build` directory. 
Concerning the `icub-main` directory, it is created when you get the source code of `icub-main` from git. 
I have added a section to better explain that part: http://wiki.icub.org/wiki/Linux:Installation_from_sources#Download_iCub_source_code .

misaki43, 2015-12-30:
Ok it is working now! The issue is finished but if you have time, I have some questions:

1. Sorry, I haven't understood I have to do " git clone https://github.com/robotology/icub-main.git"... It seems natural for you, can you explain me why? Am I supposed to call github.com/robotology/*.git all time when I read a unknown folder? 

2. I don't like to have a lot of folder in "/home/user" so I modify the path of the programm ("/home/user/Software/src", where I have put other programs). It is working but I am afraid with paths that are autocratically used by other programms. Is that going to be a problem?  

3. Another question is: what is " Setup ICUBcontrib (optional) "? Should I install it?

4. Finally, it seems that "YARP" is uninstall, so... Why it works? It seems to be currently install a lot of time:
- /home/user/Software/src (where I put all softwares)
- /home/user/gazebo-yarp-plugins/
- /home/user/codyco-superbuild/
Can you explain me? 

Anyway, thanks a lot!

traversaro, 2015-12-30:
1. That operation is used to "clone" a remote git repository to your computer. GitHub has some really nice documentation regarding the basic concepts behind git, check it: https://help.github.com/articles/cloning-a-repository/ . 

2. As long as all the enviromental variable that you define are correctly updated, everything should work fine. A trick exploit by some people is to define the directory where they download all the data as an enviromental variable itself, and define all the other environmental variables depending on this one. In this one you can change the "prefix" variable once and change all other enviromental variables automatically. 

3. Check the `icub-contrib-common` repository README https://github.com/robotology/icub-contrib-common#icub-contrib-common . In a nutshell it useful to have a single place where install configuration files for different software. 

4. What you mean by "YARP" is uninstall? The binary version?

EliseiFrederic, 2016-01-19:
@traversaro : in the uninstall add-on that you wrote in :
http://wiki.icub.org/wiki/Linux:Installation_from_binaries#Uninstall
I guess you left an extra 'icub-common' in the first 'apt-get remove' (copy paste error...)

traversaro, 2016-01-19:
@EliseiFrederic thanks for noticing, fixing.

closed 2016-01-19 by traversaro