robotology/icub-main #264: Implement "get reference" functions in Ipositioncontrol2, IvelocityControl2, IPositionDirect

valegagge, 2015-12-23:
Last setpoint value sent to board shouldn't save in motion control, but shoul asked to fw.

In order to implement this behaviour we need to modify ethernet protocol.

randaz81, 2015-12-27:
Mumble mumble...

randaz81, 2015-12-27:
Well, thinking carefully about this, we can take advantage about this feature request to introduce for the first time a radical change in our policy. Let's implement these functionalities in motionControl. This could be the first step to move away from the firmware other routines (so that the sytem is consistent and we do not have part of the logic in the firmware and part in the motion control). In principle we could handle also some low-frequency error cheks (i2t, temperature) in motioncontrol, and even the decoupling (with some caveats), greatly semplifying the fw. Let's discuss about this in the next January meeting.

randaz81, 2015-12-27:
@ale-git I'm literally enthusiast about of the idea of having joint set concepts such as coupled calibration commands, coupled control mode switching implemented at the motioncontrol level!

valegagge, 2016-01-08:
When I have spoken with @randaz81 he told me he prefered get these values from Fw, so I modified ethernet protocol and implemented these features. I saw his post late :(

Certainly, if we save values in embObjMotionControl, fw can be more simple, but we lost the possibility to check if fw receives correct values. I think "get reference" functions will be used more for debug porpose than in user applications.

valegagge, 2016-01-08:
Implementing these functions in CanBusMotionControl remains to be done. I assegn this issue a @randaz81 .

randaz81, 2017-05-22:
fixed (partially) by:
https://github.com/robotology/yarp/pull/1220

pattacini, 2017-06-06:
Merged.

closed 2017-06-06 by pattacini