robotology/icub-main #268: cfw2can not compiling on pc104

rlober, 2016-01-06:
I am compiling `yarp` and `icub-main` fresh for pc104 (`iCubParis02`). `yarp` compiles fine, but I ran into the following compile error on `icub-main`:

```
[ 27%] Building CXX object src/libraries/icubmod/cfw2Can/CMakeFiles/cfw2can.dir/Cfw2Can.cpp.o
/usr/local/src/robot/source/icub-main/src/libraries/icubmod/cfw2Can/Cfw2Can.cpp: In member function ‘virtual bool yarp::dev::Cfw2Can::open(yarp::os::Searchable&)’:
/usr/local/src/robot/source/icub-main/src/libraries/icubmod/cfw2Can/Cfw2Can.cpp:114:66: error: invalid use of incomplete type ‘class yarp::os::Value’
In file included from /usr/local/src/robot/source/yarp/src/libYARP_OS/include/yarp/os/Contact.h:14:0,
                 from /usr/local/src/robot/source/yarp/src/libYARP_OS/include/yarp/os/ConnectionReader.h:14,
                 from /usr/local/src/robot/source/yarp/src/libYARP_dev/include/yarp/dev/DeviceDriver.h:13,
                 from /usr/local/src/robot/source/yarp/src/libYARP_dev/include/yarp/dev/CanBusInterface.h:13,
                 from /usr/local/src/robot/source/icub-main/src/libraries/icubmod/cfw2Can/Cfw2Can.cpp:14:
/usr/local/src/robot/source/yarp/src/libYARP_OS/include/yarp/os/Searchable.h:18:15: error: forward declaration of ‘class yarp::os::Value’
/usr/local/src/robot/source/icub-main/src/libraries/icubmod/cfw2Can/Cfw2Can.cpp:115:66: error: invalid use of incomplete type ‘class yarp::os::Value’
In file included from /usr/local/src/robot/source/yarp/src/libYARP_OS/include/yarp/os/Contact.h:14:0,
                 from /usr/local/src/robot/source/yarp/src/libYARP_OS/include/yarp/os/ConnectionReader.h:14,
                 from /usr/local/src/robot/source/yarp/src/libYARP_dev/include/yarp/dev/DeviceDriver.h:13,
                 from /usr/local/src/robot/source/yarp/src/libYARP_dev/include/yarp/dev/CanBusInterface.h:13,
                 from /usr/local/src/robot/source/icub-main/src/libraries/icubmod/cfw2Can/Cfw2Can.cpp:14:
/usr/local/src/robot/source/yarp/src/libYARP_OS/include/yarp/os/Searchable.h:18:15: error: forward declaration of ‘class yarp::os::Value’
/usr/local/src/robot/source/icub-main/src/libraries/icubmod/cfw2Can/Cfw2Can.cpp:117:69: error: invalid use of incomplete type ‘class yarp::os::Value’
In file included from /usr/local/src/robot/source/yarp/src/libYARP_OS/include/yarp/os/Contact.h:14:0,
                 from /usr/local/src/robot/source/yarp/src/libYARP_OS/include/yarp/os/ConnectionReader.h:14,
                 from /usr/local/src/robot/source/yarp/src/libYARP_dev/include/yarp/dev/DeviceDriver.h:13,
                 from /usr/local/src/robot/source/yarp/src/libYARP_dev/include/yarp/dev/CanBusInterface.h:13,
                 from /usr/local/src/robot/source/icub-main/src/libraries/icubmod/cfw2Can/Cfw2Can.cpp:14:
/usr/local/src/robot/source/yarp/src/libYARP_OS/include/yarp/os/Searchable.h:18:15: error: forward declaration of ‘class yarp::os::Value’
/usr/local/src/robot/source/icub-main/src/libraries/icubmod/cfw2Can/Cfw2Can.cpp:118:69: error: invalid use of incomplete type ‘class yarp::os::Value’
In file included from /usr/local/src/robot/source/yarp/src/libYARP_OS/include/yarp/os/Contact.h:14:0,
                 from /usr/local/src/robot/source/yarp/src/libYARP_OS/include/yarp/os/ConnectionReader.h:14,
                 from /usr/local/src/robot/source/yarp/src/libYARP_dev/include/yarp/dev/DeviceDriver.h:13,
                 from /usr/local/src/robot/source/yarp/src/libYARP_dev/include/yarp/dev/CanBusInterface.h:13,
                 from /usr/local/src/robot/source/icub-main/src/libraries/icubmod/cfw2Can/Cfw2Can.cpp:14:
/usr/local/src/robot/source/yarp/src/libYARP_OS/include/yarp/os/Searchable.h:18:15: error: forward declaration of ‘class yarp::os::Value’
/usr/local/src/robot/source/icub-main/src/libraries/icubmod/cfw2Can/Cfw2Can.cpp:120:69: error: invalid use of incomplete type ‘class yarp::os::Value’
In file included from /usr/local/src/robot/source/yarp/src/libYARP_OS/include/yarp/os/Contact.h:14:0,
                 from /usr/local/src/robot/source/yarp/src/libYARP_OS/include/yarp/os/ConnectionReader.h:14,
                 from /usr/local/src/robot/source/yarp/src/libYARP_dev/include/yarp/dev/DeviceDriver.h:13,
                 from /usr/local/src/robot/source/yarp/src/libYARP_dev/include/yarp/dev/CanBusInterface.h:13,
                 from /usr/local/src/robot/source/icub-main/src/libraries/icubmod/cfw2Can/Cfw2Can.cpp:14:
/usr/local/src/robot/source/yarp/src/libYARP_OS/include/yarp/os/Searchable.h:18:15: error: forward declaration of ‘class yarp::os::Value’
/usr/local/src/robot/source/icub-main/src/libraries/icubmod/cfw2Can/Cfw2Can.cpp:121:69: error: invalid use of incomplete type ‘class yarp::os::Value’
In file included from /usr/local/src/robot/source/yarp/src/libYARP_OS/include/yarp/os/Contact.h:14:0,
                 from /usr/local/src/robot/source/yarp/src/libYARP_OS/include/yarp/os/ConnectionReader.h:14,
                 from /usr/local/src/robot/source/yarp/src/libYARP_dev/include/yarp/dev/DeviceDriver.h:13,
                 from /usr/local/src/robot/source/yarp/src/libYARP_dev/include/yarp/dev/CanBusInterface.h:13,
                 from /usr/local/src/robot/source/icub-main/src/libraries/icubmod/cfw2Can/Cfw2Can.cpp:14:
/usr/local/src/robot/source/yarp/src/libYARP_OS/include/yarp/os/Searchable.h:18:15: error: forward declaration of ‘class yarp::os::Value’
make[2]: *** [src/libraries/icubmod/cfw2Can/CMakeFiles/cfw2can.dir/Cfw2Can.cpp.o] Error 1
make[1]: *** [src/libraries/icubmod/cfw2Can/CMakeFiles/cfw2can.dir/all] Error 2
make: *** [all] Error 2

```

Here is my CMakeCache for `icub-main`: 
[CMakeCache_icub-main.txt](https://github.com/robotology/icub-main/files/79744/CMakeCache_icub-main.txt)

and my CMakeCache for `yarp`: 
[CMakeCache_yarp.txt](https://github.com/robotology/icub-main/files/79746/CMakeCache_yarp.txt)

traversaro, 2016-01-06:
I think we need to add a include "yarp/OS/Value.h" include to Cfw2Can.cpp .

traversaro, 2016-01-06:
See https://github.com/robotology/icub-main/commit/5444e0016e467138bc2a89bf7a265e5258edaeb4 for a similar fix (sorry I cannot fix it bit I am writing from the phone).

traversaro, 2016-01-06:
See https://github.com/robotology/yarp/commit/31bdd9ab4f8cc9a2e0e9089afffdfc67b6159f69 for che yarp commit causing the bug.

rlober, 2016-01-06:
Fixed in https://github.com/robotology/icub-main/pull/270

rlober, 2016-01-06:
Thanks @traversaro !

closed 2016-01-06 by rlober