robotology/icub-main #269: Strange problem with yarpmanager

gregoire-pointeau, 2016-01-06:
Hi.
When using in Lyon `yarpmanager` I have a strange behavior. Most of "icub-related" application cannot be launch through yarpmanager (wholebody dynamics, cartesian solvers...) and I have to launch it by hand.
But some other can (camera, ARE ...).

I have checked carefully the name of the applications I launched and the name of the host and problem is not coming from here.

Does anyone has ever seen this king of behavior ?

Thank you.

pattacini, 2016-01-07:
Hi @gregoire-pointeau 

If you're referring to modules within the [`iCubStartup`](https://github.com/robotology/icub-main/blob/master/app/iCubStartup/scripts/iCubStartup.xml.template) application, be careful, since the launch of `WBD`, `iKinCartesianSolver` and other modules is postponed up to the point when `robotInterface` has complete its start-up procedure. The reason is that they all need to communicate with `robotInterface` to retrieve useful information before proceeding with their own initialization phase.

Notice for example the line https://github.com/robotology/icub-main/blob/master/app/iCubStartup/scripts/iCubStartup.xml.template#L60:

```xml
<port timeout="60.0" request="is_ready" reply="[ok]">/icub/robotInterface</port>
```

I don't know if this mechanism can explain what you've observed, but it might be related.

gregoire-pointeau, 2016-01-07:
Hi Ugo.
Thank for your comment.
However I don't think it comes from here. I already tried it: launch `robotInterface`, launch `WBD` in console, close it, launch it through `yarpmanager` and it doesn't work neither.

pattacini, 2016-01-07:
When you launch `robotInterface`, can you please check whether the port `/icub/robotInterface` is open?

Can you also check if the line mentioned above is present in the `iCubStartup.xml` for `WBD`?

gregoire-pointeau, 2016-01-07:
Hi Ugo, I confirm that the line is present in `iCubStartup.xml` moreover, the port `/icub/robotInterface` is open.

When I try to launch `iKinCartesianSolver --part right_arm`, is doesn't work with `yarpmanager`, even is my node (`icub-b1`) is detected as resource by `yarpmanager`.

When I ssh my resource `icub-b1` and launch it by hand, it works.
However, if I launch `objectsPropertiesCollector` through `yarpmanager` on `icub-b1`, it works also fine.

pattacini, 2016-01-07:
As double-check you could remove the line under exam:

```xml
<port timeout="60.0" request="is_ready" reply="[ok]">/icub/robotInterface</port>
```
from `iCubStartup.xml` for what concerns - for example-  `iKinGazeCtrl`.

Then, you can launch `robotInterface`, wait until the calibration is over and finally launch `iKinGazeCtrl` from the gui.

If it will work, the problem is due to somehow the communication between the module and the `robotInterface`.

If it won't work yet, the problem is likely to be caused by `yarprun` and/or `yarpmanager`.

Therefore, this test will help us narrow down the fault condition.

gregoire-pointeau, 2016-01-07:
I confirm your though, if I remove the dependence, it works, if the dependence is here, it doesn't work.
So it seems to be a problem coming from our `robotInterface`

pattacini, 2016-01-07:
Ok then, since here everything is working, I tend to consider that your `PC104` is running an out-of-date code.

My suggestion therefore is to update your system, `PC104` included.

##### Caveat :exclamation: 

**The most recent `icub-main` master will require a firmware update too**

Therefore, if you'd like to schedule such an update in a more convenient moment, you could consider removing those dependencies from `iCubStartup.xml` for the time being.

pattacini, 2016-01-07:
Closing this for now.
@gregoire-pointeau feel free to reopen it

/cc @randaz81

closed 2016-01-07 by pattacini

gregoire-pointeau, 2016-01-07:
Well, it's kind of strange because we just made a full update at iit of our system in july. How can the system be already out-of-date ?

pattacini, 2016-01-07:
July? It's a life ago ...
:smirk:

pattacini, 2016-01-07:
You can see on GitHub how many new commits `icub-main` has since July ...

gregoire-pointeau, 2016-01-07:
when you say update, it's just update the sources, and recompile it, not the pc104 that need a new OS update ?

pattacini, 2016-01-07:
Yes, I'm referring to `FW` and `SW` update, not to `OS` update.

gregoire-pointeau, 2016-01-07:
Well, it is also a problem that I have detected with yarpviews. I cannot open a `yarpview` through `yarpmanager`. So I'm not sure it comes only from iCub.

pattacini, 2016-01-07:
I think the two behaviors are not related to the same problem.
Anyway, it's always a good practice to keep the code up-to-date.

gregoire-pointeau, 2016-01-07:
For exmaple in the application: `Calibrated Cameras`: 
`yarpdec`, `camcalib` work normally, but `yarpview` needs to be launched by hand.

lornat75, 2016-01-08:
@gregoire-pointeau If you are running icubcluster to execute yarprun, guis may not work if you do not enable the display option in the node description:

  <node display="true">nodename</node>

gregoire-pointeau, 2016-01-12:
Thanks. Where should I had the line ? In the `icub-cluster.py` or in the `yarpmanager` application `*.xml` ?

pattacini, 2016-01-12:
@gregoire-pointeau it's in the `icub-cluster.py`.

gregoire-pointeau, 2016-01-12:
I added the `display="true"` in my `cluster-config.xml`.
However it doesn't change anything.
I still have the `iCubStartup cannot run iKinCartesianSolver on icub-b1: Timeout while waiting for some resources.`

pattacini, 2016-01-12:
As said, they are two distinct problem.

Playing with the `display` property won't solve the `iCubStartup.xml`-related issues, unfortunately :disappointed: 

For that, you'd need a full `yarp`+`icub-main` update to begin with (bearing the above caveat in mind).

gregoire-pointeau, 2016-01-12:
Update: at least, adding the display solved the problem for applications such as `yarpview`.
Concerning `robotInterface`, I'll make a full upgrade soon (maybe I'll need some help I'm not a debian expert, and Anne-Laure is not there until March).

Thanks anyway !

pattacini, 2016-01-12:
Cool :+1:

pattacini, 2016-01-12:
Well, I also said that this is a minor issue, since you could consider removing those protections within `iCubStartup.xml` and carry out the update once you feel safer.

However, we are to help you out :smile:

lornat75, 2016-01-12:
About the display option see issue: #272