robotology/icub-main #271: [robotInterface] Interesting sensor failures with new firmware and icub-main

rlober, 2016-01-07:
Not sure if this is exactly the best place for this issue but I figured because it happens in `robotInterface` why not.

Anyways, I recently updated iCubParis02 to the newest firmware and the latest `yarp` and `icub-main` and I am getting some interesting error messages from `robotInterface`. I say interesting because I have never seen them before and I am not sure where to start resolving them. After speaking with @julijenv it seems there have been some big changes to the firmware before the holidays and I am wondering if we are seeing some new bugs. Anyways here are some examples of these errors:

```
3015		32,457671		ERROR		AxisName not found
3493		36,865772		ERROR		Trying to get fullscale data from sensor 14 net [1]: no answer received or message lost (ch:0)
5206		212,515410		ERROR		AnalogWrapper: left_leg: Sensor returned timeout error (code 3).
```
Here is the full log: [log_pc104_robotInterface_5489.txt](https://github.com/robotology/icub-main/files/81375/log_pc104_robotInterface_5489.txt)

Addmittedly these *could* be hardware related, but I am not entirely sure where to start with the debugging. 

For reference here is the `CMakeCache` of `icub-main` for pc104: [CMakeCache_icub-main-build-pc104.txt](https://github.com/robotology/icub-main/files/81368/CMakeCache_icub-main-build-pc104.txt)

cc: @randaz81

Tobias-Fischer, 2016-01-07:
See e.g. https://github.com/robotology/icub-main/commit/8c0c9ae93a77565247adca016ef9f9138578fc5c for the `AxisName` problem.

No clue about the other stuff ;)

Tobias-Fischer, 2016-01-07:
Ahh, for the warnings, add `<!DOCTYPE devices PUBLIC "-//YARP//DTD robotInterface 1.0//EN" "http://www.icub.org/DTD/robotInterfaceV1.0.dtd">` to the XML files. Again, https://github.com/robotology/icub-main/commit/8c0c9ae93a77565247adca016ef9f9138578fc5c shows an example.

rlober, 2016-01-07:
@Tobias-Fischer 

Thanks for the info! I will make those changes.

rlober, 2016-01-08:
**Updates: New error**
```
5130 311,814836 5 recv buffer overrun ( 500 >  500) 
```
After running for about 5 min `robotInterface` completely crashes pc104 and I get this `buffer overrun` error.

drdanz, 2016-01-08:
Moved to robotology/icub-support#149. Closing this

closed 2016-01-08 by drdanz