robotology/icub-main #273: Setting AxisName and AxisType in robotInterface configuration files

gsaponaro, 2016-01-13:
Hi, **robotInterface** prints errors because AxisName and AxisType parameters are not set for any joint in our robot configuration files (iCubLisboa01).

Example:
```
[INFO]created device <canmotioncontrol>. See C++ class yarp::dev::CanBusMotionControl for documentation.
[ERROR]AxisName not found
[ERROR]AxisType not found
[INFO]Position Pids section found, new format
[INFO]Position Pids successfully loaded
[INFO]TRQ_PIDS: Torque Pids section found, new format
[DEBUG]DEBUG parameters section NOT found, skipping...
[DEBUG]IMPEDANCE parameters section NOT found, skipping...
[WARNING]maxPosStep: Using default MaxPosStep=10 degs
```

Can you tell us how to set those parameters in the XMLs (probably we can "copy" them from another robot configuration, but which one?) Or can you commit the changes in the appropriate iCubLisboa01 files?

In previous issues it was suggested not to set or commit those parameters by ourselves (https://github.com/robotology/QA/issues/76, https://github.com/robotology/icub-main/issues/229).

We also have another unrelated issue related to CAN bus which prevents us from starting **robotInterface** at the moment (will open an issue in icub-support soon), so we are not sure if those AxisName errors that show up in the log are actually critical or not  - in the meantime we thought it would be a good idea for us to fix them anyway.

/cc @randaz81

Tobias-Fischer, 2016-01-18:
We also have the `AxisType` issue on `iCubLondon01`. I am not sure why not all robots were included when the configuration files were changed in https://github.com/robotology/icub-main/commit/abcfecde9b801ba4b5808d8e61758f9898240117

@randaz81: Can I just copy over the changes from e.g. `iCubGenova01` to `iCubLondon01`?

jypuigbo, 2016-01-26:
We are having the same problem with ```iCubBarcelona01```.

gsaponaro, 2016-02-04:
In addition to ```AxisName``` and ```AxisType```, now (after a general update) we also get the errors ```motorPwmLimit not found```.

Fresh **robotInterface** log here: https://gist.github.com/gsaponaro/0c74a6bc0963deea6c7b

Our software and firmware (build 87) are up to date on both pc104 and cluster. Our robot XML configuration files have been re-installed to ```/home/icub/.local/share/yarp/robots/iCubLisboa01```.

Tobias-Fischer, 2016-02-04:
@gsaponaro 
Seems like `iCubLisboa01`, `iCubLondon01` and others missed out in getting the `motorPwmLimit` configuration; see https://github.com/robotology/icub-main/commit/dba955a8944cbd6e76b9455d74f95082a261f8a5

Tobias-Fischer, 2016-02-08:
Please note that I am not planning to change any other configuration files than the ones of `iCubLondon01` .. This problem seems to be a general one though. Shall I re-assign @randaz81?

/cc @drdanz

gsaponaro, 2016-02-10:
A few messages above, @jypuigbo reported that their robot in Barcelona has the same problem (requires XML updates). Possibly other robots were left out, too. I'll open a new issue for this.

Tobias-Fischer, 2016-02-10:
Shall we open an issue regarding `motorPwmLimit` as well?

gsaponaro, 2016-02-10:
Hi @Tobias-Fischer, I'm not sure if the issue there is that we actually need the parameter, or we don't need it but **robotInterface** still reports the error for some reason. At the moment it seems like only robots with ems have that parameter set, for example it is set in ```iCubGenova02/hardware/motorControl/left_lower_leg-ems7-mc.xml```.

@randaz81, any suggestion what we should do with regard to the ```motorPwmLimit``` parameter on ```iCubLisboa01```, ```iCubLondon01``` etc. configurations?

randaz81, 2016-02-11:
Hi @gsaponaro,
the right values for parameter `motorPwmLimit` are:
* `ETH robots:` 10000
* `CAN robots:` 1333

On ETH robots this parameter is mandatory.
On CAN robots we are currently using a default value of 1333 (hidden in the source code). However, we using the label ERROR label to remember the users that soon or later this default value will be removed, and the parameter will become mandatory. So better to update the files in advance.
PS: I apologize that we didn't update the robot configuration files. We started to prepare a script that fixes all the robots at once, but given the large amount of work, I have honestly to say that currently it's scheduled with low priority :-(

gsaponaro, 2016-02-11:
Thanks a lot for the clarification, @randaz81. I'm preparing a pull request to update our CAN robot configuration by myself. Just a question about the torso file (```iCubLisboa01/hardware/motorControl/icub_torso.xml```): should the last joint also have motorPwmLimit=1333, or 0? That joint has zeros for other fields.

The relative section currently looks like this:
```
//                           "torso_yaw" "torso_roll" "torso_pitch" "none"
<group name="LIMITS">        
<param name="jntPosMin">          -50           -30           -22           0             </param>       
<param name="jntPosMax">          50            30            70            0             </param>       
<param name="motorOverloadCurrents">     7000          7000          7000          0             </param>       
```

randaz81, 2016-02-11:
Let's use 0 to emphasize the fact the the fourth joint of the torso does not exists in reality.

In any case, iCub will work even if this parameter is set to XXXX: the controlBoard firmware will simply internally store the variable but it will not apply it to any control. (The fact that in the torso we need four values even if only three are actually used is due to some simplifications in the implementation which assumes that the hardware output stages only exist in groups of two)

gsaponaro, 2016-02-11:
OK cool, thanks for the further explanation (sometimes details about these things escape me, so I prefer to ask :) ).

closed 2016-02-11 by randaz81