robotology/icub-main #276: robotInterface: corrupted double-linked list after CTRL+C

davidetome, 2016-01-19:
*** glibc detected *** robotInterface: corrupted double-linked list: 0x000000000c75c800 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x75be6)[0x7fa5ad13abe6]
/lib/x86_64-linux-gnu/libc.so.6(+0x7605d)[0x7fa5ad13b05d]
/lib/x86_64-linux-gnu/libc.so.6(+0x77518)[0x7fa5ad13c518]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x6c)[0x7fa5ad13f98c]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.1(_ZN4yarp2os4impl14PortCorePacket5resetEv+0x2e)[0x7fa5ae2ad89e]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.1(_ZN26PortWriterBufferBaseHelperD0Ev+0x15b)[0x7fa5ae2c282b]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.1(_ZN4yarp2os20PortWriterBufferBaseD1Ev+0x21)[0x7fa5ae2c1911]
/usr/local/src/robot/icub-main/build-pc104/lib/iCub/skinWrapper.so(_ZN4yarp3dev12AnalogServerD1Ev+0x447)[0x7fa5765d3dc7]
/usr/local/src/robot/icub-main/build-pc104/lib/iCub/skinWrapper.so(_ZN4yarp3dev12AnalogServerD0Ev+0x9)[0x7fa5765d4269]
/usr/local/src/robot/icub-main/build-pc104/lib/iCub/skinWrapper.so(_ZN11skinWrapper5closeEv+0x22)[0x7fa5765c3522]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.1(_ZN4yarp3dev10PolyDriver9closeMainEv+0x44)[0x7fa5adeb9d34]
robotInterface(_ZN14RobotInterface6Device5closeEv+0x20)[0x4168d0]
robotInterface(_ZN14RobotInterface5Robot7Private12closeDevicesEv+0x46)[0x41ecd6]
robotInterface(_ZN14RobotInterface5Robot10enterPhaseENS_11ActionPhaseE+0x1bee)[0x424ebe]
robotInterface(_ZN14RobotInterface6Module5closeEv+0x55)[0x41a605]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.1(_ZN4yarp2os8RFModule9runModuleEv+0x211)[0x7fa5ae2e2b61]
robotInterface(main+0xe2)[0x413c82]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xfd)[0x7fa5ad0e3ead]
robotInterface[0x413fd9]
======= Memory map: ========
00400000-00450000 r-xp 00000000 00:17 12324236                           /usr/local/src/robot/icub-main/build-pc104/bin/robotInterface
00650000-00652000 rw-p 00050000 00:17 12324236                           /usr/local/src/robot/icub-main/build-pc104/bin/robotInterface
006d2000-0e1ad000 rw-p 00000000 00:00 0                                  [heap]
7fa5447f9000-7fa5447fa000 ---p 00000000 00:00 0 
7fa5447fa000-7fa544ffa000 rw-p 00000000 00:00 0 
7fa546ffe000-7fa546fff000 ---p 00000000 00:00 0 
7fa546fff000-7fa5477ff000 rw-p 00000000 00:00 0 
7fa548000000-7fa54803b000 rw-p 00000000 00:00 0 
7fa54803b000-7fa54c000000 ---p 00000000 00:00 0 
7fa54d7cb000-7fa54d7cc000 ---p 00000000 00:00 0 
7fa54d7cc000-7fa54dfcc000 rw-p 00000000 00:00 0 
7fa54f7cf000-7fa54f7d0000 ---p 00000000 00:00 0 
7fa54f7d0000-7fa54ffd0000 rw-p 00000000 00:00 0 
7fa5517d3000-7fa5517d4000 ---p 00000000 00:00 0 
7fa5517d4000-7fa551fd4000 rw-p 00000000 00:00 0 
7fa5537d7000-7fa5537d8000 ---p 00000000 00:00 0 
7fa5537d8000-7fa553fd8000 rw-p 00000000 00:00 0 
7fa568000000-7fa5680b5000 rw-p 00000000 00:00 0 
7fa5680b5000-7fa56c000000 ---p 00000000 00:00 0 
7fa56f185000-7fa56f19b000 r-xp 00000000 00:17 12323984                   /usr/local/src/robot/icub-main/build-pc104/lib/iCub/parametricCalibrator.so
7fa56f19b000-7fa56f39a000 ---p 00016000 00:17 12323984                   /usr/local/src/robot/icub-main/build-pc104/lib/iCub/parametricCalibrator.so
7fa56f39a000-7fa56f39b000 rw-p 00015000 00:17 12323984                   /usr/local/src/robot/icub-main/build-pc104/lib/iCub/parametricCalibrator.so
7fa56f39b000-7fa56f39c000 ---p 00000000 00:00 0 
7fa56f39c000-7fa56fb9c000 rw-p 00000000 00:00 0 
7fa56fb9c000-7fa56fb9d000 ---p 00000000 00:00 0 
7fa56fb9d000-7fa57039d000 rw-p 00000000 00:00 0 
7fa57039d000-7fa57039e000 ---p 00000000 00:00 0 
7fa57039e000-7fa570b9e000 rw-p 00000000 00:00 0 
7fa570b9e000-7fa570baa000 r-xp 00000000 00:17 12324003                   /usr/local/src/robot/icub-main/build-pc104/lib/iCub/xsensmtx.so
7fa570baa000-7fa570da9000 ---p 0000c000 00:17 12324003                   /usr/local/src/robot/icub-main/build-pc104/lib/iCub/xsensmtx.so
7fa570da9000-7fa570daa000 rw-p 0000b000 00:17 12324003                   /usr/local/src/robot/icub-main/build-pc104/lib/iCub/xsensmtx.so
7fa570daa000-7fa570dab000 ---p 00000000 00:00 0 
7fa570dab000-7fa5715ab000 rw-p 00000000 00:00 0 
7fa5715ab000-7fa5715ac000 ---p 00000000 00:00 0 
7fa5715ac000-7fa571dac000 rw-p 00000000 00:00 0 
7fa571dac000-7fa571dad000 ---p 00000000 00:00 0 
7fa571dad000-7fa5725ad000 rw-p 00000000 00:00 0 
7fa5725ad000-7fa5725ae000 ---p 00000000 00:00 0 
7fa5725ae000-7fa572dae000 rw-p 00000000 00:00 0 
7fa572dae000-7fa572daf000 ---p 00000000 00:00 0 
7fa572daf000-7fa5735af000 rw-p 00000000 00:00 0 
7fa5735af000-7fa5735b0000 ---p 00000000 00:00 0 
7fa5735b0000-7fa573db0000 rw-p 00000000 00:00 0 
7fa573db0000-7fa573db1000 ---p 00000000 00:00 0 
7fa573db1000-7fa5745b1000 rw-p 00000000 00:00 0 
7fa5745b1000-7fa5745b2000 ---p 00000000 00:00 0 
7fa5745b2000-7fa574db2000 rw-p 00000000 00:00 0 
7fa574db2000-7fa574db3000 ---p 00000000 00:00 0 
[... 94 lines cut ...]
7fa5ad8e7000-7fa5ad8e8000 rw-p 00081000 07:00 10140                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/libm-2.13.so
7fa5ad8e8000-7fa5ad9d0000 r-xp 00000000 07:00 25245                      /lib/live/mount/rootfs/filesystem.squashfs/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.17
7fa5ad9d0000-7fa5adbd0000 ---p 000e8000 07:00 25245                      /lib/live/mount/rootfs/filesystem.squashfs/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.17
7fa5adbd0000-7fa5adbd8000 r--p 000e8000 07:00 25245                      /lib/live/mount/rootfs/filesystem.squashfs/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.17
7fa5adbd8000-7fa5adbda000 rw-p 000f0000 07:00 25245                      /lib/live/mount/rootfs/filesystem.squashfs/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.17
7fa5adbda000-7fa5adbef000 rw-p 00000000 00:00 0 
7fa5adbef000-7fa5adbf1000 r-xp 00000000 00:17 12321669                   /usr/local/src/robot/yarp/build-pc104/lib/libYARP_init.so.2.3.64.10
7fa5adbf1000-7fa5addf0000 ---p 00002000 00:17 12321669                   /usr/local/src/robot/yarp/build-pc104/lib/libYARP_init.so.2.3.64.10
7fa5addf0000-7fa5addf1000 rw-p 00001000 00:17 12321669                   /usr/local/src/robot/yarp/build-pc104/lib/libYARP_init.so.2.3.64.10
7fa5addf1000-7fa5adf8b000 r-xp 00000000 00:17 12322244                   /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.64.10
7fa5adf8b000-7fa5ae18a000 ---p 0019a000 00:17 12322244                   /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.64.10
7fa5ae18a000-7fa5ae19b000 rw-p 00199000 00:17 12322244                   /usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.2.3.64.10
7fa5ae19b000-7fa5ae36f000 r-xp 00000000 00:17 12321665                   /usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.2.3.64.10
7fa5ae36f000-7fa5ae56f000 ---p 001d4000 00:17 12321665                   /usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.2.3.64.10
7fa5ae56f000-7fa5ae57e000 rw-p 001d4000 00:17 12321665                   /usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.2.3.64.10
7fa5ae57e000-7fa5ae59e000 r-xp 00000000 07:00 10285                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/ld-2.13.so
7fa5ae782000-7fa5ae78a000 rw-p 00000000 00:00 0 
7fa5ae799000-7fa5ae79d000 rw-p 00000000 00:00 0 
7fa5ae79d000-7fa5ae79e000 r--p 0001f000 07:00 10285                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/ld-2.13.so
7fa5ae79e000-7fa5ae79f000 rw-p 00020000 07:00 10285                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/ld-2.13.so
7fa5ae79f000-7fa5ae7a0000 rw-p 00000000 00:00 0 
7fffda441000-7fffda462000 rw-p 00000000 00:00 0                          [stack]
7fffda561000-7fffda562000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted

francesco-romano, 2016-01-25:
I add also this log:

```
*** glibc detected *** robotInterface: corrupted double-linked list: 0x00007fee8a8f0ff0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x75be6)[0x7feeb23bcbe6]
/lib/x86_64-linux-gnu/libc.so.6(+0x7605d)[0x7feeb23bd05d]
/lib/x86_64-linux-gnu/libc.so.6(+0x77518)[0x7feeb23be518]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x6c)[0x7feeb23c198c]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.1(_ZN10StubDriverD0Ev+0x201)[0x7feeb313a261]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.1(_ZN4yarp3dev10PolyDriver9closeMainEv+0x53)[0x7feeb3144443]
robotInterface(_ZN14RobotInterface6Device5closeEv+0x20)[0x416a80]
robotInterface(_ZN14RobotInterface5Robot7Private12closeDevicesEv+0x46)[0x41ee86]
robotInterface(_ZN14RobotInterface5Robot10enterPhaseENS_11ActionPhaseE+0x1bee)[0x42506e]
robotInterface(_ZN14RobotInterface6Module5closeEv+0x55)[0x41a7b5]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.1(_ZN4yarp2os8RFModule9runModuleEv+0x211)[0x7feeb3585061]
robotInterface(main+0x140)[0x413d90]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xfd)[0x7feeb2365ead]
robotInterface[0x414109]
======= Memory map: ========
00400000-00451000 r-xp 00000000 00:16 930116                             /usr/local/src/robot/icub-main/build-pc104/bin/robotInterface
00650000-00652000 rw-p 00050000 00:16 930116                             /usr/local/src/robot/icub-main/build-pc104/bin/robotInterface
01de9000-02770000 rw-p 00000000 00:00 0                                  [heap]
7fee44000000-7fee44117000 rw-p 00000000 00:00 0 
7fee44117000-7fee48000000 ---p 00000000 00:00 0 
7fee74ce9000-7fee74cff000 r-xp 00000000 00:16 921952                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/parametricCalibratorEth.so
7fee74cff000-7fee74eff000 ---p 00016000 00:16 921952                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/parametricCalibratorEth.so
7fee74eff000-7fee74f00000 rw-p 00016000 00:16 921952                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/parametricCalibratorEth.so
7fee74f00000-7fee74f7f000 r-xp 00000000 00:16 930052                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/embObjSkin.so
7fee74f7f000-7fee7517e000 ---p 0007f000 00:16 930052                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/embObjSkin.so
7fee7517e000-7fee75183000 rw-p 0007e000 00:16 930052                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/embObjSkin.so
7fee75183000-7fee75184000 ---p 00000000 00:00 0 
7fee75184000-7fee75984000 rw-p 00000000 00:00 0 
7fee75984000-7fee75985000 ---p 00000000 00:00 0 
7fee75985000-7fee76185000 rw-p 00000000 00:00 0 
7fee76185000-7fee761a8000 r-xp 00000000 00:16 920062                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/canBusSkin.so
7fee761a8000-7fee763a7000 ---p 00023000 00:16 920062                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/canBusSkin.so
7fee763a7000-7fee763a9000 rw-p 00022000 00:16 920062                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/canBusSkin.so
7fee763a9000-7fee763aa000 ---p 00000000 00:00 0 
7fee763aa000-7fee76baa000 rw-p 00000000 00:00 0 
7fee76baa000-7fee76bab000 ---p 00000000 00:00 0 
7fee76bab000-7fee773ab000 rw-p 00000000 00:00 0 
7fee773ab000-7fee773ac000 ---p 00000000 00:00 0 
7fee773ac000-7fee77bac000 rw-p 00000000 00:00 0 
7fee77bac000-7fee77bad000 ---p 00000000 00:00 0 
7fee77bad000-7fee783ad000 rw-p 00000000 00:00 0 
7fee783ad000-7fee783ae000 ---p 00000000 00:00 0 
7fee783ae000-7fee78bae000 rw-p 00000000 00:00 0 
7fee78bae000-7fee78baf000 ---p 00000000 00:00 0 
7fee78baf000-7fee793af000 rw-p 00000000 00:00 0 
7fee793af000-7fee793d5000 r-xp 00000000 00:16 921975                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/skinWrapper.so
7fee793d5000-7fee795d5000 ---p 00026000 00:16 921975                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/skinWrapper.so
7fee795d5000-7fee795d7000 rw-p 00026000 00:16 921975                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/skinWrapper.so
7fee79dd8000-7fee79dd9000 ---p 00000000 00:00 0 
7fee79dd9000-7fee7a5d9000 rw-p 00000000 00:00 0 
7fee7adda000-7fee7addb000 ---p 00000000 00:00 0 
7fee7addb000-7fee7b5db000 rw-p 00000000 00:00 0 
7fee7f5e3000-7fee7f5ef000 r-xp 00000000 00:16 921894                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/xsensmtx.so
7fee7f5ef000-7fee7f7ee000 ---p 0000c000 00:16 921894                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/xsensmtx.so
7fee7f7ee000-7fee7f7ef000 rw-p 0000b000 00:16 921894                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/xsensmtx.so
7fee88000000-7fee8befc000 rw-p 00000000 00:00 0 
7fee8befc000-7fee8c000000 ---p 00000000 00:00 0 
7fee8d933000-7fee8d9ac000 r-xp 00000000 00:16 930054                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/embObjAnalogSensor.so
7fee8d9ac000-7fee8dbac000 ---p 00079000 00:16 930054                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/embObjAnalogSensor.so
7fee8dbac000-7fee8dbb0000 rw-p 00079000 00:16 930054                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/embObjAnalogSensor.so
7fee9e3d1000-7fee9e496000 r-xp 00000000 00:16 930053                     /usr/local/src/robot/icub-main/build-pc104/lib/iCub/embObjMotionControl.so
[... cut ...]
7feeb3a40000-7feeb3a41000 r--p 0001f000 07:00 10283                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/ld-2.13.so
7feeb3a41000-7feeb3a42000 rw-p 00020000 07:00 10283                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/ld-2.13.so
7feeb3a42000-7feeb3a43000 rw-p 00000000 00:00 0 
7fff73652000-7fff73673000 rw-p 00000000 00:00 0                          [stack]
7fff737b6000-7fff737b7000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted

```

traversaro, 2016-01-25:
Related issue to https://github.com/robotology/icub-main/issues/275 .

gsaponaro, 2016-02-04:
We also experience a segmentation fault when quitting **robotInterface** on ```iCubLisboa01``` (after having done a fresh update on pc104 & cluster, including upgrading the firmware to the latest build 87).

```
[INFO]All actions for action level 1 of interrupt1 phase finished. 
[INFO]interrupt1 phase finished. 
[INFO]shutdown phase starting... 
[INFO]Entering action level 2 of phase shutdown 
[INFO]Stopping icub/cartesianController/left_arm
[INFO]Stopping icub/cartesianController/right_arm
[INFO]All actions for action level 2 of shutdown phase started. Waiting for unfinished actions. 
[INFO]All actions for action level 2 of shutdown phase finished. 
[INFO]Entering action level 5 of phase shutdown 
[INFO]All actions for action level 5 of shutdown phase started. Waiting for unfinished actions. 
[INFO]All actions for action level 5 of shutdown phase finished. 
Segmentation fault
```

gsaponaro, 2016-02-04:
Backtrace and memory map: https://gist.github.com/gsaponaro/ffa7734357c58413deb6

lornat75, 2016-02-10:
@barbalberto can we try to replicate this and fix it?

francesco-romano, 2016-03-14:
@marcoaccame 
I don't know if the error we'd just seen is the same

traversaro, 2016-03-17:
@barbalberto some juicy details obtained by running canLoader in Valgrind : 

~~~
icub@pc104:/usr/local/src/robot/icub-main/build-pc104$ valgrind canLoader
==26087== Memcheck, a memory error detector
==26087== Copyright (C) 2002-2011, and GNU GPL'd, by Julian Seward et al.
==26087== Using Valgrind-3.7.0 and LibVEX; rerun with -h for copyright info
==26087== Command: canLoader
==26087== 
Initializing GUI version of canloader
Gtk-Message: Failed to load module "canberra-gtk-module"
==26087== Conditional jump or move depends on uninitialised value(s)
==26087==    at 0x8121AEB: myMain(int, char**) (main.cpp:1903)
==26087== 
==26087== Conditional jump or move depends on uninitialised value(s)
==26087==    at 0x8121AED: myMain(int, char**) (main.cpp:1903)
==26087== 
==26087== Syscall param ioctl(generic) points to unaddressable byte(s)
==26087==    at 0x57E1D19: ioctl (syscall-template.S:82)
==26087==    by 0x403FAC4: cfwCanRtxEnable (in /lib/modules/3.2.0-4-686-pae/iCubDrivers/cfw002/LinuxDriver/API/libcfw002.so)
==26087==    by 0xB8538A7: yarp::dev::Cfw2Can::open(yarp::os::Searchable&) (Cfw2Can.cpp:131)
==26087==    by 0x823A76F: StubDriver::open(yarp::os::Searchable&) (Drivers.cpp:199)
==26087==    by 0x8234DA7: yarp::dev::PolyDriver::coreOpen(yarp::os::Searchable&) (PolyDriver.cpp:265)
==26087==    by 0x8235788: yarp::dev::PolyDriver::open(yarp::os::Searchable&) (PolyDriver.cpp:120)
==26087==    by 0x79A04F7: ???
==26087==  Address 0x15 is not stack'd, malloc'd or (recently) free'd
==26087== 
[INFO]created device <cfw2can>. See C++ class yarp::dev::Cfw2Can for documentation.
yarp: Port /log/pc104/no_progname/26087 active at tcp://10.0.0.2:10003
yarp: Sending output from /log/pc104/no_progname/26087 to /yarplogger using tcp
==26087== Syscall param ioctl(generic) points to uninitialised byte(s)
==26087==    at 0x57E1D19: ioctl (syscall-template.S:82)
==26087==    by 0x403F986: cfwCanRead (in /lib/modules/3.2.0-4-686-pae/iCubDrivers/cfw002/LinuxDriver/API/libcfw002.so)
==26087==    by 0xB85318F: yarp::dev::Cfw2Can::canRead(yarp::dev::CanBuffer&, unsigned int, unsigned int*, bool) (Cfw2Can.cpp:77)
==26087==    by 0x812F54F: cDriver::receive_message(yarp::dev::CanBuffer&, int, double) (driver.cpp:79)
==26087==    by 0x812BD10: cDownloader::initschede() (downloader.cpp:1255)
==26087==    by 0x811EA0A: start_end_click(_GtkButton*, void*) (main.cpp:793)
==26087==    by 0x4150C56: g_cclosure_marshal_VOID__VOIDv (in /usr/lib/i386-linux-gnu/libgobject-2.0.so.0.3200.4)
==26087==    by 0x414F120: ??? (in /usr/lib/i386-linux-gnu/libgobject-2.0.so.0.3200.4)
==26087==    by 0x4168278: g_signal_emit_valist (in /usr/lib/i386-linux-gnu/libgobject-2.0.so.0.3200.4)
==26087==    by 0x4168CD2: g_signal_emit (in /usr/lib/i386-linux-gnu/libgobject-2.0.so.0.3200.4)
==26087==    by 0x4611539: gtk_button_clicked (in /usr/lib/i386-linux-gnu/libgtk-x11-2.0.so.0.2400.10)
==26087==    by 0x461293F: ??? (in /usr/lib/i386-linux-gnu/libgtk-x11-2.0.so.0.2400.10)
==26087==  Address 0xbeaf50e0 is on thread 1's stack
==26087== 
[INFO]CONNECTED: 2 Boards found on bus 1
[DEBUG]    id   type     version
[DEBUG]   14       7     3.1.2
[DEBUG]    7       3     2.48.87
==26087== Syscall param ioctl(generic) points to unaddressable byte(s)
==26087==    at 0x57E1D19: ioctl (syscall-template.S:82)
==26087==    by 0x403FB01: cfwCanRtxDisable (in /lib/modules/3.2.0-4-686-pae/iCubDrivers/cfw002/LinuxDriver/API/libcfw002.so)
==26087==    by 0xB85329A: yarp::dev::Cfw2Can::close() (Cfw2Can.cpp:147)
==26087==    by 0x823A797: StubDriver::close() (Drivers.cpp:204)
==26087==    by 0x8234006: yarp::dev::PolyDriver::closeMain() (PolyDriver.cpp:138)
==26087==    by 0x8234074: yarp::dev::PolyDriver::~PolyDriver() (PolyDriver.cpp:172)
==26087==    by 0x812F93F: cDriver::~cDriver() (driver.h:40)
==26087==    by 0x812899B: cDownloader::stopdriver() (downloader.cpp:107)
==26087==    by 0x811EAA3: start_end_click(_GtkButton*, void*) (main.cpp:815)
==26087==    by 0x4150C56: g_cclosure_marshal_VOID__VOIDv (in /usr/lib/i386-linux-gnu/libgobject-2.0.so.0.3200.4)
==26087==    by 0x414F120: ??? (in /usr/lib/i386-linux-gnu/libgobject-2.0.so.0.3200.4)
==26087==    by 0x4168278: g_signal_emit_valist (in /usr/lib/i386-linux-gnu/libgobject-2.0.so.0.3200.4)
==26087==  Address 0x5 is not stack'd, malloc'd or (recently) free'd
==26087== 
yarp: Removing output from /log/pc104/no_progname/26087 to /yarplogger
==26087== Invalid read of size 4
==26087==    at 0x81165CE: _ZN9__gnu_cxxL27__exchange_and_add_dispatchEPii.constprop.17 (atomicity.h:48)
==26087==    by 0x574129E: exit (exit.c:100)
==26087==    by 0x5728E4D: (below main) (libc-start.c:276)
==26087==  Address 0xb082440 is 8 bytes inside a block of size 24 free'd
[... 147 lines cut ...]
==26087==    at 0x815B363: yarp::os::Semaphore::wait() (POSIXSemaphoreImpl.h:26)
==26087==    by 0x574129E: exit (exit.c:100)
==26087==    by 0x5728E4D: (below main) (libc-start.c:276)
==26087==  If you believe this happened as a result of a stack
==26087==  overflow in your program's main thread (unlikely but
==26087==  possible), you can try to increase the size of the
==26087==  main thread stack using the --main-stacksize= flag.
==26087==  The main thread stack size used in this run was 8388608.
==26087== 
==26087== HEAP SUMMARY:
==26087==     in use at exit: 1,304,772 bytes in 10,160 blocks
==26087==   total heap usage: 553,633 allocs, 543,536 frees, 44,817,919 bytes allocated
==26087== 
==26087== LEAK SUMMARY:
[... cut ...]
==26087==    indirectly lost: 10,539 bytes in 471 blocks
==26087==      possibly lost: 841,073 bytes in 3,526 blocks
==26087==    still reachable: 426,392 bytes in 6,147 blocks
==26087==         suppressed: 0 bytes in 0 blocks
==26087== Rerun with --leak-check=full to see details of leaked memory
==26087== 
==26087== For counts of detected and suppressed errors, rerun with: -v
==26087== Use --track-origins=yes to see where uninitialised values come from
==26087== ERROR SUMMARY: 250950 errors from 16 contexts (suppressed: 177 from 10)
~~~

pattacini, 2016-03-17:
cc @Tobias-Fischer

traversaro, 2016-03-17:
I don't have time to further investigate it, but I guess the problem could be related to the fact that `yarp::os::Companion` class and the fact that somewhere it is defined as a static instance. that is then duplicated in the `robotInterface/canLoader` binary and in the `cfw2can.so` device that is loaded at runtime... but in this case it should happen with all devices loaded at runtime.

barbalberto, 2016-03-17:
AFAIK it seems to happen only on can robot and not in eth ones... maybe @randaz81 can help more on this regard?

randaz81, 2016-03-17:
Well, every ETH robot has at least one CAN network (the head). I'm familiar with this error on ETH robots too, but I never tried on a ETH robot with the head disabled.
Additionally, I saw this error also running robotInterface on non-robot-devices!! (e.g. the rpLidar)

pattacini, 2016-03-24:
Sooner or later this bug should be put at the top of our to-fix list.

traversaro, 2016-04-12:
Just to clarify, this happends also on ETH robots, here an example on `iCubDarmstadt01`:
~~~
[DEBUG]right_arm_calibrator finished park device right_arm_mc_wrapper 
[DEBUG]Head_Calibrator : Moving to park position, joint: 1 
[DEBUG]Head_Calibrator : Moving to park position, joint: 2 
[DEBUG]Head_Calibrator : Moving to park position, joint: 3 
[DEBUG]Head_Calibrator : Moving to park position, joint: 4 
[DEBUG]Head_Calibrator : Moving to park position, joint: 5 
[DEBUG]Head_Calibrator : Park  completed 
[DEBUG]cfw2can [0] msg from board 1: PWM DIS CH1 
[DEBUG]cfw2can [0] msg from board 1: PWM DIS CH0 
[DEBUG]cfw2can [0] msg from board 1: PWM DIS COUPLED:1 
[DEBUG]cfw2can [0] msg from board 1: PWM DIS COUPLED:0 
[DEBUG]cfw2can [0] msg from board 2: PWM DIS:0 
[DEBUG]cfw2can [0] msg from board 2: PWM DIS:1 
[DEBUG]cfw2can [0] msg from board 2: PWM DIS COUPLED:3 
[DEBUG]cfw2can [0] msg from board 2: PWM DIS COUPLED:2 
[DEBUG]head_calibrator finished park device head_mc_wrapper 
[INFO]All actions for action level 1 of interrupt1 phase finished. 
[INFO]interrupt1 phase finished. 
[INFO]shutdown phase starting... 
[INFO]Entering action level 2 of phase shutdown 
[INFO]Stopping icub/cartesianController/left_arm
[INFO]Stopping icub/cartesianController/right_arm
[INFO]All actions for action level 2 of shutdown phase started. Waiting for unfinished actions. 
[INFO]All actions for action level 2 of shutdown phase finished. 
[INFO]Entering action level 5 of phase shutdown 
[INFO]All actions for action level 5 of shutdown phase started. Waiting for unfinished actions. 
[INFO]All actions for action level 5 of shutdown phase finished. 
yarp: Removing input from /icub/yarpmotorgui0/head/rpc:o to /icub/head/rpc:i
yarp: Removing input from /icub/yarpmotorgui0/head/command:o to /icub/head/command:i
yarp: Removing output from /icub/head/state:o to /icub/yarpmotorgui0/head/state:i
yarp: Removing output from /icub/head/stateExt:o to /icub/yarpmotorgui0/head/stateExt:i
yarp: Removing input from /icub/yarpmotorgui0/torso/rpc:o to /icub/torso/rpc:i
yarp: Removing input from /icub/yarpmotorgui0/torso/command:o to /icub/torso/command:i
yarp: Removing output from /icub/torso/state:o to /icub/yarpmotorgui0/torso/state:i
yarp: Removing output from /icub/torso/stateExt:o to /icub/yarpmotorgui0/torso/stateExt:i
yarp: Removing input from /icub/yarpmotorgui0/left_arm/rpc:o to /icub/left_arm/rpc:i
yarp: Removing input from /icub/yarpmotorgui0/left_arm/command:o to /icub/left_arm/command:i
yarp: Removing output from /icub/left_arm/state:o to /icub/yarpmotorgui0/left_arm/state:i
yarp: Removing output from /icub/left_arm/stateExt:o to /icub/yarpmotorgui0/left_arm/stateExt:i
yarp: Removing input from /icub/yarpmotorgui0/right_arm/rpc:o to /icub/right_arm/rpc:i
yarp: Removing input from /icub/yarpmotorgui0/right_arm/command:o to /icub/right_arm/command:i
yarp: Removing output from /icub/right_arm/state:o to /icub/yarpmotorgui0/right_arm/state:i
yarp: Removing output from /icub/right_arm/stateExt:o to /icub/yarpmotorgui0/right_arm/stateExt:i
yarp: Removing input from /icub/yarpmotorgui0/left_leg/rpc:o to /icub/left_leg/rpc:i
yarp: Removing input from /icub/yarpmotorgui0/left_leg/command:o to /icub/left_leg/command:i
yarp: Removing output from /icub/left_leg/state:o to /icub/yarpmotorgui0/left_leg/state:i
yarp: Removing output from /icub/left_leg/stateExt:o to /icub/yarpmotorgui0/left_leg/stateExt:i
yarp: Removing input from /icub/yarpmotorgui0/right_leg/rpc:o to /icub/right_leg/rpc:i
yarp: Removing input from /icub/yarpmotorgui0/right_leg/command:o to /icub/right_leg/command:i
yarp: Removing output from /icub/right_leg/state:o to /icub/yarpmotorgui0/right_leg/state:i
yarp: Removing output from /icub/right_leg/stateExt:o to /icub/yarpmotorgui0/right_leg/stateExt:i
[INFO]Closing Server Inertial...
[INFO]Server Intertial thread finished
*** glibc detected *** robotInterface: corrupted double-linked list: 0x00007f1e99ab2b50 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x75be6)[0x7f1ec3bc5be6]
/lib/x86_64-linux-gnu/libc.so.6(+0x7605d)[0x7f1ec3bc605d]
/lib/x86_64-linux-gnu/libc.so.6(+0x77518)[0x7f1ec3bc7518]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x6c)[0x7f1ec3bca98c]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.1(_ZN4yarp2os4impl20PortReaderBufferBaseD1Ev+0x2e)[0x7f1ec4d918be]
/usr/local/src/robot/icub-main/build-pc104/lib/iCub/skinWrapper.so(_ZN4yarp3dev12AnalogServerD1Ev+0x3fc)[0x7f1e8ede4a5c]
/usr/local/src/robot/icub-main/build-pc104/lib/iCub/skinWrapper.so(_ZN4yarp3dev12AnalogServerD0Ev+0x9)[0x7f1e8ede4f49]
/usr/local/src/robot/icub-main/build-pc104/lib/iCub/skinWrapper.so(_ZN11skinWrapper5closeEv+0x22)[0x7f1e8edd4522]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.1(_ZN4yarp3dev10PolyDriver9closeMainEv+0x44)[0x7f1ec494f674]
robotInterface(_ZN14RobotInterface6Device5closeEv+0x20)[0x416a70]
robotInterface(_ZN14RobotInterface5Robot7Private12closeDevicesEv+0x46)[0x41ee76]
robotInterface(_ZN14RobotInterface5Robot10enterPhaseENS_11ActionPhaseE+0x1bee)[0x42505e]
robotInterface(_ZN14RobotInterface6Module5closeEv+0x55)[0x41a7a5]
[... 114 lines cut ...]
7f1ec2f9e000-7f1ec319d000 ---p 00032000 00:17 275999                     /usr/local/src/robot/yarp/build-pc104/lib/libYARP_sig.so.2.3.64.14
7f1ec319d000-7f1ec319e000 rw-p 00031000 00:17 275999                     /usr/local/src/robot/yarp/build-pc104/lib/libYARP_sig.so.2.3.64.14
7f1ec319e000-7f1ec319f000 rw-p 00000000 00:00 0 
[... cut ...]
7f1ec3b4b000-7f1ec3b4c000 rw-p 00017000 07:00 10307                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/libpthread-2.13.so
7f1ec3b4c000-7f1ec3b50000 rw-p 00000000 00:00 0 
7f1ec3b50000-7f1ec3cd1000 r-xp 00000000 07:00 10150                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/libc-2.13.so
7f1ec3cd1000-7f1ec3ed1000 ---p 00181000 07:00 10150                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/libc-2.13.soAborted

~~~

pattacini, 2016-04-29:
@randaz81 @barbalberto @drdanz
Please, apply any possible bug fix to `yarprobotinterface` instead.

pattacini, 2016-04-29:
Issue moved to [robotology/yarp #745](https://github.com/robotology/yarp/issues/745) via [**ZenHub.io**](https://www.zenhub.io/)

closed 2016-04-29 by pattacini