robotology/icub-main #279: problem of `icub_firmware_shared`

gregoire-pointeau, 2016-01-20:
Hello, I a updating my iCub which is mounted on the iKart and I have the following errors:
(ps: I updated and compiled YARP without problem just before.)
It seems to be a problem of shared configuration but I'm not an expert on it. Does anyone has a clue ?

```
  The following configuration files were considered but not accepted:
     /usr/local/src/robot/icub-firmware-shared/build-pc104/icub_firmware_shared-config.cmake, version:
 0.1.0
     /usr/local/src/robot/icub-firmware-shared/build-pc104/icub_firmware_shared-config.cmake, version:
 0.1.0
 Call Stack (most recent call first):
   CMakeLists.txt:84 (include)

 CMake Warning at conf/iCubFindDependencies.cmake:19 (message):
   You requested to use the package icub_firmware_shared, but it is
   unavailable (or was not found).  This might lead to compile errors, we
   recommend you turn off the ICUB_USE_icub_firmware_shared flag.
 Call Stack (most recent call first):
   conf/iCubFindDependencies.cmake:108 (checkandset_dependency)
   CMakeLists.txt:84 (include)

 ERROR canBusMotionControl: cannot find icub_firmware_shared library, turn off device
   embObj library can now be found in the icub-firmware-shared package that
   can be downloaded from here:
        https://github.com/robotology/icub-firmware-shared
 ERROR canBusAnalogSensor: cannot find icub_firmware_shared library, turn off device
   embObj library can now be found in the icub-firmware-shared package that
   can be downloaded from here:
        https://github.com/robotology/icub-firmware-shared

 ERROR canBusDoubleFTSensor: cannot find icub_firmware_shared library, turn off device
   embObj library can now be found in the icub-firmware-shared package that
   can be downloaded from here:
        https://github.com/robotology/icub-firmware-shared

 ERROR canBusVirtualAnalogSensor: cannot find icub_firmware_shared library, turn off device
   embObj library can now be found in the icub-firmware-shared package that
   can be downloaded from here:
        https://github.com/robotology/icub-firmware-shared

 ERROR canBusInertialMTB: cannot find icub_firmware_shared library, turn off device
   embObj library can now be found in the icub-firmware-shared package that
   can be downloaded from here:
        https://github.com/robotology/icub-firmware-shared

 ERROR canBusSkin: cannot find icub_firmware_shared library, turn off device
   embObj library can now be found in the icub-firmware-shared package that
   can be downloaded from here:
        https://github.com/robotology/icub-firmware-shared
```

traversaro, 2016-01-20:
Hi, 
  have you updated `icub-firmware-shared` to the latest version?

gregoire-pointeau, 2016-01-20:
Hum, No I don't think so...
I'm a bit ashamed but, what is that ?
(I'm alone with the robot for 4 months...)

traversaro, 2016-01-20:
For the explanation of what `icub-firmware-shared` is, check https://github.com/robotology/icub-firmware-shared#b2-case-of-use-with-icub-main . 
To see how it is related to the compilation on the robot pc, see http://wiki.icub.org/wiki/Compilation_on_the_pc104#Compile_icub-firmware-shared .

gregoire-pointeau, 2016-01-20:
Ok Thanks.
Now, I have another problem with the `canBusMotionControl` for my `pc104`
```
Building CXX object src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/CanBusMotionControl.cpp.o
/usr/local/src/robot/icub-main/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp: In member function ‘unsigned char yarp::dev::CanBusMotionControl::from_modevocab_to_modeint(int)’:
/usr/local/src/robot/icub-main/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:3901:16: warning: large integer implicitly truncated to unsigned type [-Woverflow]
/usr/local/src/robot/icub-main/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp: In member function ‘virtual bool yarp::dev::CanBusMotionControl::getJointTypeRaw(int, yarp::dev::JointTypeEnum&)’:
/usr/local/src/robot/icub-main/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:4163:16: error: ‘JointTypeEnum’ is not a class or namespace
make[2]: *** [src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/CanBusMotionControl.cpp.o] Error 1
make[1]: *** [src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/all] Error 2
make: *** [all] Error 2
```

And with the simulator for the other blades:
```
Building CXX object src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/wrapper/iCubSimulationControl.cpp.o
/mnt/data/Data/robot/icub/src/simulators/iCubSimulation/wrapper/iCubSimulationControl.cpp: In member function ‘virtual bool yarp::dev::iCubSimulationControl::getJointTypeRaw(int, yarp::dev::JointTypeEnum&)’:
/mnt/data/Data/robot/icub/src/simulators/iCubSimulation/wrapper/iCubSimulationControl.cpp:1781:27: error: ‘yarp::dev::JointTypeEnum’ is not a class or namespace
         type = yarp::dev::JointTypeEnum::VOCAB_JOINTTYPE_REVOLUTE;
                           ^
src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/build.make:238: recipe for target 'src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/wrapper/iCubSimulationControl.cpp.o' failed
make[2]: *** [src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/wrapper/iCubSimulationControl.cpp.o] Error 1
CMakeFiles/Makefile2:3527: recipe for target 'src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/all' failed
make[1]: *** [src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2
```

But this seems to be related to so stuff directly in the ode of `icub-main` no ?

traversaro, 2016-01-20:
The fix for that is arriving in a minute. : )

gregoire-pointeau, 2016-01-20:
Ok, cool.
Thank you very much !

closed 2016-01-20 by gregoire-pointeau

traversaro, 2016-01-20:
It should be fixed by https://github.com/robotology/icub-main/commit/1161b7e1f75add9b10add9facdc96c7be6f31c24 .