robotology/icub-main #282: ***ERROR*** driver <gazecontrollerclient> was found but could not open Segmentation fault (core dumped)

wainting, 2016-01-26:
Hi to all,

I'm tryng to Express the fixation point of icubSim in Cartesian Coordinates using this code 
```
 #include <yarp/os/Network.h>
 #include <yarp/os/RFModule.h>
 #include <yarp/os/RateThread.h>
 #include <yarp/os/Time.h>
 #include <yarp/sig/Vector.h>
 #include <yarp/math/Math.h>
 #include <yarp/dev/Drivers.h>
 #include <yarp/dev/ControlBoardInterfaces.h>
 #include <yarp/dev/GazeControl.h>
 #include <yarp/dev/PolyDriver.h>
 #include <gsl/gsl_math.h>
 #include <stdio.h>
 #include <deque>
#include <cstdio>
#include <yarp/os/Time.h>
 
 #define CTRL_THREAD_PER     0.02        // [s]
 #define PRINT_STATUS_PER    1.0         // [s]
 #define STORE_POI_PER       3.0         // [s]
 #define SWITCH_STATE_PER    10.0        // [s]
 #define STILL_STATE_TIME    5.0         // [s]
 #define STATE_TRACK         0
 #define STATE_RECALL        1
 #define STATE_WAIT          2
 #define STATE_STILL         3
 
 using namespace std;
 using namespace yarp::os;
 using namespace yarp::dev;
 using namespace yarp::sig;
 using namespace yarp::math;

int main()
{
Property option;
option.put("device","gazecontrollerclient");
option.put("remote","/iKinGazeCtrl");
option.put("local","/client/gaze");
 
PolyDriver clientGazeCtrl(option);
IGazeControl *igaze=NULL;

if (clientGazeCtrl.isValid()) {
   clientGazeCtrl.view(igaze);
Vector fp(3);
fp[0]=-0.50;                                    // x-component [m]
fp[1]=+0.00;                                    // y-component [m]
fp[2]=+0.35;                                    // z-component [m]
igaze->lookAtFixationPoint(fp);                 // move the gaze to the desired fixation point
igaze->waitMotionDone();                        // wait until the operation is done
Vector x;
igaze->getFixationPoint(x);                     // retrieve the current fixation point
cout<<"final error = "<<norm(fp-x)<<endl;       // return a measure of the displacement error

}

}
```
first of all I run ``` iKinGazeCtrl --from configSim.ini --name icubSim  ``` this is the output:
```
||| clearing context
||| adding context [iKinGazeCtrl]
||| configuring
||| no policy found
||| default config file specified as configSim.ini
||| checking [/home/daniele/configSim.ini] (pwd)
[... 88 lines cut ...]
[INFO]Localizer started successfully
yarp: Port /icubSim/rpc active at tcp://192.168.0.22:10071
``` 
after this I run the code with the fixation coordinates but i get this error:
```
daniele@daniele-HP-Pavilion-dv6-Notebook-PC:~/Documents/Prova/build$ ./Fixation
yarp: Port /client/gaze/xd:o active at tcp://192.168.0.22:10072
yarp: Port /client/gaze/angles:o active at tcp://192.168.0.22:10073
yarp: Port /client/gaze/mono:o active at tcp://192.168.0.22:10074
yarp: Port /client/gaze/stereo:o active at tcp://192.168.0.22:10075
yarp: Port /client/gaze/x:i active at tcp://192.168.0.22:10076
yarp: Port /client/gaze/angles:i active at tcp://192.168.0.22:10077
yarp: Port /client/gaze/q:i active at tcp://192.168.0.22:10078
yarp: Port /client/gaze/events:i active at tcp://192.168.0.22:10079
yarp: Port /client/gaze/rpc active at tcp://192.168.0.22:10080
[ERROR]unable to connect to the server rpc port!
yarpdev: ***ERROR*** driver <gazecontrollerclient> was found but could not open
Segmentation fault (core dumped)
``` 

Can someone  help me? Thanks.

Best 

Daniele

pattacini, 2016-01-26:
Hi @wainting 

Try to launch just: `iKinGazeCtrl --from configSim.ini`

The option `name` serves a different goal (look it up [**here**](http://wiki.icub.org/iCub/main/dox/html/group__iKinGazeCtrl.html)) and putting it equal to the robot name might cause ports names clash. If so, this might explain the problem you're having.

pattacini, 2016-01-26:
Uhm, maybe not, 'cause you're opening the local ports with the `/client/gaze` prefix. However, I think there might be still some ports conflicts. Did you try to start from a neat situation, meaning that no other ports in the name server?

Also you could query directly the gaze controller by doing:

```sh
>>yarp rpc /iKinGazeCtrl/rpc
status
```

wainting, 2016-01-26:
OK. This time I run ``` iKinGazeCtrl --from configSim.ini ``` and after the code for the fixation :
``` 
daniele@daniele-HP-Pavilion-dv6-Notebook-PC:~/Documents/Prova/build$ ./Fixation
yarp: Port /client/gaze/xd:o active at tcp://10.40.230.217:10072
yarp: Port /client/gaze/angles:o active at tcp://10.40.230.217:10073
yarp: Port /client/gaze/mono:o active at tcp://10.40.230.217:10074
yarp: Port /client/gaze/stereo:o active at tcp://10.40.230.217:10075
yarp: Port /client/gaze/x:i active at tcp://10.40.230.217:10076
yarp: Port /client/gaze/angles:i active at tcp://10.40.230.217:10077
yarp: Port /client/gaze/q:i active at tcp://10.40.230.217:10078
yarp: Port /client/gaze/events:i active at tcp://10.40.230.217:10079
yarp: Port /client/gaze/rpc active at tcp://10.40.230.217:10080
yarp: Sending output from /client/gaze/rpc to /iKinGazeCtrl/rpc using tcp
yarp: Sending output from /client/gaze/xd:o to /iKinGazeCtrl/xd:i using udp
yarp: Sending output from /client/gaze/angles:o to /iKinGazeCtrl/angles:i using udp
yarp: Sending output from /client/gaze/mono:o to /iKinGazeCtrl/mono:i using udp
yarp: Sending output from /client/gaze/stereo:o to /iKinGazeCtrl/stereo:i using udp
yarp: Receiving input from /iKinGazeCtrl/x:o to /client/gaze/x:i using udp
yarp: Receiving input from /iKinGazeCtrl/angles:o to /client/gaze/angles:i using udp
yarp: Receiving input from /iKinGazeCtrl/q:o to /client/gaze/q:i using udp
yarp: Receiving input from /iKinGazeCtrl/events:o to /client/gaze/events:i using udp
[INFO]created wrapper <gazecontrollerclient>. See C++ class ClientGazeController for documentation.
 ```
The problem is that the icubSim loses control of the head joints and starts to look around randomly.. maybe there is something wrong in the fixation code.
The output of ```   >>yarp rpc /iKinGazeCtrl/rpc
                            status 
                             Response: [ack] running   ```

pattacini, 2016-01-26:
In line with the previous standalone test, we could still continue doing (don't launch your module):

```sh
>> yarp write ... /iKinGazeCtrl/xd:i
-0.5 0.0 0.35
```

Are the resultant movements still crazy?
If so, try again with `--imu::mode off` command line option.

wainting, 2016-01-27:
With the values you give me the system works... Now I put a box in the icub world in 0.0 0.92 0.35
[image]

after I write  
       ```
      >> yarp write ... /iKinGazeCtrl/xd:i
        0.0 0.92 0.35
             ```
and the simulator crash.... is that happen because the encoders go out of range? The coordinates I've passed on ```yarp write``` are not world coordinates? Why can't foveate the box?

pattacini, 2016-01-27:
For the crash of the simulator I don't have any explanation, but remember that the world coordinate system is different from that of the robot. Find out more on our [**wiki**](http://wiki.icub.org/wiki/Simulator_README#Object_reference_frame:).

wainting, 2016-01-29:
OK!  The problem was that I wrong the reference frame system... with the coordinates  -10.0 0.0 1.35 the code works.. Can I ask you before closing this issue why the rototraslational matrix for switch between robot system coordinates and and world coordinates it's a 4x4 (in the readme simulator)?  I would expect for an homogeneus trasformation a 3x4 matrix to obtain 3 values points... how can I use the matrix in the readme simulator for convert from  -0.5 0.0 0.35 (robot reference) to world reference frame?

pattacini, 2016-01-29:
Any homogeneous transformation representing a rotation along with a translation is always described by means of a 4x4 matrix belonging to the **SE3** group.

In a few words, you end up having a 4x1 vector representing a point in the space, whose first three numbers account for the x,y,z coordinates, whereas the last component is always 1.

wainting, 2016-01-29:
OK. Thanks  Ugo!

closed 2016-01-29 by wainting