robotology/icub-main #285: Update head IMU location for iCub Head v2 COM Express

traversaro, 2016-01-29:
The new iCub v2 head with the new eletronics based on COM Express will have a different IMU. The sensor frame of this new IMU will be translated with respect to the frame of the old IMU: 
[image]

We then need to: 
* Update the documentation in http://wiki.icub.org/wiki/ICubForwardKinematics 
* Add a new variant to [iCubInertialSensor](http://wiki.icub.org/brain/classiCub_1_1iKin_1_1iCubInertialSensor.html#a0d9da729f9e983a168a1bfaf72b083a6)
* Modify the iKinGazeCtrl to use the correct IMU chain (even if the angular velocity model should not be affected).
* Modify the URDF model pipeline generator. 

cc @pattacini @matejhof @Mick3Lozzo

barbalberto, 2016-01-29:
Is the IMU a new device? Do we have the yarp driver for it?

randaz81, 2016-01-29:
* [ ] modify legacy SW (iDyn/wholeBodyDynamics/gravitycompensator) adding a new option --head_V2.5
* [ ] put in this page and in the official icub manual a link to the sensor datasheet.
* [ ] check the yarp devicedriver protocol

traversaro, 2016-11-16:
The first robot with this head is now in the lab (`iCubGenova04`). The offset is just in position so it is not a big issue, but it should be solved.

pattacini, 2016-11-16:
@traversaro, wilco soon

I'm also thinking of refactoring a bit the way version is specified, for `iKinGazeCtrl` at least, moving from 
```
headV2
```
to something like
```
hw-version 2.0
```

traversaro, 2016-11-16:
The tricky part is modify the dh chain in http://wiki.icub.org/wiki/ICubInertiaSensorKinematicsV2 , but the automatic converter can help on this.

pattacini, 2016-11-16:
Wouldn't we get away with less by resorting to `HN`? Is there any hitch on this?

pattacini, 2016-11-26:
Working on this next **IFI**

pattacini, 2016-12-01:
I'm done on my side

traversaro, 2017-01-27:
@Mick3Lozzo told me f2f yesterday evening that the head recently mounted on `iCubGenova02` (the one with "python" cameras and the talking mouth) has a different IMU orientation w.r.t. to both the head v2.5 described and the previous version of the head. 
I am waiting more details from him, but it seems that in this case the IMU is mounted upside-down w.r.t. to previous heads, and consequently `gravityCompensator` and `wholeBodyDynamics` will not work on `iCubGenova02` until they are properly modified.

pattacini, 2017-01-27:
I tend to take this new info as a bad **joke**, unless there are very very very good reasons to create yet another version.

Mick3Lozzo, 2017-01-27:
It was due to simplify the assemblying and substitution of MC4-plus rack with an hinge system, but this solution has been abandoned because it's impractical (too many ca
[image]
bles and connectors).
The IMU has been repositioned in the previous position.

traversaro, 2018-09-12:
I guess we can move this issue in icub-model-generator,  in https://github.com/robotology/icub-model-generator/issues/110 and close this issue. Feel free to reopen if necessary.

closed 2018-09-12 by traversaro

traversaro, 2020-11-12:
For reference, this head is part of what is now called iCub 2.6 .