robotology/icub-main #287: YARP_DATA_DIRS missing path?

davidetome, 2016-02-05:
demoRedBall can't find config.ini unless I add /usr/local/share/ICUBcontrib to YARP_DATA_DIRS env variable in .bashrc_iCub file... is that right?

pattacini, 2016-02-05:
Uhm ... judging from the info you gave, it seems that `demoRedBall` has been installed in the user path (i.e. `/user/local`), so it should suffice to rely on `path.d`: therefore, no need for using `YARP_DATA_DIRS` at all.

Nonetheless, a mixed configuration of `path.d` + `YARP_DATA_DIRS` brings to unexpected results, as far as I know.

/cc @traversaro

pattacini, 2016-02-05:
(`icub-main` is not the correct repo for this issue :smirk:)

randaz81, 2016-02-05:
http://wiki.icub.org/wiki/YARP_ResourceFinder

> Nonetheless, a mixed configuration of path.d + YARP_DATA_DIRS brings to unexpected results, as far as I know.
Let's check then Davide's system: I think what he did (installing in /usr/local) was valid a choice, so why path.d didn't find the files? why he was required to add the folder to YARP_DATA_DIRS?

pattacini, 2016-02-05:
@randaz81 I agree

@davidetome see also https://github.com/robotology/QA/issues/42 for future reference.

pattacini, 2016-02-05:
@drdanz might be of some help too.

traversaro, 2016-02-05:
You hit a (at least for me) rather counterintuitive aspect of the YARP configuration files policy. See one of the last lines in http://www.yarp.it/yarp_external_build.html :

>Unless the NO_PATH_D option is passed, this macro checks if the installation directory of the package is the same as YARP's, in which case it sets up automatic recognition of data directories; otherwise, it warns the user to set up appropriately the YARP_DATA_DIRS environment variable.

 Basically if the 3rd party repository (in this case `icub-contrib`) is not being installed in the same installation prefix as YARP the `path.d` file **is not installed** and the user is warned when configuring `icub-contrib` with the following message:

> ${_msg}: no file will we be installed into path.d folder, you need to set YARP_DATA_DIRS environment variable to ${_data_dirs}${_path_separator}${_path}

The relevant code is in https://github.com/robotology/yarp/blob/83466b733d89d2f3687542b047bcd5185d3f9299/conf/YarpInstallationHelpers.cmake#L211 . 

In this case I imagine that yarp was used in the build directory, while icub-contrib was installed in `/usr/local/`.

pattacini, 2016-02-05:
@traversaro thanks for the insight.
Yep, I believe `yarp` is used from within the build.

lornat75, 2016-02-10:
It seems to me that the use of YARP_DATA_DIRS here is required, therefore the reply to @davidetome should be "yes". 

Are we missing some documentation info?

pattacini, 2016-02-10:
I believed that configuration files in `/usr/local` (i.e. standard location) are found by means of `path.d`; on the other hand whenever we install in different paths we should rely on `YARP_DATA_DIRS`, to tell yarp where to seek.

lornat75, 2016-02-10:
I think the problem is that yarp in this case has not been installed, therefore there is no path.d.

pattacini, 2016-02-10:
Got it, I missed the link between the _yarp-in-the-build_ and `path.d`.

drdanz, 2016-02-11:
Maybe this point is not very clear yet... `path.d` is a very specific use case and should work only if yarp and others are
1. **installed** (not from the build directory)
2. installed **using the SAME `CMAKE_INSTALL_PREFIX`**
3. **`YARP_DATA_DIRS` is NOT set**.

For all the other use cases, you should be using `YARP_DATA_DIRS`

I don't know how this is implemented, and I suspect that there is some issue, but according to the original specifications, if `YARP_DATA_DIRS` is set, `path.d` should just be disregarded.

pattacini, 2016-02-11:
Yes, for me it was not that clear.
Thanks @drdanz for the clarification!

closed 2016-02-12 by pattacini