robotology/icub-main #288: wholeBodyDinamics and robot calibration

davidetome, 2016-02-05:
it seems that wholeBodyDinamics doesn't wait for robot calibration completion to start. If I run the module before calibration is completed, all torque values are 0 (verified by Randaz)

pattacini, 2016-11-03:
Hi @davidetome 

If you use the proper [**xml script**](https://github.com/robotology/icub-main/blob/master/app/iCubStartup/scripts/iCubStartup.xml.template) to launch `yarprobotinterface`  alongside with `wholeBodyDynamics` you shouldn't experience that problem anymore. This is the correct way of launching these components indeed.

In particular, `yarpmanager` will check whether it's the right moment to start `wholeBodyDynamics` using the `dependencies` made explicit in the following lines:

```xml
<module>
    <name>wholeBodyDynamics</name>
    <!-- Remember to add the "--headV2" parameter for robots equipped with head v. 2.0 -->
    <parameters>--autoconnect</parameters>
    <node>pwrNode6</node>
    <dependencies>
        <port timeout="60.0" request="is_ready" reply="[ok]">/icub/yarprobotinterface</port>
    </dependencies>
</module>
```

pattacini, 2016-11-03:
Let's close this until we'll have news.

closed 2016-11-03 by pattacini