robotology/icub-main #291: yarpdev: ***ERROR*** could not find device <cartesiancontrollerclient>

ceroloy, 2016-02-07:
I am using:
- iCub version 1.1.13 and Yarp 2.3.22. 
- I compiled with icubmode_cartesiancontrolclient/cartesiancontrolserver as ON. 
- Those are also listed in my icubmoddev --list
- The error I am receiving is: yarpdev:  *** ERROR *** could not find device <cartesiancontrollerclient> 

I have tried:
- gaze interface client, which also **fails**. 
- remote_controlboard works **fine**.
- I can work with simulation when I directly connect to the right-arm from terminal. So, simulation
and related yarp ports seems to work **fine**.
- yarpserver & iCub_SIM work **fine**.
- simCartesianControl & iKin also work **fine**.
- After executing the code I check for live ports and **I do not see** ports for "/icub/cartesianController/right_arm" and /client/right_arm".

-So **Is there a driver file missing for cartesiancontrollerclient?**

- The following code fails at clientCartCtrl(option). 

`  
        #include yarp/os/Network.h
        #include yarp/dev/Drivers.h
        #include yarp/dev/CartesianControl.h
        #include yarp/dev/PolyDriver.h
        #include stdio.h

        using namespace yarp::os;
        using namespace yarp::dev;
        using namespace std;

        int main() {
                Property option;
	        option.put("device","cartesiancontrollerclient");
	        option.put("remote","/icub/cartesianController/right_arm");
	        option.put("local","/client/right_arm");
	
                PolyDriver clientCartCtrl(option);//fails here

        }

pattacini, 2016-02-08:
Hi @ceroloy 

Let me suggest you to update your `yarp` and `icub` binaries first.
Your current versions seem to be really old: apparently, they date back to 2 years and half ago.

In the meantime many things have changed and it might be quite difficult for us to provide useful help.

By the way, since you're playing with the simulator - I suppose -, the ports stem name must begin with `/icubSim` instead of `/icub`. This does not explain the reason of the error in the original post, but certainly will help out later on.

ceroloy, 2016-02-08:
Hello sir,
Thank you for your response. The reason I am using old binaries is that the
robotic platform we own only works with those binaries. In my case how
should I proceed ? Should I create custom drivers or try to understand the
updates made on the binaries throughout the years and try to hack into it?

ceroloy, 2016-02-08:
Ok here is the solution,
Our CMakeFiles were generated by yarp cmake and we needed to include 
"${ICUB_LIBRARIES} icubmod" in the
target_link_libraries(${KEYWORD} ${YARP_LIBRARIES} ${ICUB_LIBRARIES} icubmod).

closed 2016-02-08 by ceroloy

pattacini, 2016-02-08:
Happy yo know you solved the problem.
Cheers