robotology/icub-main #295: portaudio packages not installed in the icub live pc104 image

traversaro, 2016-02-11:
@mbrunettini :
This morning on pc104 of the iCubGenova02, @DanielePucci was following the instructions at http://wiki.icub.org/wiki/Compilation_on_the_pc104 to compile yarp on the pc104, but we got an error because yarp could not find portaudio. 
We installed the necessary packets with `sudo apt-get libportaudio0 libportaudio2 portaudio19-dev` and now yarp is compiling fine.
As the portaudio device is mentioned in the official pc104 compilation instructions, we need to:
* Either mention that we need to install those packages in the documentation.
* Install those packages directly in the pc104 OS image. 
Using the default packages could actually cause problems, see https://github.com/robotology/icub-main/issues/265 .

mbrunettini, 2016-02-11:
What about adding those packages to the icub-common dependancies (as many others)?
@lornat75

traversaro, 2016-02-11:
Option added to the bullet list.

barbalberto, 2016-02-11:
Originally icub-common was meant to be used on laptop computers and pc104 was not using it.

mbrunettini, 2016-02-11:
@barbalberto : you're right.
@traversaro : the package libportaudio-dev can be enough? The package libportaudio2 is already installed on PC1104 image. I'm trying to keep the image as thin as possible. Can we test it?

traversaro, 2016-02-11:
I tested on the pc104 today first with `libportaudio-dev` and it was not enough to compile the `portaudio` driver in yarp.

mbrunettini, 2016-02-11:
Added the required packages to the pc104 image. Available in next build.
Issue closed.

closed 2016-02-11 by mbrunettini

mbrunettini, 2016-02-12:
A new live image with required package has been uploaded to icub.org website