robotology/icub-main #298: iCubSim should be deprecated

randaz81, 2016-02-12:
This issue wants to promote the development of the missing features in https://github.com/robotology/gazebo-yarp-plugins in order to have a complete icub simulation with gazebo.

Let's stop any further development on iCubSIM, it's not possibile to maintain two simulators.
With your (reader) contribution we can easily implement the missing features. Help us!

@traversaro is tracking the current status of development of the simulator with:
https://github.com/robotology/gazebo-yarp-plugins/issues/132.

randaz81, 2016-06-11:
In the next weeks, I'm planning to move iCubSim away from icub-main to a new icubSim repository.

pattacini, 2016-06-11:
I'd suggest to move `iCubSim` to https://github.com/orgs/robotology-legacy/dashboard :cry: 

@randaz81 tell me when you're done, I'll have to update a bunch of documentation then

pattacini, 2016-09-02:
It appears that the new simulator in `Gazebo` is not properly tuned and **we might have quite hard time to work** with it.

See discussion in https://github.com/robotology/QA/issues/131 and in particular the [review](https://gist.github.com/misaki43/5014f5076118ab134cc8c9ad89c3ea17#gistcomment-1856736) of this [gist](https://gist.github.com/misaki43/5014f5076118ab134cc8c9ad89c3ea17).

@randaz81 I'd definitely wait a bit longer before deprecating `icubSim`.

randaz81, 2016-11-03:
@pattacini ???

closed 2017-03-28 by pattacini