robotology/icub-main #302: Python bindings not working

Tobias-Fischer, 2016-02-23:
Hi all,

unfortunately the python bindings for `icub-main` do not work at all.
After building the bindings (successfully), the following error occurs:
```
In [1]: import icub
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-434db9cd35c6> in <module>()
----> 1 import icub

/home/tobias/robot/icub-main/bindings/build_python/icub.py in <module>()
     26                 fp.close()
     27             return _mod
---> 28     _icub = swig_import_helper()
     29     del swig_import_helper
     30 else:

/home/tobias/robot/icub-main/bindings/build_python/icub.py in swig_import_helper()
     22         if fp is not None:
     23             try:
---> 24                 _mod = imp.load_module('_icub', fp, pathname, description)
     25             finally:
     26                 fp.close()

ImportError: ./_icub.so: undefined symbol: _ZN20ClientGazeControllerC1ERN4yarp2os10SearchableE
```

Furthermore, a look in `icub.i` reveals that only very little functionality is exposed in the swift interface (gaze controller and cartesian controller). I guess it would be great if more libraries are accessible from other programming languages.

I am happy to help with this, but unfortunately I couldn't resolve the errors listed above. As soon as they are fixed, I'll go ahead and integrate more functionality.

Best,
Tobi

frank-foerster, 2016-11-22:
Hi @Tobias-Fischer ,
I just tried using Python in conjunction with the iCub repository for the first time, and came across exactly the same problem. Did you find any solution to this?

Tobias-Fischer, 2016-11-22:
Hi @frank-foerster,
Nope, unfortunately I didn't have time to look into this. I am not sure who the best person would be to talk to. Maybe @pattacini can give us a hint?

Best,
Tobi

traversaro, 2016-11-22:
I am in the process of enabling the build of the Python bindings in [`codyco-superbuild`](https://github.com/robotology/codyco-superbuild) for each sub-project that support them, including `icub`, so eventually (no short-term guarantees  : ) ) I could look into this. 

However, if you are just interested in using the `CartesianController` and the `GazeController`, I think you can avoid to instantiate directly the `CartesianController` class by spawning it at runtime using the `yarp::dev::PolyDriver` class factory. 

The relevant `view` method is present directly in the yarp bindings : https://github.com/robotology/yarp/blob/e0ca8e6f73c37286734d06e3afb8015529ee36b9/bindings/yarp.i#L739 , so if the `yarp` bindings are correctly configured you should be able to use the Cartesian Controller . 

A quick search on github for that method ( https://github.com/search?p=1&q=viewICartesianControl&type=Code&utf8=%E2%9C%93 ) found two possible examples, one in Python (not actually using the `cartesiancontrollerclient` device, but the idea is similar) and the other in Matlab:
* https://github.com/bharat72007/wallyR/blob/master/roboticslab-code/asibot/contrib/superseded/test_asibot.py#L31
* https://github.com/GiuliaP/icubBMM15/blob/941f7d41da52503cbd0c66601379963e8d645863/matlab/cartesian_interface_template/handController.m#L69

I think they should be correct but again remember that this are just random piece of code I found on the internet. : )

drdanz, 2016-11-22:
This error is something related to wrong linkage of the libraries:

```
ImportError: ./_icub.so: undefined symbol: _ZN20ClientGazeControllerC1ERN4yarp2os10SearchableE
```

Have you tried with a clean build?

frank-foerster, 2016-11-22:
I'm not experienced with SWIG, so I'm not sure how it cobbles together the various required libraries and the paths to them.

It was my first build of the bindings, that is, in a newly created build dir in the `$ICUB_ROOT/bindings/build` with `build` a newly created subdir, as recommended in the instructions. Also the fact that I got precisely the same error as Tobias Fischer above indicates that it's not some leftover artifact of a prior build which is causing this. 

I agree, that somehow the executable can't find the required symbol, meaning some path is probably generated wrongly by SWIG and/or cmake or something of that kind. Maybe some undocumented environment variable is not set?

traversaro, 2016-11-22:
I checked the CMake for the bindings, and I noticed in horror that is using the `link_libraries` CMake command (https://github.com/robotology/icub-main/blob/master/bindings/CMakeLists.txt#L17). 

Perhaps it is worth try to link the bindings to the iCub libraries using a proper `target_link_libraries`, as we usually do and in the way that we know that works.

frank-foerster, 2016-11-23:
So, I've had a further look into this issue and tried several things.
First of all I followed traversaro's recommendation above to look for replacement for the `link_libraries`
command. Replacing it with `target_link_libraries` won't as SWIG uses it special CMake macros which don't seem to be compatible with the the latter command. 
Instead I replaced it with a `swig_link_libraries` command as indicated [here](https://cmake.org/cmake/help/v3.2/module/UseSWIG.html) and in terms of building that seems to work (that is: it doesn't throw any more warnings than the CMakeLists.txt with the link_libraries command and the build process completes).
This doesn't solve the problem though, the same runtime-error ensues. 
I subsequently used `nm` to look into the generated `_icub.so` and the symbol `_ZN20ClientGazeControllerC1ERN4yarp2os10SearchableE` is listed as undefined, which means that it is expected to be defined in some other library.  
This triggered me to run `nm` in "batch-mode" over all icub-related libraries in order to find the library which defines it. The relevant library, as the symbol name already hints at, is `$ICUB_DIR/lib/iCub/gazecontrollerclient.so`, where `$ICUB_DIR' is the installation directory of icub-main.
The latter, together with `cartesiancontrollerserver.so` are the only ones located in this `iCub' subdir of the library directory, all the other are non-nested so to speak.
If we print the content of `${ICUB_LIBRARIES}` against which our `_icub.so`is linked against  (`$ICUB_ROOT/bindings/CMakeLists.txt`) `gazecontrollerclient.so` is suspiciously absent. 
So I guess this is the problem - we actually never provide the linker with the correct library. 
As I haven't been around for a while, I'm not sure if `gazecontrollerclient.so`is ought to be in the list of `${ICUB_LIBRARIES}` or not. I'll try now to either add it to this list or look whether it is already member of some other cmake variable that I can use for linking and will let you know whether this really solves the problem.

Tobias-Fischer, 2016-11-23:
This sounds like a cmake issue. Maybe @drdanz can help? In `iCubHelpers.cmake` the `ICUB_TARGETS` variable (which in turn is used to determine `ICUB_LIBRARIES` is only extended for `icub_export_library`, but not in `icub_export_plugin` which is used for `gazecontrollerclient`.

frank-foerster, 2016-11-23:
I found a temporary fix in order to get rid of the problem. Bit I guess it would be nicer to have a more principled solution in terms of including the two libraries `gazecontrollerclient` and `cartesiancontrollerclient` to the list of to-be-linked libraries.

Here's the fix: 
The above mentioned two libraries need to be linked against the to-be-produced SWIG library `_icub.so`.
Both of them are located in `$ICUB_DIR/lib/iCub`, where `$ICUB_DIR` points to the local installation directory of `icub-main`. 
In order to add them to the list of to-be-linked libraries and for the linker to be happy I did the following:

1. In the bindings' [CMakeLists.txt](https://github.com/robotology/icub-main/blob/master/bindings/CMakeLists.txt), line 44 I added the two lines:

`find_library(GAZECTRLCLIENT gazecontrollerclient "$ENV{ICUB_DIR}/lib/iCub")`
`find_library(CARTESIANCTRLCLIENT cartesiancontrollerclient "$ENV{ICUB_DIR}/lib/iCub")`

in order to get the paths to the missing libraries. As mentioned, this is a hack as I'm half-hardcoding the directory. I guess there is some more elegant way using other icub-related CMake variables. As I couldn't find any during my short search I fell back to the use of the environment variable `$ICUB_DIR`. 

2. In the same  [CMakeLists.txt](https://github.com/robotology/icub-main/blob/master/bindings/CMakeLists.txt) add underneath line 62 (`swig_add_module(..)`) the following line:

 `swig_link_libraries(icub ${ICUB_LIBRARIES} ${PYTHON_LIBRARIES} ${GAZECTRLCLIENT} ${CARTESIANCTRLCLIENT})`

3. The above should in theory be sufficient, but for some reason the linker does not like libraries which are missing the `lib` prefix. So I created copies of both `gazecontrollerclient.so` and  `cartesiancontollerclient.so` in the directory where they reside, renaming them to `libgazecontrollerclient.so` and `libcartesiancontrollerclient.so`. That makes the linker happy. 

Now all the errors originating from the generated `icub.py' are gone. Instead we get an error of the test file `test.py`which is mentioned in the [README file](https://github.com/robotology/icub-main/blob/master/bindings/README.txt), because `d.toString()` apparently has no member `c_str()`. Luckily `toString()` on its own appears to be sufficient and prints a list of all kinds of YARP and ICUB devices including the gazecontrollerclient.

traversaro, 2016-11-23:
Great catch @frank-foerster ! 
I think I can explain why I think not having `gazecontrollerclient.so` in `${ICUB_LIBRARIES}`. 

The main point is that  `gazecontrollerclient.so` is a YARP Device, and is meant to be used through the `yarp::dev::PolyDriver` factory, not by linking directly to it. 
This can be seen in the C++ tutorials in http://wiki.icub.org/brain/icub_cartesian_interface.html , section " Opening and Closing the Cartesian Interface" or in http://wiki.icub.org/brain/icub_gaze_interface.html, section: Opening and Closing the Gaze Interface . The equivalent procedure for bindings is the one that I explained in https://github.com/robotology/icub-main/issues/302#issuecomment-262232544 . 

This iCub bindings code is probably a leftover of the era in which all the yarp device plugins contained in `icub-main` where actually library statically linked, and hence they were added to `${ICUB_LIBRARIES}`. At the moment I think the proper way of using the Gaze and Cartesian Interface is the one described in https://github.com/robotology/icub-main/issues/302#issuecomment-262232544 . This bindings code should just be deleted, or replaced by bindings that wrap the actual libraries in `icub-main`, such as `iKin`.

frank-foerster, 2016-11-23:
I have used the cartesian interface as described in the tutorial a few years ago. Currently I'm more interested in using ARE, in conjunction with the cartesian interface, via a Python script. 
I'm not totally sure how to go about this, but it doesn't look too hard to write a SWIG interface for a given module / library. To that purpose I probably can't exclusively rely on the YARP interface and will have to go via icub-main, right?

traversaro, 2016-11-23:
> Currently I'm more interested in using ARE, in conjunction with the cartesian interface, via a Python script. 
> ...
> To that purpose I probably can't exclusively rely on the YARP interface and will have to go via icub-main, right?

I am not familiar with ARE, but if you just want to write a Bottle to a port, as described in http://wiki.icub.org/brain/group__actionsRenderingEngine.html , section "Issuing command", it should be doable with just the **YARP** bindings, without requiring anything icub-main specific.

closed 2017-03-28 by pattacini