robotology/icub-main #318: [iKin] Arms joint limits - wrist pronosupination

vicentepedro, 2016-04-13:
Hi all!

I was digging inside the iKin library and I found something that seems to be a bug...

Inside the file iKinfwd.cpp - Lines 1756-58
https://github.com/robotology/icub-main/blob/master/src/libraries/iKin/src/iKinFwd.cpp#L1756

The limits of the wrist pronosupination of the right arm are [-50,50].

According to the hardware XML of the our iCub (with V1 arms), the limits are [-90,90]. And according with the issue:
https://github.com/robotology/icub-support/issues/213

for the v2 arms are [-60,60].

@pattacini

pattacini, 2016-04-13:
Hi @vicentepedro 

Those hard-coded limits are actually updated at run time by means of [**`iCub::iKin::iKinLimb::alignJointsBounds`**](http://wiki.icub.org/iCub/main/dox/html/classiCub_1_1iKin_1_1iKinLimb.html#ad1f5813de8c56fce20d117bf268eb8ca) method, which is specialized within each single limb object.

There's no point in keeping those bounds always aligned with ever-changing CAD values. The important numbers are indeed written in the robot configuration files and then - as said - retrieved at run time by `iKin` related modules.

closed 2016-04-13 by pattacini

vicentepedro, 2016-04-13:
Ok.
I was not aware of this ;)

I'm using a custom eye-arm kinematic chain. So I will need to re-implement the alignJointsBounds method as well.

Thanks for the help.