robotology/icub-main #319: Information on embObjVirtualAnalogSensor device

traversaro, 2016-04-14:
A simple question, mainly for @barbalberto .

As far I can see, now the logic to send the torque estimates to the low level boards for ETH robots is handled by the `embObjMotionControl` device (see [1] for an example). This situation is different from the one on CAN robots, in which a specific device `CanBusVirtualAnalogSensor` is used for this (see [2][3] for an example). 

I guess that in the past the `embObjVirtualAnalogSensor` was used, but now it is not used anymore, but it is still present in the repo. The plan for the future is to revive it, or perhaps it could be a good idea to just remove the  `embObjVirtualAnalogSensor` device from the repo, to avoid inconsistency in the `updateMeasure` implementation between the two devices? 

[1] :  https://github.com/robotology/icub-main/blob/bf82fee30e7809343cfed8edc0d3dc8178723318/app/robots/iCubHeidelberg01/wrappers/VFT/right_leg-VFT_wrapper.xml
[2] : https://github.com/robotology/icub-main/blob/bf82fee30e7809343cfed8edc0d3dc8178723318/app/robots/iCubParis01/wrappers/VFT/right_arm_VFT_wrapper.xml
[3] : https://github.com/robotology/icub-main/blob/bf82fee30e7809343cfed8edc0d3dc8178723318/app/robots/iCubParis01/hardware/VFT/right_arm_virtual_strain.xml

barbalberto, 2016-04-14:
The original idea was to have the embObjVirtualAnalogSensor to do its job, like the can one, but at the end of the day was never implemented, therefore never used, if I remember correctly.

@marcoaccame has more updates on this

traversaro, 2016-04-14:
Thanks @barbalberto , I imagined something like this. Let's wait for @marcoaccame if he wants to add something, and then we can close the issue.

marcoaccame, 2016-05-03:
Hi @traversaro, sorry for the late intervention: I have nothing to add to the discussion. 
To tell the truth @barbalberto, I have never looked at this device at all.

closed 2016-05-03 by traversaro

randaz81, 2016-05-03:
> The original idea was to have the embObjVirtualAnalogSensor to do its job, like the can one, but at the end of the day was never implemented, therefore never used, if I remember correctly.

The goal is still the same. 

> The plan for the future is to revive it, or perhaps it could be a good idea to just remove the embObjVirtualAnalogSensor device from the repo, to avoid inconsistency in the updateMeasure implementation between the two devices?

The plan to revive it. I always found updateMeasure() awful.