robotology/icub-main #320: motorPwmLimit not found error

traversaro, 2016-04-14:
In `iCubDarmstatd01` and `iCubGenova02` while starting the `robotInterface` we always get the error (i.e. red in yarplogger):
~~~~
motorPwmLimit not found
~~~~
but then the robot starts fine because there are some default values for `motorPwmLimit` in the `CanBusMotionControl` device:
https://github.com/robotology/icub-main/blob/bf82fee30e7809343cfed8edc0d3dc8178723318/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp#L1603 . 

I saw that other parameters that have working fallback values (such as `jntVelMax`) call the `validate_optional` method, that for a parameter not found prints a warning (i.e. yellow in yarplogger) instead of an error. 

Question (mainly @randaz81 @barbalberto ): what is the right way to get rid of this error?

* Add the motorPwmLimit in the configuration files of the neck controlboard

or 

* Transform the `motorPwmLimit` in a optional configuration value, such as  `jntVelMax`
?

Tobias-Fischer, 2016-04-14:
See also https://github.com/robotology/icub-main/issues/180 ..

valegagge, 2016-05-03:
I added motorPwmLimit for all ethernet robots, but could't modify configuration files of body parts controlled by boards on CAN bus (in this case the head), because when I did this work, firmware of CAN boards did't handle motorPwmLimit yet.

After, (about first days of Febbrary)  the firmware of CAN boards has been updated and all configuration files of CAN robots have been modified accordingly.
Probably,  head' files of  `iCubDarmstatd01` and `iCubGenova02` have been forgotten. :(

So, I  think motorPwmLimit should not be a optional parameter, therefore we need to add the motorPwmLimit in the configuration files of the neck controlboard (using the default present in the source code).

Cheers
Vale

randaz81, 2016-05-03:
MotorPwmLimit is a mandatory parameter.
The default value on a CAN board (either in an CAN robot or a ETH robot) is 1333.
robotInterface is supposed to terminate if MotorPwmLimit is not found however, **for debug purpose only**, robotInterface can currently continue. So, @valegagge you did not forgot to update any file. :)

closed 2016-05-03 by randaz81