robotology/icub-main #323: iCubGui wrongly visualizes specific force vectors

matejhof, 2016-04-21:
I created a module that talks to /iCubGui/forces and sends a skinContactList. All my skinContacts have only a single component - along the normal to the skin. At the right palm, this is exactly opposite the z axis of the link reference frame. However, when sending force of the form (0 0 -1), it gets visualized exactly in the opposite direction than it should be. I found a workaround, when setting (0.00001 0.00001 -1), it gets visualized correctly. So the problem seems to be with the exact direction.
I suspect the problem is around here
https://github.com/robotology/icub-main/blob/master/src/tools/iCubGui/src/bvhnode.h#L76
possibly in the atan.
Maybe @ale-git can have a look?

matejhof, 2016-09-01:
Any updates here @ale-git ?

ale-git, 2016-09-08:
Fixed by bb8ce3d966088274e598ff74361e08d13aaecb33

closed 2016-09-08 by ale-git