robotology/icub-main #324: Can't change light source position in iCub-sim world

wainting, 2016-04-22:
Hi to all, 

I'm trying to change the light source position in the icub simulator world (sun position) one the back of the robot in order to to enlighten directly the screen. I've found a code line in iCub_Sim.cpp ``` drawSkyDome(0,0,0,50,50,50);  ``` but it can only change the dimension of the sky without performing rotation. I've thought alternatively to rotate the icub position in order to take the sun on the back of robot but i can't find a way to to rotate the robot 'entirely' without changing every single part position.

Best Regards

Daniele

vtikha, 2016-09-05:
@wainting, i apologise for the late reply, I just wanted to check if you managed to change the light source.

pattacini, 2016-09-07:
Closing in the meantime.

closed 2016-09-07 by pattacini