robotology/icub-main #326: calibrators and wrappers

randaz81, 2016-04-29:
Check if the following snippet of configuration:

```
        <param name="CALIB_ORDER">(8 9 11 13) (10 12 14 15) </param>                       
        <action phase="startup" level="10" type="calibrate">           
            <param name="target">left_arm_mc_wrapper</param>    
        </action> 
```        

is a correct alternative to the one currently used on the robots:

```
        <param name="CALIB_ORDER">(0 1 3 5) (2 4 6 7) </param>                       
        <action phase="startup" level="10" type="calibrate">           
            <param name="target">left_hand_mc</param>    
        </action>   
``` 

(The calibrator attaches to the **arm wrapper** instead of the **hand motor controller**)

Tobias-Fischer, 2016-08-04:
Did someone try this? @randaz81

Tobias-Fischer, 2016-11-27:
Any updates? @randaz81

randaz81, 2016-11-27:
I think no one tested this on a iCub... even if other kind of robots (e.g. R1) use a similar calibration technique (so it should work, in principle).
Btw, I cannot remember anymore: why this issue is referenced by #334?

Tobias-Fischer, 2016-11-27:
Ahhh cool, it might be good to try on a IFI day. The issue is referenced as it causes the following warning: `WARNING Device left_arm_mc is not implementing a yarp::dev::IRemoteCalibrator, therefore it cannot attach to a Calibrator device`

barbalberto, 2016-11-28:
I don't remember the initial issue, anyway I can confirm that parametric calibrators work on wrappers, not motion control devices.

valegagge, 2017-05-26:
I updated all calibration files of all ethernet robots.

@barbalberto Could you finish this work updating CAN robots also?

closed 2021-03-27 by pattacini