robotology/icub-main #331: Outdated documentation of device to compile on the pc104 for ETH robot

traversaro, 2016-05-04:
Since the merge of https://github.com/robotology/icub-main/pull/327 , to successfully run the `yarprobotinterface` on ETH robots it is necessary to compile also the `embObjStrain`, `embObjMais`. `embObjInertials` devices. Failing to compile this device will result in an error (quite clear, fortunatly) of the yarprobotinterface initialization. 

Consequently it would be nice to announce this change in https://github.com/robotology/QA , update the relevant list of devices necessary to be compiled in http://wiki.icub.org/wiki/Compilation_on_the_pc104 , and update the ChangeLog in http://wiki.icub.org/wiki/ChangeLog (see https://github.com/robotology/icub-main/issues/180#issuecomment-216809096 ).

lornat75, 2016-05-04:
@marcoaccame

marcoaccame, 2016-05-11:
Ok, I have updated documentation and announced the change.

closed 2016-07-11 by marcoaccame