robotology/icub-main #332: iCub_SIM does not start. [Mac]

tkelestemur, 2016-05-04:
Hello, I have installed Yarp, iCub and all the dependencies following from these tutorials:
1. http://www.yarp.it/install_yarp_mac.html
2. http://wiki.icub.org/wiki/Homebrew_Mavericks

when I try to start iCub_SIM, I get this error:

```
Tariks-MacBook-Pro:~ tkelestemur$ iCub_SIM 
[DEBUG]
ODE configuration: ODE ODE_EXT_trimesh ODE_EXT_opcode ODE_OPC_new_collider ODE_EXT_threading ODE_double_precision

[DEBUG]SimConfig::configure
||| clearing context
||| adding context [simConfig]
||| configuring
||| default config file specified as simulator.ini
||| checking [/Users/tkelestemur/simulator.ini] (pwd)
||| checking [/Users/tkelestemur/Library/Application Support/yarp/config/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/Users/tkelestemur/Library/Application Support/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/Library/Preferences/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/Users/tkelestemur/Library/Application Support/yarp/config/contexts/simConfig] (context YARP_CONFIG_HOME)
||| checking [/Users/tkelestemur/Library/Application Support/yarp/contexts/simConfig] (context YARP_DATA_HOME)
||| checking [/Library/Preferences/yarp/contexts/simConfig] (context YARP_CONFIG_DIRS)
||| checking [/etc/yarp/contexts/simConfig] (context YARP_CONFIG_DIRS)
||| checking [/usr/local/share/yarp/contexts/simConfig] (context YARP_DATA_DIRS)
||| checking [/usr/share/yarp/contexts/simConfig] (context YARP_DATA_DIRS)
||| checking [/Users/tkelestemur/Library/Application Support/yarp/config/simulator.ini] (YARP_CONFIG_HOME)
||| checking [/Users/tkelestemur/Library/Application Support/yarp/simulator.ini] (YARP_DATA_HOME)
||| checking [/Library/Preferences/yarp/simulator.ini] (YARP_CONFIG_DIRS)
||| checking [/etc/yarp/simulator.ini] (YARP_CONFIG_DIRS)
||| checking [/usr/local/share/yarp/simulator.ini] (YARP_DATA_DIRS)
||| checking [/usr/share/yarp/simulator.ini] (YARP_DATA_DIRS)
||| did not find simulator.ini
yarp: cannot read from simulator.ini
[INFO]default module name:  /icubSim 
[INFO]default verbosity level:  0 
||| finding file []
||| checking [/Users/tkelestemur/] (pwd)
||| found /Users/tkelestemur/
||| finding file [parts]
||| checking [/Users/tkelestemur/parts] (pwd)
||| checking [/Users/tkelestemur/Library/Application Support/yarp/config/parts] (YARP_CONFIG_HOME)
||| checking [/Users/tkelestemur/Library/Application Support/yarp/parts] (YARP_DATA_HOME)
||| checking [/Library/Preferences/yarp/parts] (YARP_CONFIG_DIRS)
||| checking [/etc/yarp/parts] (YARP_CONFIG_DIRS)
||| checking [/usr/local/share/yarp/parts] (YARP_DATA_DIRS)
||| checking [/usr/share/yarp/parts] (YARP_DATA_DIRS)
||| did not find parts
yarp: cannot read from 
||| finding file [parts]
||| did not find parts
||| finding file [general]
||| checking [/Users/tkelestemur/general] (pwd)
||| checking [/Users/tkelestemur/Library/Application Support/yarp/config/general] (YARP_CONFIG_HOME)
||| checking [/Users/tkelestemur/Library/Application Support/yarp/general] (YARP_DATA_HOME)
||| checking [/Library/Preferences/yarp/general] (YARP_CONFIG_DIRS)
||| checking [/etc/yarp/general] (YARP_CONFIG_DIRS)
||| checking [/usr/local/share/yarp/general] (YARP_DATA_DIRS)
||| checking [/usr/share/yarp/general] (YARP_DATA_DIRS)
||| did not find general
yarp: cannot read from 
[ERROR]

There seems to be a conflict with the  actElevation  configuration file 
[ERROR]Check the iCub_parts_activation.ini under $ICUB_ROOT/app/simConfig  
[ERROR]If it still does not work (update svn) and re-install the app folder

 
[ERROR]The iCub simulator will not start..
```

and this is my .bash_profile file

```
export Qt5_DIR=/usr/local/Cellar/qt5/5.4.2/lib/cmake
export GDK_PIXBUF_MODULEDIR="/usr/local/lib/gdk-pixbuf-2.0/2.10.0/loaders"
export OpenCV_DIR=/Users/tkelestemur/Development/frameworks/opencv-3.1.0/build
export code=/Users/tkelestemur/Development/frameworks
export YARP_ROOT=/Users/tkelestemur/Development/frameworks/yarp
export YARP_DIR=/Users/tkelestemur/Development/frameworks/yarp/build
export PATH=$YARP_DIR/bin:$PATH 
export IPOPT_DWN_DIR=/Users/tkelestemur/Development/frameworks/Ipopt-3.12.1
export IPOPT_DIR=/Users/tkelestemur/Development/frameworks/Ipopt-3.12.1/build
export ICUB_ROOT=/Users/tkelestemur/Development/frameworks/icub-main
export ICUB_DIR=/Users/tkelestemur/Development/frameworks/icub-main/build
export PATH=$ICUB_DIR/bin:$PATH

if [ -f `brew --prefix`/etc/bash_completion ]; then
      . `brew --prefix`/etc/bash_completion
fi
source $YARP_ROOT/scripts/yarp_completion
```

I guess this is a PATH issue but I couldn't figure out what I should do. Any idea?

tkelestemur, 2016-05-05:
I have tried starting 'iCub_SIM' from `/Users/tkelestemur/Development/frameworks/icub-main/build/share/iCub/contexts/simConfig` folder and it didn't give any error however the simulation display didn't show up. I changed timestep to 25 but still it doesn't show up.

pattacini, 2016-05-05:
Hi @tkelestemur 

At first sight it seems a problem due to the fact you didn't define `YARP_DATA_DIRS` in your env.
Find out more on this here: https://github.com/robotology/QA/issues/42.

pattacini, 2016-05-05:
Aside from the context, we've been experiencing troubles running the simulator with `El Capitan`.
OS X people can provide more details than me.

tkelestemur, 2016-05-05:
Hi @pattacini thanks for the reply.

I've defined `YARP_DATA_DIRS` as `/Users/tkelestemur/Development/frameworks/yarp/build/share/yarp`
like stated in the installation tutorial but it didn't solve the problem. Anyway it is not important until the simulator problem is fixed with El Capitan.

pattacini, 2016-05-05:
@tkelestemur 

Actually, `iCub_SIM` is in `icub-main`, therefore you ought to append to `YARP_DATA_DIRS` the path to the `icub-main` contexts too.

@vtikha @francesco-romano @claudiofantacci and other OS-X people
Is there already an issue on `El Capitan` so that we can reference that one and close the game here?

pattacini, 2016-05-09:
Closing since the problem is already described in #222.

closed 2016-05-09 by pattacini