robotology/icub-main #339: Tagging icub-* repositories whenever icub-main is tagged

traversaro, 2016-05-16:
This was something that was already discussed by email and in https://github.com/robotology/icub-firmware-shared/issues/4 , but I think it would be nice to discuss it properly. 

Currently, `yarp` and `icub-main` releases are made in the same time instant, but they use different version numbers for historical reasons. 

However there are some repositories that are tightly coupled to `icub-main` : 
* [`icub-firmare`](https://github.com/robotology/icub-firmware)
* [`icub-firmware-shared`](https://github.com/robotology/icub-firmware-shared)
* [`icub-firmware-build`](https://github.com/robotology/icub-firmware-build)
* [`icub-basis-demos`](https://github.com/robotology/icub-basic-demos)

Less coupled repo as they don't depend software-wise on `icub-main`, but definitely related:
* [`icub-tests`](https://github.com/robotology/icub-tests)
* [`icub-gazebo`](https://github.com/robotology/icub-gazebo)

It would be nice to tag all this repos (at least `icub-firmware-shared`, `icub-firmware-build` and `icub-basic-demos`) whenever there is a release of `icub-main`, so that it is actually possible to rollback a robot to a tagged version. As version number I think that they should all use the relative `icub-main` version, to avoid endless tables of matching version numbers (`icub-firmware-shared` 0.2.0 that requires `icub-main` 1.2.12 etc etc).

traversaro, 2016-05-16:
cc @mbrunettini @drdanz

traversaro, 2016-05-16:
Ref: https://github.com/robotology/QA/issues/20 .

mbrunettini, 2016-05-17:
icub-firmware-shared was already tagged in  when I closed [QA#20](https://github.com/robotology/QA/issues/20).
The remaining repositories are going to be tagged in this icub release (1.1.17)
@lornat75 : what about changing version number to match the one in icub-main?

mbrunettini, 2016-05-17:
all requested repo have been tagged with icub 1.1.17

pattacini, 2016-05-17:
Honestly, it doesn't have much sense to tag **`icub-basic-demos`**. Given that the code therein hasn't changed since ages, we'll end up having many tags that are coincident.

What makes more sense is to enforce a dependency in that repository from a specific tag of `icub-main` onward.

pattacini, 2016-05-17:
I'm going to remove tag from `icub-basic-demos`...

lornat75, 2016-05-17:
We need to make sure we have a way to identify versions of repositories that work together. One way to do this is to tag these repositories every time we do a release. Maybe the policy could be to skip tagging a repository (e.g. ``icub-basic-demos'') if there are no new commits since the previous tag.  Does this work?

pattacini, 2016-05-17:
This will reduce the clutter of having multiple coincident tags, but eventually will make the use of tagging less useful on `icub-basic-demos` since there won't be any match among tags across repositories.

However we can go with that if you like.

traversaro, 2016-05-17:
If `icub-basic-demos` normally does not change, I think it is ok to not tag it. Whenever we have a change in them, we can always start tagging it.

Tobias-Fischer, 2016-05-26:
Related o this issue, I think it would be useful to be able to check the version of `icub-main` in subprojects. For now, the version information seems to be wrong: https://github.com/robotology/icub-main/blob/master/conf/iCubVersion.cmake (1.2.17 instead of 1.1.17).

lornat75, 2016-05-27:
The cmake version information should be used exactly for that purpose. @mbrunettini @drdanz can you check this?

mbrunettini, 2016-05-27:
fixed

traversaro, 2016-05-27:
@mbrunettini Please note that the bump to version 1.2.17 was discussed and decided with @pattacini  in https://github.com/robotology/QA/issues/68#issuecomment-150624649 . 

This change is breaking the CMake code in which we have: 
~~~
find_package(ICUB 1.2.17)
~~~
fortunately we never seems we never add that explicit dependency on the CMake, but it would be good to actually announce a change like that before reverting the version of the repo.

traversaro, 2016-11-28:
@drdanz Reopening, we can close when we update https://github.com/robotology/icub-main/blob/devel/admin/scripts/README_RELEASE#L54 .

traversaro, 2016-11-28:
(BTW: we should investigate tooling for doing releases in all this repos at the same time, otherwise we will get crazy and go back to a monolithic repo sooner or later : ) ).

closed 2017-03-28 by pattacini