robotology/icub-main #348: objectsPropertiesCollector not closing properly

Tobias-Fischer, 2016-06-02:
Hi @pattacini,
We still observe the same issue that the OPC does not close properly. It gets stuck somewhere, and even sending the stop command repeatedly in the yarpmanager does not close the OPC. I assume it is somehow related to other modules which simultaneously interrupt / close the port to `/OPC/rpc` and `/OPC/broadcast:o`. It happens all the time in the WYSIWYD demo.
It would be great if we can have a look at this. If you let me know where to start debugging, I can help.

Best,
Tobi

pattacini, 2016-06-02:
Hi @Tobias-Fischer 

Quite difficult to figure out. We ought to first isolate where `OPC` gets stuck on closure by filling its code with lots of traps everywhere.

pattacini, 2016-06-04:
@Tobias-Fischer 

First off, I would check the functionality of this [**`read`**](https://github.com/robotology/icub-main/blob/master/src/modules/objectsPropertiesCollector/main.cpp#L1468) method when we're on closure, especially in view of the mutex and the flag `enabled`. Filling that function with `yDebug` would be my initial step.

Unfortunately, I'm on others demos these days and, also, replicating verbatim the wysiwyd demo here, that is where the problems occur, is not that straightforward. So, I'm afraid I'll be of little help...

Tobias-Fischer, 2016-06-04:
Okay @pattacini, the next time we'll have a demo I'll put debug messages in the `read`, `interrupt` and `close` methods and will report back. Maybe we can figure it from there. As if it wasn't tricky enough already, it seems like it's random as well, as sometimes the OPC seems to close okay.

Tobias-Fischer, 2016-06-22:
Recently I didn't encounter this problem anymore. I'm not sure whether there were some underlying changes in `yarpmanager` or somewhere else inside `yarp` which might have solved it?!?

I'll continue observing ..

pattacini, 2016-06-22:
I'm not aware of any possible changes in this respect...

pattacini, 2016-09-01:
Any other occurrence of this problem @Tobias-Fischer ?

Tobias-Fischer, 2016-09-01:
I think we can close it for now. I'll reopen if this happens again.

closed 2016-09-01 by Tobias-Fischer