robotology/icub-main #349: Broken sprintf in trajectoryPlayer

drdanz, 2016-06-09:
@randaz81 These sprintf are missing the arguments, I don't know what they are supposed to write on the string:

[utils.cpp:209](https://github.com/robotology/icub-main/blob/268ccc8489576a0e355e04c1ed8a321a651bb069/src/tools/trajectoryPlayer/utils.cpp#L209)
```c++
    sprintf(command_line_format, "%d %lf    ");
```
[utils.cpp:213](https://github.com/robotology/icub-main/blob/268ccc8489576a0e355e04c1ed8a321a651bb069/src/tools/trajectoryPlayer/utils.cpp#L213)
```c++
 sprintf(command_line_format+off, "%lf ");
```

randaz81, 2016-06-09:
The code is ok... the intention was to write the string "%d %lf" into the command_line_format variable...

randaz81, 2016-06-09:
no, escapes are missing

commit a060f800 (2016-06-09): Fix -Wformat warnings in trajectoryPlayer

Fixes #349
closed 2016-06-09 by drdanz