robotology/icub-main #352: iCubSim Self-Collision

tunaonur, 2016-06-10:
I am trying to avoid self collision in iCub simulator. Although I saw comments in the code saying "these are needed for the iCub self-collisions", I cannot avoid self collision in iCub when I used yarpmotorgui. Is there way for me to avoid self collision in iCub simulator?

claudiofantacci, 2016-06-10:
Hi @tunaonur ,
could you please give us some more details about "I cannot avoid self collision in iCub when I used yarpmotorgui"?

barbalberto, 2016-06-10:
The real robot does not have (yet) a self-collision avoidance check on board, therefore neither the simulator has it.
The yarpmotorgui is simply a gui allowing a user friendly way to send command and it has no robot-specific logic inside.

So the short answer is no, there is no way to right now to check for collision using a yarpmotorgui directly connected to the robot.

I don't know if there is a library or module doing self-collision avoidance already implemented but in this case I hardly see how it could be used with the yarpmotorgui. I guess a 'module-in-the-middle' will be required.

tunaonur, 2016-06-10:
Hi @claudiofantacci ,
What I meant was the robot's hands are passing through each other instead of colliding and I am trying to avoid this to happen. As I explained below, I am just looking for a way to prevent the parts of the robot to pass through each other when I move them. Thanks a lot!

@barbalberto Thanks for your answer. What if I don't use yarpmotorgui? If I write a code to move the arm of robot in iCub simulator, is there a way for me to check for collision. Besides, I realized that in this file: /usr/local/share/iCub/contexts/simConfig iCub_parts_activation.ini  there is a part called self_collision. What changes when I change that from on to off or vice versa? Do you have an idea? Thanks in advance!

claudiofantacci, 2016-06-10:
The **self_collision** and **covers_collisions** properties are described here: https://www.icub.org/wiki/Simulator_README#COLLISIONS_AND_SELF-COLLISIONS:
It may be of help as far as the iCub simulator is concerned.
On the other hand, @barbalberto comments apply for the real iCub.

barbalberto, 2016-06-10:
Thanks @claudiofantacci I didn't know about collision flags in the iCub_SIM

randaz81, 2016-06-11:
OffTopic: I recommend to use iCubGazebo and not iCubSim, which is old and it will be (soon or later) dismissed.

tunaonur, 2016-06-17:
@claudiofantacci I read about the self-collision and collision flags for iCub Simulator. I have been working on them for a while. Although the flags are on, the arms of the robot in the simulator are still passing through each other both when I tried to move them using yarpmotorgui and when I tried to move them using a code which I share below. Do you have an idea about why this is happening? Thank you very much! 

https://gist.github.com/tunaonur/28fe63a5566757af90030488cb8e5d18

claudiofantacci, 2016-06-20:
Hi @tunaonur , unfortunately I'm not an expert user of iCub Simulator collisions, but I've just noticed that you found out PR #69 and @matejhof could provide help. One thing that you may be missing is the proper compilation of the ODE library (as suggested by @matejhof in #69 ).
I would also suggest you to try using Gazebo, as suggested by @randaz81 .

matejhof, 2016-06-20:
Hi @tunaonur , yes indeed, I have now amended the documentation here: [Simulator README](https://www.icub.org/wiki/Simulator_README#COLLISIONS_AND_SELF-COLLISIONS:) with the following sentence: It is also recommended to have ODE compiled with the --enable-libccd option on (see http://eris.liralab.it/wiki/ICub_Simulator_Installation), which is necessary for cylinder-cylinder collisions support (for example, all iCub simulator fingers are cylinders). 

Can you check whether it's just cylinder-cylinder collisions you are missing (e.g. fingers going through each other) or if it happens also elsewhere - when you touch the covers of the forearm for example. An example of correct behavior is in this video: [iCub simulator self-touch](https://www.dropbox.com/s/lvwpzqo817mz16u/icubSimDoubleTouchSkinEmulation.mkv?dl=0)

tunaonur, 2016-06-22:
Dear @matejhof,
Although I did what you had recommended, I am still having self-collision problem. I am still missing cylinder-cylinder collision. I am sending the current situation in a video. Thanks a lot for your help! 

https://drive.google.com/file/d/0B1Y8sN9-hriZS2JhYmlxU2ZMQjQ/view?usp=sharing

matejhof, 2016-06-24:
I looked at your video - you're missing all collisions, not just cylinder-cylinder. In your video, everything is penetrated - also the palm and the covers, which are not cylinders.
I don't know what the problem could be then. Are you sure the right config params (flags) are loaded from the right .ini file? iCub_SIM will write it out to the console when started.

tunaonur, 2016-06-27:
Dear @matejhof,
I think the right flags are loaded. Might have you watched only the first half of the video? The second half is the one in which the flags are on. Thanks!

matejhof, 2016-06-28:
Aha, yes, you're right. So in the second part, it really seems only the cylinder-cylinder collisions are missing. Which OS are you using? Did you compile ODE from source and is it the version that is actually used by your system?
Do not use the old debian package ODE 0.11 (that comes with icub-common dependencies), but compile on your own the latest ODE with the `libccd` option enabled. Have a look [here](http://eris.liralab.it/wiki/Linux:_Installing_ODE).
You may have to set the `ODE_DIR` for yarp/icub-main configuration.

matejhof, 2016-09-01:
@tunaonur are you still having problems there? Anybody else having problems with the self-collisions? On my stone age system (Ubuntu 12.04) everything's working nicely.

pattacini, 2016-09-03:
Closing for time being.

closed 2016-09-03 by pattacini