robotology/icub-main #353: [OSX] Problem with GCC 6.1

claudiofantacci, 2016-06-13:
After ```brew update && brew upgrade```-ing I found myself with GCC 6.1 brew-linked and with several modules not working properly or not running at all due to some missing/wrong library paths.

I tried to recompile everything from scratch deleting my *build* folder, without success. The modules still were not able to run properly as some of them were still pointing to some GCC 5 libraries.
Here are some screenshots of the ```ccmake``` terminal GUI with brew-linked GCC 6.1.0:
[image]
[image]

To have everything back working properly I had to ```brew switch gcc 5.3.0``` (that was my last GCC 5.* installed version) and compile everything again.

Looking closely to red highlighted paths, they should be pointing to the more general ```/usr/local/opt/gcc```, which is used by ```brew``` to symlink each package to its very latest installed version (see the following picture).
[image]

kt10aan, 2016-06-15:
I am not having the same issue. Everything works fine with ``gcc 6.1`` without having to use ``brew switch gcc 5.3.0``. Maybe you have some issue with your ``Homebrew`` configuration?

claudiofantacci, 2016-06-15:
@kt10aan no problem with ```Homebrew```, ```brew doctor``` reported no issues whatsoever.

I solved the problem recompiling from scratch, in order:
1. ```Ipopt```, downloading the ZIP archive and using the *ThirdParty* scripts as suggest in http://wiki.icub.org/wiki/Homebrew_Mavericks (cc @pattacini )
2. ```YARP```
3. ```iCub-main```

The problem was related to ```Ipopt``` compilation made with GCC 5.*.
The proper 6.1 library links were updated (and logged on the terminal by) the Ipopt ```configure``` script.

closed 2016-06-15 by claudiofantacci