robotology/icub-main #355: yarplogger error when starting yarprobotinterface for ETH iCub's

traversaro, 2016-06-23:
At startup of ETH robots (just tried on `iCubGenova05`) the following error is printed by the `yarprobotinterface` : 
~~~
39		0,313448		ERROR		HasSpeedEncoder  parameter not found 
77		0,525780		ERROR		HasSpeedEncoder  parameter not found 
121		0,727998		ERROR		HasSpeedEncoder  parameter not found 
160		0,940414		ERROR		HasSpeedEncoder  parameter not found 
~~~
I guess this was added in https://github.com/robotology/icub-main/commit/294caccd76fd94e668219376851ea6b80db52ef9 .
 
Can we had a line of  
~~~
<param name="HasSpeedEncoder"> 0 0 0 0 0 0 0 0 0 0 0 0 </param>
~~~
in 
https://github.com/robotology/icub-main/blob/master/app/robots/iCubGenova05/hardware/mechanicals/left_lower_arm-ems2-mec.xml (and all the relative files) to get rid of this error? 
Should we add it to all ETH robots? 

cc @ale-git @randaz81 @barbalberto

ale-git, 2016-06-23:
It is reported with the ERROR tag but it is not a mandatory parameter, anyway I'm going to add it to all the eth robots.

traversaro, 2016-06-29:
Fixed by https://github.com/robotology/icub-main/commit/1d55d3ed2a83f3f653d7cd5abdd7f6d468217ca0 .

closed 2016-06-29 by traversaro