robotology/icub-main #360: Dedicated repository for robots configuration

pattacini, 2016-08-10:
We've been discussing since a while about the opportunity to host robots configuration files from a dedicated repository.

This solution brings the following advantages:
- Make `icub-main` much _slimmer_ :smile: 
- Clearly identify _responsibilities_ and _duties_ among code developers (operating merely on `icub-main`) and robots maintainers (operating mostly on the new repo).
- As a consequence, the two _workflows_ remain separated and do not interfere each other, which is a nice pro since we're gonna introduce the new branching in `icub-main` (aka `master` and `devel` lines) very soon. Branching is an important feature for developers, while robot maintainers can continue working happily and unaware of the new code policies :wink:

All along with the following downsides:
- Moving robots files in the new repo will likely cause some headaches to users during the transition. However, with proper announcements and support we can deal with this.

Therefore, I've made an attempt to implement this solution and provided those files in the brand new repo: [**`robotology/robots-configuration`**](https://github.com/robotology/robots-configuration). Of course, it represents only an **experiment** as of now and `icub-main` is still in place to export the configuration files.

The migration is done keeping the history of the files and the `icub-contrib-common` helper package needs to be used to install them correctly on the system.

I've carried out a bunch of tests confirming that everything is working fine.

@lornat75 @randaz81 @drdanz please have a look, revise and decide what's next.

cc @vtikha @marcoaccame @barbalberto @valegagge @julijenv @davidetome 
cc @robotology/icub-developers @robotology/icub-support-team

alecive, 2016-08-18:
A suggestion: to limit disruption at the user level, a partial transition can be done by using `git submodules`. You instantiate the new `robots-configuration` repo as a submodule of `icub-main` and what would be exposed to the user is the same folder structure as before. After that you can tell the users that the submodule is going to be removed from `icub-main`, and the `robots-configuration` repo should be cloned as a proper repo when needed.

pattacini, 2016-08-18:
@alecive That would be a milder transition indeed, thanks for pointing it out, but you know how much I root for radical cuts :wink: 

However, it could be certainly a solution to assess for @lornat75 @drdanz @randaz81

drdanz, 2016-08-18:
:-1: for `git submodules` they will just complicate everyone's existence

francesco-romano, 2016-08-18:
@alecive  I don't know if in the last 2/3 years they have improved, but I've always found very complex to manage a repository with submodules

pattacini, 2016-08-18:
[**Why your company shouldn’t use Git submodules**](https://codingkilledthecat.wordpress.com/2012/04/28/why-your-company-shouldnt-use-git-submodules/)

... I didn't even finish reading this post that I developed the opinion it's a **bad idea** to go with `git submodules` :disappointed:

lornat75, 2016-08-18:
The main issue is to make icub-main and firmware(?) repositories depend from the robot configuration repository, and, more importantly, define and follow a policy to handle versions.

If we manage vsioning  and dependencies correctly I think the separation is a good thing.

pattacini, 2016-08-18:
@lornat75 isn't the other way round more correct, that is `robots-configuration` depending on `yarp`, `icub-main` and/or `icub-firmware`?

For example, an update to `yarprobotinterface` could induce an update to `robots-configuration`.

Am I missing something else?

lornat75, 2016-08-18:
You are right, anyway it is more of a run-time dependency. The main issue is that we need to find a way to notify users when they need to update their configuration files. When everything was within a unique repository this was easier.

Maybe we can rely on mechanisms that are already in place for run-time checking of firmware version etc. We need to talk with the firmware developers and @randaz81.

pattacini, 2016-09-16:
@randaz81 to keep track of the commands we need to run to [_extract parts of a repository keeping the history_](https://ariya.io/2014/07/extracting-parts-of-git-repository-and-keeping-the-history).

pattacini, 2016-09-22:
Agreed!
Let's do it ✨

closed 2016-09-22 by pattacini