robotology/icub-main #361: IPOPT link missing during compilation

gregoire-pointeau, 2016-08-16:
Hello.
I have a problem while compiling `icub-main` and `iol`.

I hadn't updated the robot since a bit of time (I would say 2 months, we stayed on a stable version).
Since I have a compilation problem that seems to come from Ipopt.

On my console I was able by  re-building from scratch, to compile `yarp icub-main contrib-common kinect-wrapper wysiwyd stereo-vison segmentation` but it fails when it comes to `iol`:

> icub@console1:/mnt/data/Data/robot/icub-contrib-iit/iol/build$ make
[  8%] Built target blobSelector
[ 16%] Built target iolHelper
[ 50%] Built target iolStateMachineHandler
Linking CXX executable ../../bin/iolReachingCalibration
//usr/local/lib/liboptimization.a(calibReference.cpp.o): dans la fonction « iCub::optimization::CalibReferenceWithMatchedPoints::calibrate(yarp::sig::Matrix&, yarp::sig::Vector&, double&) »:
calibReference.cpp:(.text+0x1dcc): référence indéfinie vers « Ipopt::IpoptApplication::IpoptApplication(bool, bool) »
//usr/local/lib/liboptimization.a(calibReference.cpp.o): dans la fonction « iCub::optimization::CalibReferenceWithMatchedPoints::calibrate(yarp::sig::Matrix&, double&, double&) »:
calibReference.cpp:(.text+0x2e8a): référence indéfinie vers « Ipopt::IpoptApplication::IpoptApplication(bool, bool) »
//usr/local/lib/liboptimization.a(calibReference.cpp.o): dans la fonction « iCub::optimization::CalibReferenceWithMatchedPoints::calibrate(yarp::sig::Matrix&, double&) »:
calibReference.cpp:(.text+0x3f07): référence indéfinie vers « Ipopt::IpoptApplication::IpoptApplication(bool, bool) »
//usr/local/lib/liboptimization.a(calibReference.cpp.o): dans la fonction « Ipopt::RegisteredOption::~RegisteredOption() »:
calibReference.cpp:(.text._ZN5Ipopt16RegisteredOptionD0Ev[_ZN5Ipopt16RegisteredOptionD5Ev]+0x12): référence indéfinie vers « vtable for Ipopt::RegisteredOption »
//usr/local/lib/liboptimization.a(calibReference.cpp.o): dans la fonction « Ipopt::RegisteredOption::~RegisteredOption() »:
calibReference.cpp:(.text._ZN5Ipopt16RegisteredOptionD2Ev[_ZN5Ipopt16RegisteredOptionD5Ev]+0x12): référence indéfinie vers « vtable for Ipopt::RegisteredOption »
//usr/local/lib/liboptimization.a(calibReference.cpp.o): dans la fonction « std::_Rb_tree<std::string, std::pair<std::string const, Ipopt::SmartPtr<Ipopt::RegisteredOption> >, std::_Select1st<std::pair<std::string const, Ipopt::SmartPtr<Ipopt::RegisteredOption> > >, std::less<std::string>, std::allocator<std::pair<std::string const, Ipopt::SmartPtr<Ipopt::RegisteredOption> > > >::_M_erase(std::_Rb_tree_node<std::pair<std::string const, Ipopt::SmartPtr<Ipopt::RegisteredOption> > >*) »:
calibReference.cpp:(.text._ZNSt8_Rb_treeISsSt4pairIKSsN5Ipopt8SmartPtrINS2_16RegisteredOptionEEEESt10_Select1stIS6_ESt4lessISsESaIS6_EE8_M_eraseEPSt13_Rb_tree_nodeIS6_E[_ZNSt8_Rb_treeISsSt4pairIKSsN5Ipopt8SmartPtrINS2_16RegisteredOptionEEEESt10_Select1stIS6_ESt4lessISsESaIS6_EE8_M_eraseEPSt13_Rb_tree_nodeIS6_E]+0x7e): référence indéfinie vers « vtable for Ipopt::RegisteredOption »
//usr/local/lib/liboptimization.a(calibReference.cpp.o): dans la fonction « Ipopt::RegisteredOptions::~RegisteredOptions() »:
calibReference.cpp:(.text._ZN5Ipopt17RegisteredOptionsD2Ev[_ZN5Ipopt17RegisteredOptionsD5Ev]+0x10): référence indéfinie vers « vtable for Ipopt::RegisteredOptions »
//usr/local/lib/liboptimization.a(calibReference.cpp.o): dans la fonction « Ipopt::RegisteredOptions::~RegisteredOptions() »:
calibReference.cpp:(.text._ZN5Ipopt17RegisteredOptionsD0Ev[_ZN5Ipopt17RegisteredOptionsD5Ev]+0x10): référence indéfinie vers « vtable for Ipopt::RegisteredOptions »
//usr/local/lib/liboptimization.a(calibReference.cpp.o): dans la fonction « Ipopt::SmartPtr<Ipopt::RegisteredOptions>::ReleasePointer_() »:
calibReference.cpp:(.text._ZN5Ipopt8SmartPtrINS_17RegisteredOptionsEE15ReleasePointer_Ev[_ZN5Ipopt8SmartPtrINS_17RegisteredOptionsEE15ReleasePointer_Ev]+0x27): référence indéfinie vers « vtable for Ipopt::RegisteredOptions »
//usr/local/lib/liboptimization.a(calibReference.cpp.o): dans la fonction « Ipopt::OptionsList::~OptionsList() »:
calibReference.cpp:(.text._ZN5Ipopt11OptionsListD2Ev[_ZN5Ipopt11OptionsListD5Ev]+0x10): référence indéfinie vers « vtable for Ipopt::OptionsList »
calibReference.cpp:(.text._ZN5Ipopt11OptionsListD2Ev[_ZN5Ipopt11OptionsListD5Ev]+0x6b): référence indéfinie vers « vtable for Ipopt::RegisteredOptions »
//usr/local/lib/liboptimization.a(calibReference.cpp.o): dans la fonction « Ipopt::OptionsList::~OptionsList() »:
calibReference.cpp:(.text._ZN5Ipopt11OptionsListD0Ev[_ZN5Ipopt11OptionsListD5Ev]+0x10): référence indéfinie vers « vtable for Ipopt::OptionsList »
calibReference.cpp:(.text._ZN5Ipopt11OptionsListD0Ev[_ZN5Ipopt11OptionsListD5Ev]+0x6b): référence indéfinie vers « vtable for Ipopt::RegisteredOptions »
//usr/local/lib/liboptimization.a(calibReference.cpp.o): dans la fonction « Ipopt::SmartPtr<Ipopt::OptionsList>::ReleasePointer_() »:
calibReference.cpp:(.text._ZN5Ipopt8SmartPtrINS_11OptionsListEE15ReleasePointer_Ev[_ZN5Ipopt8SmartPtrINS_11OptionsListEE15ReleasePointer_Ev]+0x27): référence indéfinie vers « vtable for Ipopt::OptionsList »
collect2: error: ld returned 1 exit status
src/iolReachingCalibration/CMakeFiles/iolReachingCalibration.dir/build.make:165: recipe for target 'bin/iolReachingCalibration' failed
make[2]: *** [bin/iolReachingCalibration] Error 1
CMakeFiles/Makefile2:303: recipe for target 'src/iolReachingCalibration/CMakeFiles/iolReachingCalibration.dir/all' failed
make[1]: *** [src/iolReachingCalibration/CMakeFiles/iolReachingCalibration.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2

While on the blade of the cluster, I recompiled `yarp` from scratch, but `icub-main` fails:

> icub@icub-b1:/usr/local/src/robot/icub-main/build-x86_64$ make
[  0%] Copying manager.py to /usr/local/src/robot/icub-main/build-x86_64/bin/.
[  0%] Built target python-scripts
[  0%] Built target robotSymLink
[  1%] Copying icub-cluster scripts to /usr/local/src/robot/icub-main/build-x86_64/bin/.
[  1%] Built target cluster-scripts
[  2%] Built target iCubDev
[  6%] Built target ctrlLib
[  9%] Built target skinDynLib
[ 10%] Building CXX object src/libraries/iKin/CMakeFiles/iKin.dir/src/iKinIpOpt.cpp.o
/usr/local/src/robot/icub-main/src/libraries/iKin/src/iKinIpOpt.cpp:21:22: fatal error: IpTNLP.hpp: No such file or directory
compilation terminated.
make[2]: *** [src/libraries/iKin/CMakeFiles/iKin.dir/src/iKinIpOpt.cpp.o] Error 1
make[1]: *** [src/libraries/iKin/CMakeFiles/iKin.dir/all] Error 2
make: *** [all] Error 2

Ipopt seems to be a problem, but I can't see how to solve it.

drdanz, 2016-08-16:
on which debian/ubuntu version? Is the package `coinor-libipopt-dev` installed?
Can you paste the output of the command `dpkg -l coinor-libipopt-dev icub-common`?

gregoire-pointeau, 2016-08-16:
Console:

> Souhait=inconnU/Installé/suppRimé/Purgé/H=à garder
| État=Non/Installé/fichier-Config/dépaqUeté/échec-conFig/H=semi-installé/W=attend-traitement-déclenchements
|/ Err?=(aucune)/besoin Réinstallation (État,Err: majuscule=mauvais)
||/ Nom                                            Version                      Architecture                 Description
+++-==============================================-============================-============================-==================================================================================================
un  coinor-libipopt-dev                            <aucune>                     <aucune>                     (aucune description n'est disponible)
ii  icub-common                                    1.1.16-2~wheezy              amd64                        List of dependencies for iCub software

version: 

> icub@console1:~$
 lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	DebianEdu/Skolelinux
Release:	8.5
Codename:	jessie

blades:

> Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                           Version                      Architecture                 Description
+++-==============================================-============================-============================-==================================================================================================
un  coinor-libipopt-dev                            <none>                                                    (no description available)
ii  icub-common                                    1.1.14-0+hack-opencv~wheezy  amd64                        List of dependencies for iCub software

version:

> icub@icub-b1:~$ lsb_release -a
LSB Version:	core-2.0-amd64:core-2.0-noarch:core-3.0-amd64:core-3.0-noarch:core-3.1-amd64:core-3.1-noarch:core-3.2-amd64:core-3.2-noarch:core-4.0-amd64:core-4.0-noarch:core-4.1-amd64:core-4.1-noarch:languages-3.2-amd64:languages-3.2-noarch:languages-4.0-amd64:languages-4.0-noarch:languages-4.1-amd64:languages-4.1-noarch:security-4.0-amd64:security-4.0-noarch:security-4.1-amd64:security-4.1-noarch
Distributor ID:	Debian
Description:	Debian GNU/Linux 7.2 (wheezy)
Release:	7.2
Codename:	wheezy

drdanz, 2016-08-16:
Please update the icub-common package on all the systems and try again, the current package version is `1.4.0-1~wheezy` for `wheezy` and `1.4.0-4~jessie` for `jessie` (on `jessie` it should download the `coinor-libipopt-dev` package, and on `wheezy` it will conflict with the same package, this is ok, the `wheezy` package is missing some build flag, therefore `icub-common` offers a custom version of the library that conflicts with the `coinor-libipopt-dev`, luckily for newer versions this is no longer necessary).
Also I'm not sure where the `1.1.14-0+hack-opencv~wheezy` version comes from, is it some custom build, or is it something that was in icub repository?

gregoire-pointeau, 2016-08-17:
Hi,
concerning the console, I updated the package but it seems blocked at 1.1.15-1:

Update:
> icub@console1:/mnt/data/Data/robot$ sudo apt-get install icub-common 
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances
Lecture des informations d'état... Fait
icub-common est déjà la plus récente version disponible.
0 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.

Version:
> icub@console1:/mnt/data/Data/robot$ dpkg -p icub-common 
Package: icub-common
Priority: optional
Section: contrib/science
Installed-Size: 6848
Maintainer: Alberto Cardellino <alberto.cardellino@iit.it>
Architecture: amd64
Version: 1.1.15-1~wheezy
Depends: libace-dev, libgsl0-dev, libc6, python (<= 3), libncurses5-dev, libgtkmm-2.4-dev, libglademm-2.4-dev, libqt4-dev, libqt4-opengl-dev, libcv-dev, libhighgui-dev, libcvaux-dev, libsdl1.2-dev, subversion, git, gfortran, freeglut3-dev, cmake (>= 2.8.7), libxmu-dev, libode1 (>= 0.11.1), libode-dev, libopencv-dev
Conflicts: coinor-libipopt0, coinor-libipopt-dev
Size: 2630766
Description: List of dependencies for iCub software
 This package lists all the dependencies needed to install the icub 
 platform software or to download the source code and compile it directly
 onto your machine. It contains also a compiled version of IpOpt library
Homepage: http://www.robotcub.org, https://projects.coin-or.org/Ipopt

While it seems a bit more complicated on the blades (the cluster was initialized last year at IIT):
Update:
> icub@icub-b1:/usr/local/src/robot$ sudo apt-get install icub-common 
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:
The following packages have unmet dependencies:
 icub-common : Depends: qtbase5-dev but it is not installable
               Depends: qtdeclarative5-dev but it is not installable
               Depends: qtmultimedia5-dev but it is not installable
               Depends: libqt5svg5 but it is not installable
               Depends: qml-module-qtquick2 but it is not installable
               Depends: qml-module-qtquick-window2 but it is not installable
               Depends: qml-module-qtmultimedia but it is not installable
               Depends: qml-module-qtquick-dialogs but it is not installable
               Depends: qml-module-qtquick-controls but it is not installable
E: Unable to correct problems, you have held broken packages.

Version:
> icub@icub-b1:/usr/local/src/robot$ dpkg -p icub-common
Package: icub-common
Priority: optional
Section: contrib/science
Installed-Size: 6384
Maintainer: Alberto Cardellino <alberto.cardellino@iit.it>
Architecture: amd64
Version: 1.1.14-0+hack-opencv~wheezy
Depends: libace-dev, libgsl0-dev, libc6, python (<= 3), libncurses5-dev, libgtkmm-2.4-dev, libglademm-2.4-dev, libqt4-dev, libqt4-opengl-dev, libsdl1.2-dev, subversion, git, gfortran, freeglut3-dev, cmake (>= 2.8.7), libxmu-dev, libode1 (>= 0.11.1), libode-dev
Suggests: libcv-dev, libhighgui-dev, libcvaux-dev, libopencv-dev
Conflicts: coinor-libipopt0, coinor-libipopt-dev
Size: 2417692
Description: List of dependencies for iCub software
 This package lists all the dependencies needed to install the icub 
 platform software or to download the source code and compile it directly
 onto your machine. It contains also a compiled version of IpOpt library
Homepage: http://www.robotcub.org, https://projects.coin-or.org/Ipopt

drdanz, 2016-08-17:
If I understand correctly, on the blades you have debian wheezy, since now icub-common requires qt5 for the new GUIs, you need also the wheezy-backports repository. You have to add a file `/etc/apt/sources.list.d/wheezy-backports.list` with this content:

```
deb http://backports.debian.org/debian-backports/ wheezy-backports main
deb-src http://backports.debian.org/debian-backports/ wheezy-backports main
```

Can you show me the content of the file(s) `/etc/apt/sources.list.d/icub.list` and the output of the command `apt-cache policy icub-common`?

gregoire-pointeau, 2016-08-17:
`/etc/apt/sources.list.d/icub.list` : `deb http://www.icub.org/debian wheezy contrib/science`

> icub@icub-b1:/etc/apt/sources.list.d$ apt-cache policy icub-common
icub-common:
  Installed: 1.1.14-0+hack-opencv~wheezy
  Candidate: 1.1.16-1~wheezy
  Version table:
     1.1.16-1~wheezy 0
        500 http://www.icub.org/debian/ wheezy/contrib/science amd64 Packages
 *** 1.1.14-0+hack-opencv~wheezy 0
        100 /var/lib/dpkg/status

I'll try to add the file. However I wonder, to I really need Qt on the blades because we cannot display anything on the blades?

drdanz, 2016-08-17:
Unfortunately I think that at the moment the icub-common package (that for wheezy contains ipopt) is has qt5 packages as dependencies. The situation is going to improve with jessie, where the icub-common is just a meta-package that downloads all the dependencies and therefore not required, you can install all of them manually.

>  Candidate: 1.1.16-1~wheezy

That's weird... according to http://www.icub.org/debian/dists/wheezy/contrib/science/binary-amd64/Packages version `1.4.0-1~wheezy` should be available.
Can you run `apt-get update` and try again `apt-cache policy icub-common`?

gregoire-pointeau, 2016-08-17:
Ok, after the update and the newback-port list, I have a target for 1.4 but still cannot install it:

> icub@icub-b1:/usr/local/src/robot$ apt-cache policy icub-common
icub-common:
  Installed: 1.1.14-0+hack-opencv~wheezy
  Candidate: 1.4.0-1~wheezy
  Version table:
     1.4.0-1~wheezy 0
        500 http://www.icub.org/debian/ wheezy/contrib/science amd64 Packages
 *** 1.1.14-0+hack-opencv~wheezy 0
        100 /var/lib/dpkg/status
icub@icub-b1:/usr/local/src/robot$ apt-get install icub-common
E: Could not open lock file /var/lib/dpkg/lock - open (13: Permission denied)
E: Unable to lock the administration directory (/var/lib/dpkg/), are you root?
icub@icub-b1:/usr/local/src/robot$ sudo apt-get install icub-common
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 icub-common : Depends: qtbase5-dev but it is not installable
               Depends: qtdeclarative5-dev but it is not installable
               Depends: qtmultimedia5-dev but it is not installable
               Depends: libqt5svg5 but it is not installable
               Depends: qml-module-qtquick2 but it is not installable
               Depends: qml-module-qtquick-window2 but it is not installable
               Depends: qml-module-qtmultimedia but it is not installable
               Depends: qml-module-qtquick-dialogs but it is not installable
               Depends: qml-module-qtquick-controls but it is not installable
E: Unable to correct problems, you have held broken packages.

drdanz, 2016-08-17:
Hmm, probably the backports repo does not have enough priority to be installed

Try this on the blades:
```
sudo apt-get install -t wheezy-backports qtbase5-dev qtdeclarative5-dev qtmultimedia5-dev qml-module-qtquick2 qml-module-qtquick-window2 qml-module-qtmultimedia qml-module-qtquick-dialogs qml-module-qtquick-controls libqt5svg5
sudo apt-get install icub-common
```

drdanz, 2016-08-17:
About the console, `lsb_release` reports that you are running `jessie`, but the package is for `wheezy`... Also apt-cache policy does not show the version on the repo.
This probably means that the repo is not correctly configured.

Try running:

```
sudo sh -c 'echo "deb http://www.icub.org/debian jessie contrib/science" > /etc/apt/sources.list.d/icub.list'
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 57A5ACB6110576A6
```

Then run:

```bash
sudo apt-get update
sudo apt-get install icub-common
```

gregoire-pointeau, 2016-08-17:
Thanks.
Concerning the blades:

> icub@icub-b1:/usr/local/src/robot$ sudo apt-get install -t wheezy-backports qtbase5-dev qtdeclarative5-dev qtmultimedia5-dev qml-module-qtquick2 qml-module-qtquick-window2 qml-module-qtmultimedia qml-module-qtquick-dialogs qml-module-qtquick-controls libqt5svg5
Reading package lists... Done
E: The value 'wheezy-backports' is invalid for APT::Default-Release as such a release is not available in the sources

While on the console:

> icub@console1:~$ sudo sh -c 'echo "deb http://www.icub.org/debian jessie contrib/science" > /etc/apt/sources.list.d/icub.list'

> icub@console1:~$ sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 57A5ACB6110576A6
Executing: gpg --ignore-time-conflict --no-options --no-default-keyring --homedir /tmp/user/0/tmp.9Rut962eB9 --no-auto-check-trustdb --trust-model always --keyring /etc/apt/trusted.gpg --primary-keyring /etc/apt/trusted.gpg --keyring /etc/apt/trusted.gpg.d/deb-multimedia-keyring.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-jessie-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-jessie-security-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-jessie-stable.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-squeeze-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-squeeze-stable.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-wheezy-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-wheezy-stable.gpg --keyserver keyserver.ubuntu.com --recv-keys 57A5ACB6110576A6
gpg: demande de la clef 110576A6 sur le serveur hkp keyserver.ubuntu.com
gpg: le délai d'attente du serveur de clefs a expiré
gpg: échec de réception depuis le serveur de clefs : erreur du serveur de clefs

drdanz, 2016-08-17:
> E: The value 'wheezy-backports' is invalid for APT::Default-Release as such a release is not available in the sources

https://github.com/robotology/icub-main/issues/361#issuecomment-240359574

 > gpg: le délai d'attente du serveur de clefs a expiré

I don't speak french, but this looks like a timeout error... try again

gregoire-pointeau, 2016-08-17:
It is indeed a timeout error. But I get it all the times...

drdanz, 2016-08-17:
what does `ping -c3 keyserver.ubuntu.com` print?
Do the blades have access to the network?
Can you post the content of `/etc/resolv.conf`?

gregoire-pointeau, 2016-08-17:
Both ping (console and blades) get nothing (100% loss).
I guess that the blades should have access to the network.

resolv.conf blade:

> domain icub.local
search icub.local
nameserver 10.0.0.1

resolv.conf console: 

> # Generated by NetworkManager
search icub.local icub.local.
nameserver 10.0.0.1

drdanz, 2016-08-17:
what about `nslookup keyserver.ubuntu.com`? `ping -c3 www.google.com`?

drdanz, 2016-08-17:
also `ping -c3 91.189.89.49`

gregoire-pointeau, 2016-08-17:
100% loss everywhere...
But I can however goes on internet on this computers... (I'm on the console right now even if `ping google` doesn't work).

drdanz, 2016-08-17:
Can you please post the output of the commands above? I need it to understand if the problem is the connection or the nameserver...

gregoire-pointeau, 2016-08-17:
> icub@icub-b1:~$ nslookup keyserver.ubuntu.com
Server:		10.0.0.1
Address:	10.0.0.1#53

> Non-authoritative answer:
Name:	keyserver.ubuntu.com
Address: 91.189.90.55
Name:	keyserver.ubuntu.com
Address: 91.189.89.49

> icub@icub-b1:~$ sudo ping -c3 91.189.89.49
PING 91.189.89.49 (91.189.89.49) 56(84) bytes of data.
--- 91.189.89.49 ping statistics ---
3 packets transmitted, 0 received, 100% packet loss, time 2015ms

> icub@icub-b1:~$ sudo ping -c3 91.189.89.49
PING 91.189.89.49 (91.189.89.49) 56(84) bytes of data.
--- 91.189.89.49 ping statistics ---
3 packets transmitted, 0 received, 100% packet loss, time 2014ms

> icub@icub-b1:~$ sudo ping -c3 www.google.com
PING www.google.com (77.95.65.121) 56(84) bytes of data.
--- www.google.com ping statistics ---
3 packets transmitted, 0 received, 100% packet loss, time 1999ms

Console: 

> icub@console1:~$ nslookup keyserver.ubuntu.com
Server:		10.0.0.1
Address:	10.0.0.1#53
Non-authoritative answer:
Name:	keyserver.ubuntu.com
Address: 91.189.89.49
Name:	keyserver.ubuntu.com
Address: 91.189.90.55

> icub@console1:~$ ping -c3 www.google.com
PING www.google.com (77.95.65.123) 56(84) bytes of data.
--- www.google.com ping statistics ---
3 packets transmitted, 0 received, 100% packet loss, time 2017ms

> icub@console1:~$ ping -c3 91.189.89.49
PING 91.189.89.49 (91.189.89.49) 56(84) bytes of data.
--- 91.189.89.49 ping statistics ---
3 packets transmitted, 0 received, 100% packet loss, time 1999ms

drdanz, 2016-08-17:
Ok, so the nameserver is working, but there is no internet access... But you are saying that the computer goes on the internet, so it makes no sense...
Can you post the output of these 2 commands?

```
sudo ip addr
sudo ip route show
```

gregoire-pointeau, 2016-08-17:
Console: 

> icub@console1:~$ sudo ip addr
[sudo] password for icub: 
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default 
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 00:90:f5:ec:3a:07 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.10/24 brd 10.0.0.255 scope global dynamic eth0
       valid_lft 4893sec preferred_lft 4893sec
    inet6 fe80::290:f5ff:feec:3a07/64 scope link 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 0c:8b:fd:29:ec:17 brd ff:ff:ff:ff:ff:ff

> icub@console1:~$ sudo ip route show
default via 10.0.0.1 dev eth0  proto static  metric 1024 
10.0.0.0/24 dev eth0  proto kernel  scope link  src 10.0.0.10 
169.254.0.0/16 dev eth0  scope link  metric 1000 

Blades: 
> icub@icub-b1:~$ sudo ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 16436 qdisc noqueue state UNKNOWN 
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP qlen 1000
    link/ether b0:83:fe:df:1d:c1 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.51/24 brd 10.0.0.255 scope global eth0
    inet6 fe80::b283:feff:fedf:1dc1/64 scope link 
       valid_lft forever preferred_lft forever
3: eth7: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN qlen 1000
    link/ether b0:83:fe:df:1d:c2 brd ff:ff:ff:ff:ff:ff
4: eth6: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN qlen 1000
    link/ether 00:0a:f7:7c:43:34 brd ff:ff:ff:ff:ff:ff
5: eth9: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN qlen 1000
    link/ether 00:0a:f7:7c:43:35 brd ff:ff:ff:ff:ff:ff

> icub@icub-b1:~$ sudo ip route show
default via 10.0.0.1 dev eth0 
10.0.0.0/24 dev eth0  proto kernel  scope link  src 10.0.0.51

drdanz, 2016-08-17:
Ok, can you try to run connect with ssh on 10.0.0.1 and run the same `ip` commands, and some `ping`?

gregoire-pointeau, 2016-08-17:
on 10.0.0.1 `icub-srv`
>icub@icub-srv:~$ nslookup keyserver.ubuntu.com
Server:		127.0.0.1
Address:	127.0.0.1#53
Non-authoritative answer:
Name:	keyserver.ubuntu.com
Address: 91.189.90.55
Name:	keyserver.ubuntu.com
Address: 91.189.89.49

>icub@icub-srv:~$ sudo ping -c3 www.google.com
PING www.google.com (74.125.206.104) 56(84) bytes of data.
--- www.google.com ping statistics ---
3 packets transmitted, 0 received, 100% packet loss, time 2016ms

>icub@icub-srv:~$ sudo ping -c3 91.189.89.49
PING 91.189.89.49 (91.189.89.49) 56(84) bytes of data.
--- 91.189.89.49 ping statistics ---
3 packets transmitted, 0 received, 100% packet loss, time 2015ms

>icub@icub-srv:~$ /sbin/ifconfig
eth0      Link encap:Ethernet  HWaddr 00:0c:29:03:97:89  
          inet addr:192.168.10.2  Bcast:192.168.10.3  Mask:255.255.255.252
          inet6 addr: fe80::20c:29ff:fe03:9789/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:52871116 errors:0 dropped:253 overruns:0 frame:0
          TX packets:28363230 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:67572675773 (62.9 GiB)  TX bytes:4456863983 (4.1 GiB)
          Interrupt:18 Base address:0x2024 
eth1      Link encap:Ethernet  HWaddr 00:0c:29:03:97:93  
          inet addr:10.0.0.1  Bcast:10.0.0.255  Mask:255.255.255.0
          inet6 addr: fe80::20c:29ff:fe03:9793/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:110043681 errors:0 dropped:446 overruns:0 frame:0
          TX packets:119318787 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:66590996743 (62.0 GiB)  TX bytes:94922474018 (88.4 GiB)
          Interrupt:19 Base address:0x20a4 
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:16436  Metric:1
          RX packets:47666 errors:0 dropped:0 overruns:0 frame:0
          TX packets:47666 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:4226456 (4.0 MiB)  TX bytes:4226456 (4.0 MiB)

drdanz, 2016-08-17:
`sudo ip route show`?

gregoire-pointeau, 2016-08-17:
> icub@icub-srv:~$ sudo ip route
[sudo] password for icub: 
default via 192.168.10.1 dev eth0 
10.0.0.0/24 dev eth1  proto kernel  scope link  src 10.0.0.1 
192.168.10.0/30 dev eth0  proto kernel  scope link  src 192.168.10.2

drdanz, 2016-08-17:
`ping 192.168.10.1`?

gregoire-pointeau, 2016-08-17:
Got a answer: 

> icub@icub-srv:~$ ping 192.168.10.1
PING 192.168.10.1 (192.168.10.1) 56(84) bytes of data.
64 bytes from 192.168.10.1: icmp_req=1 ttl=64 time=0.157 ms
64 bytes from 192.168.10.1: icmp_req=2 ttl=64 time=0.213 ms
64 bytes from 192.168.10.1: icmp_req=3 ttl=64 time=0.176 ms
64 bytes from 192.168.10.1: icmp_req=4 ttl=64 time=0.170 ms

gregoire-pointeau, 2016-08-17:
I have to admit, I didn't set up the network on neither the console or the cluster... I might not be very helpful to debut that... Thanks anyway for helping !

drdanz, 2016-08-17:
@gregoire-pointeau: no problem... the issue with the network seems to be outside of the cluster, so I cannot help.
When you fix the network let me know if you still have issues with icub-common

gregoire-pointeau, 2016-08-17:
Ok, thanks ! will do !

pattacini, 2016-09-01:
@gregoire-pointeau any update on this?

gregoire-pointeau, 2016-09-01:
Yep, maybe @AnneLaureM will explain better than me. But now it's solved.

pattacini, 2016-09-01:
Ok thanks! :+1: 
Closing.

closed 2016-09-01 by pattacini