robotology/icub-main #362: VirtualAnalogWrapper deviceId has been deprecated

gregoire-pointeau, 2016-08-23:
Hello.
After an update, `robotInterface` is failing at starting on iCubLyon01.
The errors are the following:

> ERROR		VirtualAnalogWrapper: the parameter 'deviceId' has been deprecated, please use parameter 'name' instead. 
 e.g. In the VFT wrapper configuration files of your robot, replace '<param name=deviceId> left_arm </param>' 
 with '/icub/joint_vsens/left_arm:i'  
Cannot open device torso_VFTserver 
...
ERROR		lAnalogWrapper: the parameter 'deviceId' has been deprecated, please use parameter 'name' instead. 
 e.g. In the FT wrapper configuration files of your robot, replace '<param name=deviceId> left_arm </param>' 
 with '/icub/left_arm/analog:o'  
yarpdev: ***ERROR*** driver <analogServer> was found but could not open

Everything was updated this morning.
Could someone help me concerning this AnalogWrapper please?
Thanks.

traversaro, 2016-08-23:
Hi @gregoire-pointeau , 
   the configuration files in `icub-main` for `iCubLyon01` seems to be updated (see https://github.com/robotology/icub-main/blob/master/app/robots/iCubLyon01/wrappers/VFT/torso_VFT_wrapper.xml , the modification mentioned in the error message were performed in commit https://github.com/robotology/icub-main/commit/605b7683e62eafca80caa79b0ebfd01723591835 ).

From the error message, it seems that you may have some old `torso_VFT_wrapper.xml` (for example in the `.local` directories) that is loaded by `robotInterface` instead of the new one from the repo. To quickly check all the copies of `torso_VFT_wrapper.xml`, run this two commands on the `pc104`: 
~~~
sudo updatedb
locate torso_VFT_wrapper.xml
~~~

Since you have mentioned that you are using `robotInterface`, please note that it has been deprecated in favor of the `yarprobotinterface`, see https://github.com/robotology/QA/issues/109 .

gregoire-pointeau, 2016-08-24:
Hi @traversaro , thanks for your answer.
Concerning `robotInterface` to `yarprobotinterface`, it's done now without problem thanks.

However our `pc104` doesn't know the command `updatedb`. Is there an homologue function?

traversaro, 2016-08-24:
It seems a similar issue (old files left in the `.local`) with these files : https://github.com/robotology/icub-main/blob/master/app/robots/iCubLyon01/wrappers/FT/left_arm_FT_wrapper.xml .

gregoire-pointeau, 2016-08-24:
Thanks, now we have a firmware version problem apparently. Do you know the procedure and the version to update?

>AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
sharedcan [1] thread:0 msg:91 joint:0 timed out
getFirmwareVersion: message timed out
Unable to read firmware version 
sharedcan [1] thread:0 msg:91 joint:1 timed out
getFirmwareVersion: message timed out
Unable to read firmware version 
sharedcan [1] thread:0 msg:91 joint:2 timed out
getFirmwareVersion: message timed out
Unable to read firmware version 
###################################################################################
###################################################################################
  It has been detected that your control boards are not using the same
  CAN protocol used by iCubinterface. iCubInterface cannot continue.
  Please update your system (iCubInterface and/or your control board firmware.
  For further information please visit: http://wiki.icub.org/wiki/Firmware
###################################################################################
###################################################################################
checkFirmwareVersions() failed. CanBusMotionControl::open returning false, 
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
sharedcan [5] thread:0 msg:91 joint:0 timed out
getFirmwareVersion: message timed out
Unable to read firmware version 
sharedcan [5] thread:0 msg:91 joint:1 timed out
getFirmwareVersion: message timed out
Unable to read firmware version 
###################################################################################
###################################################################################
  It has been detected that your control boards are not using the same
  CAN protocol used by iCubinterface. iCubInterface cannot continue.
  Please update your system (iCubInterface and/or your control board firmware.
  For further information please visit: http://wiki.icub.org/wiki/Firmware
###################################################################################
###################################################################################
checkFirmwareVersions() failed. CanBusMotionControl::open returning false, 
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
AxisType not found
motorPwmLimit not found
lAnalogWrapper: the parameter 'deviceId' has been deprecated, please use parameter 'name' instead. 
lAnalogWrapper: the parameter 'deviceId' has been deprecated, please use parameter 'name' instead. 
One or more devices failed opening... see previous log messages for more info

traversaro, 2016-08-25:
From the messages, it seems that the boards are not replying to the `pc104`. Perhaps it could make sense to open an issue on [`icub-support`](https://github.com/robotology/icub-support).

gregoire-pointeau, 2016-08-25:
Ok, thanks, I'll do it.

closed 2016-08-25 by gregoire-pointeau