robotology/icub-main #366: Fail to start ActionsRenderingEngine

rouzinho, 2016-09-02:
Hello,

I'm trying to use the ActionsRenderingEngine in the icub Simulator by starting first these modules :
`simCartesianControl --robot icubSim
iKinCartesianSolver --robot icubSim --context simCartesianControl --part right_arm
iKinCartesianSolver --robot icubSim --context simCartesianControl --part left_arm
iKinGazeCtrl --robot icubSim
wholeBodyDynamics --robot icubSim --autoconnect --dummy_ft
actionsRenderingEngine --from /usr/local/share/iCub/contexts/actionsRenderingEngine/sim/config.ini`
But when I start the wholeBodyDynamics, I noticed the following output (even if the calibration is fine) : 
`Failure: could not find source port /icubSim/left_arm/analog:o
Failure: could not find source port /icubSim/right_arm/analog:o
Failure: could not find source port /icubSim/left_leg/analog:o
Failure: could not find source port /icubSim/right_leg/analog:o
Failure: could not find source port /icubSim/left_foot/analog:o
Failure: could not find source port /icubSim/right_foot/analog:o`

And these ports are necessary to start the actionsRenderingEngine...

Which module is supposed to open these ports ?

Thank you for your help.

ps: I also copy some files (network.ini exploration_poses.ini that were absent of the sim/ folder)

pattacini, 2016-09-02:
Hi @rouzinho 

The dependencies to launch `WBD` along with the simulator are pretty much here: [**icubSimStartup.xml**](https://github.com/robotology/icub-main/blob/master/app/iCubStartup/scripts/icubSimStartup.xml.template).

I usually use the application below. Try it out.

```xml
<application>
<name>simulation</name>

        <dependencies>
        </dependencies>

        <module>
                <name>iCub_SIM</name>
                <node>localhost</node>
        </module>

        <module>
                <name>simCartesianControl</name>
                <parameters>--no_legs</parameters>
                <node>localhost</node>
        </module>

        <module>
                <name>iKinCartesianSolver</name>
                <parameters>--context simCartesianControl --part right_arm</parameters>
                <node>localhost</node>
        </module>

        <module>
                <name>iKinCartesianSolver</name>
                <parameters>--context simCartesianControl --part left_arm</parameters>
                <node>localhost</node>
        </module>

        <module>
                <name>iKinGazeCtrl</name>
                <parameters>--from configSim.ini</parameters>
                <node>localhost</node>
        </module>

        <module>
                <name>wholeBodyDynamics</name>
                <parameters>--robot icubSim --autoconnect --dummy_ft --no_legs</parameters>
                <node>localhost</node>
        </module>

        <module>
                <name>gravityCompensator</name>
                <parameters>--robot icubSim --no_legs</parameters>
                <node>localhost</node>
        </module>

        <module>
                <name>actionsRenderingEngine</name>
                <parameters>--from sim/config.ini</parameters>
                <node>localhost</node>
        </module>

</application>
```

rouzinho, 2016-09-02:
Thanks for the answer, It seems that all the modules start except the ARE. I tried to start it in a terminal :

pattacini, 2016-09-02:
You may already know it, but just as reminder, don't start them up all at once. The sequence is:

1. `iCub_SIM`
2. Everything else but `WBD`
3. `WBD`, finally.

rouzinho, 2016-09-02:
I tried again and it works (tested with [expect] command). But after closing and restarting again, I got some trouble with other modules. Is there a proper way to close everything in order to avoid future issues ?

pattacini, 2016-09-02:
So I imagine that the command
```
yarp resource --context actionsRenderingEngine --from sim/config.ini
```
returns the correct path to that file.

Usually, to close modules gracefully you should follow the **LIFO policy**: close first the last module you launched (i.e. reverse order).

Sometimes, an **`address conflict`** might occur because a module didn't shut down correctly not releasing its resources (e.g. sockets). This prevents from launching the same application again, for example.

Solving this issue is not that swift though.

rouzinho, 2016-09-02:
I tried again after a reboot and it worked just fine. I guess I was not closing the modules in a right order.
Anyway, thank you very much for the help ;-)

closed 2016-09-02 by rouzinho