robotology/icub-main #367: Grasping

rouzinho, 2016-09-05:
Hello,

I try to make the icub (in simulator) grasp to an object (the default blue ball on the table). I experience some issues to use the grasping module. According to my colleagues, the grasping module inside ActionsRenderingEngine is not so good due to the physics of the simulator, and the best option could be to REACH the ball and use the "magnet" : `world grab (obj) (hand) 1`
But I could not see any REACH module in ARE...
If you have some advices on grasping inside the simulator, I will be glad to listen to them.

Thank you.

vtikha, 2016-09-05:
@rouzinho, indeed the `grasp` command from **`ARE`** would not work well on the simulator due to the fact that it is using a springy model of iCub's hand which is not present in the simulator, therefore it would not control the grasping force which would lead to errors in simulating the contact points. 

Why do you need to reach with **`ARE`**? You could simply do it using the cartesian controller, please find the tutorial ([here](http://wiki.icub.org/brain/icub_cartesian_interface.html)) and then command the simulator to use the `magnet` function. Otherwise, if, for some reason, you specifically need to do it with **`ARE`**, you can use the command:
```sh
$ [touch] [target] "param1" 
```
where `param1` would be `still` which would prevent the  robot from bringing the arm back home after accomplishing the action, and then make use of the simulators `magnet` function.

rouzinho, 2016-09-05:
Thank you for the reply, I will investigate on the cartesian controller side.

closed 2016-09-05 by rouzinho

frank-foerster, 2016-11-17:
Thanks for the info - I was trying to do the same as ARE is not working yet on Gazebo.
Just one comment regarding the `touch` command mentioned above.
I just tried that one and it doesn't seem to work properly neither - it seems the robot isn't picking up on the pressure once contact is established.
Here's a [video](https://vimeo.com/191962876) of it. Note that I'm using a custom model as the ball would slip away when trying to grasp / touch it - I tried that one as well.

pattacini, 2016-11-19:
Hi @frank-foerster 

Remember that force detection is not properly handled within `iCub_SIM`, hence `ARE` doesn't perceive any contact in that particular environment. This was one of the main reasons motivating the use of `Gazebo`, which is under development.