robotology/icub-main #368: Crash of icubSimulator after importing 3D Model

rouzinho, 2016-09-05:
Hello,

I tried to import a 3D model I did with blender (I followed the example [http://wiki.icub.org/wiki/Simulator_README#Importing_3D_models_into_the_simulator:](url)). The thing is the import is a success if made as static model but not as dynamic one. And I need a dynamic one in order to rotate it and place it where I want. 
The crash of the simulator shows the following error : 
`ODE Message 2: mass must be > 0 in dMassCheck() [mass.cpp:49]`
`ODE INTERNAL ERROR 1: assertion "dMassCheck(mass)" failed in dBodySetMass() [ode.cpp:485]`

By looking on internet I saw that the issue might come from a double precision in ODE, so I compiled the more recent ODE [(http://wiki.icub.org/wiki/Linux:_Installing_ODE)](url) and recompiled icub-main. But I still have the same issue...

Thank you for your help.

MtRivWin, 2016-09-07:
Hello.
I had the same problem but I succeeded to import my models.

Please check options below when you save the model files.
Export Meshes
Export Normals
    **Flip Normals**
Export UV Coordinates
Export Materials
[image]

and make sure that you choose Right-Handed for export system.

In my case it works but I don't know why because I'm a beginner of modeling.

Good luck.

pattacini, 2016-09-07:
@MtRivWin seemed to catch the point.

From the [manual](http://wiki.icub.org/wiki/Simulator_README):

> make sure that you are using the right handed system by clicking on "flip z" and then export to the desired location.

@rouzinho did this help you out?

pattacini, 2016-09-07:
BTW, @MtRivWin if you'll accept the invitation to join our team, I'll be able to assign this issue to you 😄

rouzinho, 2016-09-08:
Thanks for the answers, I managed to do it with a static model.

closed 2016-09-08 by rouzinho