robotology/icub-main #370: VOCAB more than 4 letters

rouzinho, 2016-09-08:
Hello,

I want to produce a touch gesture with ActionsRenderingEngine in my source code. To gain some time, I just wanted to send a Bottle to the `/actionsRenderingEngine/cmd:io`. The command works with the simple rpc : `[touch] (-0.35 0.15 0) "still" "left"`.
The thing is I don't know how to send the Vocab [touch] since you can't convert more than 4 letters...

Thank you for your answers.

pattacini, 2016-09-08:
Hi @rouzinho 

Try this:

```cpp
int cmd=yarp::os::Vocab::encode("touch");

yarp::sig::Vector v(3,0.0);
v[0]=-0.35;
v[1]=+0.15;

yarp::os::Bottle b;
b.addVocab(cmd);
b.addList().read(pos);
b.addString("still");
b.addString("left");
```

barbalberto, 2016-09-08:
Actually Vocab::encode will truncate the string up to the forst 4 char so for example encoding 'left_hand' and 'left_arm' will generate the same vocab.
Vocab are encoded as 4bytes integer so there is no way to use more then 4 chars.
That's because they are not meant to be strings but as a 'readable' enum and they just provide hints to understand the command.
To have a better user experience, usually there is a parser from string to vocab from the client, like the simpleClient does.

rouzinho, 2016-09-08:
Perfect it works !
@barbalberto  I understood that later but apparently the vocab was not my only error. 

Anyway, thank you for the help and the reactivity.

closed 2016-09-08 by barbalberto