robotology/icub-main #372: Add hardware limits paramenter in motion control xml file

valegagge, 2016-09-15:
The hardware limits represent the limits beyond which the joint can not be moved physically unless it is not broken.

This param let joint to calibrate towards hardware limit, exceeding user limits without set joint in error.
(for example pronosupination of forearm of icub and neck of R1.)

This param will be a couple (hardwareJntPosMin  and hardwareJntPosMax) and are optional: if there aren't  then `hardwareJntPosMin  = jntPosMin` and `hardwareJntPosMax  >= jntPosMax`  

The motion control device should be check following constraints:
 * `hardwareJntPosMin  <= jntPosMin`
 * `hardwareJntPosMax  >= jntPosMax`

pattacini, 2016-09-15:
Hi @valegagge 

I would consider #133 altogether with this enhancement.

cc @randaz81

valegagge, 2017-05-02:
I implemented this enhacement in embObjMotionControl with the only difference that `hardwareJntPosMin` and `hardwareJntPosMax` are mandatory parameters.

@randaz81 What about canBusMotionControl?

closed 2021-03-27 by pattacini