robotology/icub-main #373: Problems with Dragonfly2 camera when using Firewire hardware clock

gsaponaro, 2016-09-20:
1. this fix should probably be committed in any case: https://github.com/robotology/icub-main/commit/3b0616ca6b81591c5414bfeda7c43d3adc92ed5d#commitcomment-19018030
2. starting two Dragonfly2 devices with the Firewire hardware clock option (```use_network_time 0```) results in the following behaviour: if I start the left device and the right device almost simultaneously on the PC104, and then I compute the difference in the respective image timestamps, then the difference is small; however, if I wait a few seconds between starting the left and starting the right drivers, the difference is much bigger; I guess that this is not the intended behaviour of the "hardware clock option", which should be independent of the time at which the drivers are launched on the computer, right?
3. in addition, the integer part of the timestamp with the Firewire hardware clock option seems strange, example: ```left 18347.169742, right 18198.183742 => deviation between cameras 148.986000``` (148 seconds?!). Probably a problem related to the computation of ```nSecond```, ```nCycleCount```, ```nCycleOffset``` in [FirewireCameraDC1394-DR2_2.cpp](https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/dragonfly2/linux/FirewireCameraDC1394-DR2_2.cpp). When using the YARP system time all is fine: ```left 1474373513.852414, right 1474373513.882302 => deviation between cameras -0.029888```
4. starting two Dragonfly2 devices with the Firewire hardware clock option (```use_network_time 0```), the time at which images are acquired tends to **drift**, as mentioned by @lornat75. I verified that this happens quite fast (10 seconds is enough to observe the time difference to double), while this problem does not happen with the YARP system time option (```use_network_time 1```).

Note: in the tests I only ran the two ```yarpdev``` processes on the PC104, without ```camCalib``` or ```frameGrabberGui2```, and I used the plain UDP carrier (no Bayer). The small program that prints the timestamps and the time difference between them is here: https://github.com/gsaponaro/tools/tree/master/checkCameraSynch, it expects the two images in the ```left:i``` and ```right:i``` ports.

See also: https://github.com/robotology/QA/issues/139

/cc @ale-git @lornat75 @barbalberto @plinioMoreno

lornat75, 2016-09-20:
Just to clarify, the drift I mentioned was within one period (~33ms) and happened without network time (`use_network_time 1`).

gsaponaro, 2016-09-20:
Ah OK, two different things then. I clarified point 4 in the initial post.

apaikan, 2016-09-21:
As @lornat75 mentioned the time drift doesn't go behind the frame grabbing period (1/30 ~= 33ms). We have done some tests before and noticed that the problem with the Firewire grabber raises due to time drift in multi core system. 

- So far I remember, in the older multi-core systems (cpus), there was the problem of cpu's clock drifts. Most recent Linux kernel solves the problem using 'Invariant TSC'.  Thus `yarp::os::Time::now()` retrieve the correct time independent from cpus.  

- I assume dragonfly Firwire devices have different mechanism for synchronization. We did not get so deep into this problem, but if you force the two dragonfly device driver to run on the same cpu (i.e. setting their cpu affinity) the time drift problem disappears. You can set the cpu affinity of the two drivers after running them using `taskset` (https://linux.die.net/man/1/taskset). 

@gsaponaro would you try running both dragonfly driver on the same CPU (e.g., using taskset), measure their time stamp and let us know how it is?

lornat75, 2016-09-21:
My opinion is that instead of fighting getting image timestamps (that as suggested by @ale-git cannot be compared with timestamps of other sensors), we should make sure the grabbers are synchronized. In this way as long as the YARP timestamp is within one frame, the frames should be exactly taken at the same time.

This test can be done by pointing the cameras to a digital chronometer which shows milliseconds and comparing two frames with closest YARP timestamp. @apaikan and @rezaahmadzadeh did such a test and found that the two frames show exactly the same time.

gsaponaro, 2016-09-21:
Update:

I realized that yesterday I was running my tests with the experimental "fix" of point 1 enabled (i.e., ```m_pFrame->timestamp```). However, from [the discussion](https://github.com/robotology/icub-main/commit/3b0616ca6b81591c5414bfeda7c43d3adc92ed5d#commitcomment-19106621) with @paulfitz it turns our that that part of the driver code is fine as it is on the repo (i.e., ```m_pFrame->image``` providing the timestamp embedded in the image). As a result, the problems I listed in the beginning disappear. Apologies!

Thanks for all the information about the CPU core aspect where we run the drivers, good to know for the future.

Closing issue as we get good grabber synchronization both with Firewire hardware clock (this issue) and with the YARP system clock (https://github.com/robotology/QA/issues/139).

closed 2016-09-21 by gsaponaro