robotology/icub-main #377: EmbObjMotioncontrol: use ImplementXXX measure convertion layer.

valegagge, 2016-10-03:
In embObjMotionControl, values read by xml are converted using local variables insted of ImplementXXX measure convertion functions

This  caused bugs. Using the same layer of convertion of measure should be avoid these issues.

pattacini, 2017-03-28:
Hi @valegagge 
any update on this?

valegagge, 2017-09-21:
In embObjMotionControl there are a lot of value conversions from and to firmware and same type of conversion is done using different way and functions. 
So, as first step, I reorganized all this conversions in a single object "measureConverter" and embObMotioncontrol calls its function when nees to convert a value.

I pushed the code on branch `eoMC_measureConverter`, but I have not tested the code yet.

valegagge, 2017-10-26:
The branch  `eoMC_measureConverter` has beed tested and merged. 
see 616a87f9044359809deb7d609f0270e616ce7137

Using ImplementXXX measure convertion layer inside embObjMotionControl requires a lot of effort that is not justified from the little improvement. 
By the measureConverter object we obtain a good level of reuse of code and limited  the possibility of bugs.
So I'll close this issue, as long as this improvement will be not necessary.

closed 2017-10-26 by valegagge