robotology/icub-main #379: icub-main does not compile anymore after removing GSL from yarp

Tobias-Fischer, 2016-10-04:
Hi all,
`icub-main` fails to compile as the `getGslVector` / `getGslMatrix` methods have been moved in another namespace (see https://github.com/robotology/yarp/commit/f3614bf349698b254a439b1da50a607638cb7b32).
I am not sure whether it's planned to use the new Eigen methods, or whether `yarp::sig` should be replaced by `yarp::gsl`.

Best, Tobi

pattacini, 2016-10-04:
Hi @Tobias-Fischer 

`icub-main/devel` compiles fine; are you using `icub-main/master` instead?

pattacini, 2016-10-04:
Now the `devel` branches shoud go along together.

Tobias-Fischer, 2016-10-04:
Ahh nice, I must have missed the announcement of the `devel` branch on `icub-main`. All good, thanks @pattacini!

closed 2016-10-04 by Tobias-Fischer

pattacini, 2016-10-04:
Actually, we didn't announce it properly yet, but only touched on this matter within the **closing remarks** of https://github.com/robotology/QA/issues/143.

In two days we'll be running **IFI Day 8**, which is a good moment to deal with all these points.