robotology/icub-main #382: iCubSIM not working in macOS Sierra

kt10aan, 2016-10-16:
``iCub_SIM`` is not starting in macOS Sierra.

``YARP`` and ``icub-main`` were compiled from the ``master`` branch. Compilation was without problems. Newest versions of the dependencies were installed with ``Homebrew``.

I should note that this was the first time I attempted to compile/run ``yarp``/``iCub`` software on macOS Sierra (I did a clean install of macOS).
So the main thing I want to know first is if someone can reproduce the problem or I should keep digging around my setup.

Starting the ``iCub_SIM`` fails just before drawing the main window, with:

```
mbp:~ kostas$ iCub_SIM
[DEBUG]
ODE configuration: ODE ODE_EXT_trimesh ODE_EXT_opcode ODE_OPC_new_collider ODE_EXT_threading ODE_double_precision

[DEBUG]SimConfig::configure
||| clearing context
||| adding context [simConfig]
||| configuring
||| default config file specified as simulator.ini
||| checking [/Users/kostas/simulator.ini] (pwd)
||| checking [/Users/kostas/Library/Application Support/yarp/config/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/Users/kostas/Library/Application Support/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/Library/Preferences/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/Users/kostas/robotcub/yarp/install/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/Users/kostas/robotcub/icub-main/install/share/iCub/robots/default] (robot YARP_DATA_DIRS)
||| checking [/Users/kostas/robotcub/contrib/share/ICUBcontrib/robots/default] (robot YARP_DATA_DIRS)
||| checking [/Users/kostas/robotcub/yarp/install/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/Users/kostas/robotcub/icub-main/install/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/Users/kostas/robotcub/contrib/share/ICUBcontrib/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/Users/kostas/Library/Application Support/yarp/config/contexts/simConfig] (context YARP_CONFIG_HOME)
||| checking [/Users/kostas/Library/Application Support/yarp/contexts/simConfig] (context YARP_DATA_HOME)
||| checking [/Library/Preferences/yarp/contexts/simConfig] (context YARP_CONFIG_DIRS)
||| checking [/etc/yarp/contexts/simConfig] (context YARP_CONFIG_DIRS)
||| checking [/Users/kostas/robotcub/yarp/install/share/yarp/contexts/simConfig] (context YARP_DATA_DIRS)
||| checking [/Users/kostas/robotcub/icub-main/install/share/iCub/contexts/simConfig] (context YARP_DATA_DIRS)
||| found /Users/kostas/robotcub/icub-main/install/share/iCub/contexts/simConfig
||| checking [/Users/kostas/robotcub/contrib/share/ICUBcontrib/contexts/simConfig] (context YARP_DATA_DIRS)
||| checking [/Users/kostas/robotcub/icub-main/install/share/iCub/contexts/simConfig/simulator.ini] (context)
||| found /Users/kostas/robotcub/icub-main/install/share/iCub/contexts/simConfig/simulator.ini
[INFO]default module name:  /icubSim
[INFO]default verbosity level:  0
||| finding file [ode_params.ini]
||| checking [/Users/kostas/ode_params.ini] (pwd)
||| checking [/Users/kostas/robotcub/icub-main/install/share/iCub/contexts/simConfig/ode_params.ini] (context)
||| found /Users/kostas/robotcub/icub-main/install/share/iCub/contexts/simConfig/ode_params.ini
||| finding file [parts]
||| checking [/Users/kostas/iCub_parts_activation.ini] (pwd)
||| checking [/Users/kostas/robotcub/icub-main/install/share/iCub/contexts/simConfig/iCub_parts_activation.ini] (context)
||| found /Users/kostas/robotcub/icub-main/install/share/iCub/contexts/simConfig/iCub_parts_activation.ini
||| finding file [parts]
[INFO]The iCub simulator will start with the following configuration:

 Elevation :  off
 startHomePos :  on
 Legs :  on
 Torso :  on
 Left arm :  on
 Left hand :  on
 Right arm :  on
 Right hand :  on
 Head :  on
 Fixed Hip :  on

 Self-collisions :  off
 Collisions for covers :  off

 Pressure sensors:  off
 Whole body skin emulation:  off
 Cameras : on
[... 519 lines cut ...]
||| checking [/Users/kostas/robotcub/icub-main/install/share/iCub/contexts/simConfig/data/texture/face.raw] (context)
||| found /Users/kostas/robotcub/icub-main/install/share/iCub/contexts/simConfig/data/texture/face.raw
2016-10-16 19:13:09.345 iCub_SIM[98921:430452] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'nextEventMatchingMask should only be called from the Main Thread!'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff8bbd052b __exceptionPreprocess + 171
	1   libobjc.A.dylib                     0x00007fffa02a8cad objc_exception_throw + 48
	2   AppKit                              0x00007fff89eeec34 -[NSApplication(NSEvent) shouldBeTreatedAsInkEvent:] + 0
	3   libSDL-1.2.0.dylib                  0x000000010ca15e12 QZ_PumpEvents + 337
	4   libSDL-1.2.0.dylib                  0x000000010c9f47f9 SDL_PumpEvents + 35
	5   libSDL-1.2.0.dylib                  0x000000010c9f482c SDL_PollEvent + 14
	6   iCub_SIM                            0x000000010c85ca66 _ZN16OdeSdlSimulation14process_eventsEv + 1414
	7   iCub_SIM                            0x000000010c86b225 _ZN16OdeSdlSimulation7simLoopEii + 1093
	8   iCub_SIM                            0x000000010c808eaf _ZN15SimulatorModule9runModuleEv + 47
	9   iCub_SIM                            0x000000010c7fe7dd _ZN13SimulationRun3runEP16SimulationBundleiPPc + 685
	10  iCub_SIM                            0x000000010c7f63ed _ZN10MainThread3runEv + 29
	11  libYARP_OS.1.dylib                  0x000000010d1c3495 _Z18theExecutiveBranchPv + 213
	12  libACE.dylib                        0x000000010d715c3a _ZN21ACE_OS_Thread_Adapter6invokeEv + 254
	13  libsystem_pthread.dylib             0x00007fffa0d9cabb _pthread_body + 180
	14  libsystem_pthread.dylib             0x00007fffa0d9ca07 _pthread_body + 0
	15  libsystem_pthread.dylib             0x00007fffa0d9c231 thread_start + 13
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6
```

claudiofantacci, 2016-11-03:
Hi @kt10aan,

unfortunately iCub_SIM is not working on macOS since a long time.
You can have a glimpse of this [here](https://github.com/robotology/icub-main/issues/222).

I don't think that the problem will be solved soon since we are trying to migrate to Gazebo.

As far as I'm concerned, I am a tiny virtualized Debian distro on which i run iCub_SIM.

closed 2017-03-28 by pattacini

muratgunana, 2017-05-04:
Did you resolve the issue: the iCub_SIM crashing on mac OS Sierra #386 ?

pattacini, 2017-05-04:
Hi @muratgunana 
It's not sorted out and I believe there's a very little chance it will be in the future, given that most of the development on iCub_SIM is now discontinued in favor of the Gazebo counterpart, as @claudiofantacci mentioned one comment above.

You may consider following @claudiofantacci's suggestion and use a tiny VM for iCub_SIM as fallback or switch to Gazebo.