robotology/icub-main #385: Python getEncoders seems return wrong value

kkjh0723, 2016-11-01:
I'm having trouble while trying to run tutorial in (http://wiki.icub.org/brain/icub_python_basic_motor.html)

It seems I correctly did yarp-python binding following (http://wiki.icub.org/wiki/Python_bindings).
I didn't bind icub-python because it seems not necessary.

When I run the tutorial code, the right arm actually moves but not to proper position.
So I tried to print what encoder value using `print encs.get(0), encs.get(1)` 

but printed value is `6.92746259798e-310` and `7.57506981739e-317` which is trash value.

I'm not sure the reason. Please give me some advice.
Below is the full code I used.

    import readline # workaround in anaconda env to import yarp 
    import yarp
    
    yarp.Network.init()
    
    #prepare a property object
    props = yarp.Property()
    props.put("device","remote_controlboard")
    props.put("local","/client/right_arm")
    props.put("remote","/icubSim/right_arm")
    
    #creat remote driver
    armDriver = yarp.PolyDriver(props)
    
    #query motor control interfaces
    iPos = armDriver.viewIPositionControl()
    iEnc = armDriver.viewIEncoders()
    
    #retrieve number of joints
    jnts = iPos.getAxes()
    print 'Controlling ',jnts,' joints'
    
    #read encoders
    encs = yarp.Vector(jnts)
    iEnc.getEncoders(encs.data())
    print encs.get(0), encs.get(1)
    
    #store as home position
    home = yarp.Vector(jnts, encs.data())
    
    #initialize a new tmp vector identical to encs
    tmp = yarp.Vector(jnts)
    
    #add 10 degrees to the first joints
    tmp.set(0, tmp.get(0)+10)
    tmp.set(1, tmp.get(1)+20)
    tmp.set(2, tmp.get(2)+10)
    tmp.set(3, tmp.get(3)+10)
    
    #move to new position
    iPos.positionMove(tmp.data())
    
    #back to home position
    #iPos.positionMove(home.data())

pattacini, 2016-11-01:
Hi @kkjh0723 

It's likely that when you call the first time the `getEncoders()` method, the thread running underneath that collects those data from the network didn't have the chance to receive the first message yet, hence the crazy values you saw. 

You could either introduce a suitable delay before the first call or check continuously (with a small time step) against its return value until it becomes `true`

kkjh0723, 2016-11-01:
Thanks very much!! It helps a lot. The problem was exactly what you wrote.

pattacini, 2016-11-01:
Glad it helped.

closed 2016-11-01 by pattacini