robotology/icub-main #388: calibrationDelta is ignored by parametricCalibrator device, but it is present on some configuration file devices

traversaro, 2016-11-25:
We need to either: 
 * Remove that line from the configuration files, to avoid misunderstanding 
 * Add support for calibrationDelta parameter in parametricCalibrator . 

I guess @randaz81 is the one taking the decision. 

Related: https://github.com/robotology/icub-main/issues/240 , because  the documentation should describe the fine calibration procedure correctly regardless of the decision that we take. : ) 

@nunoguedelha @gabrielenava @S-Dafarra

randaz81, 2016-11-25:
I need few days to think about this..

nunoguedelha, 2016-11-25:
@randaz81 , while waiting for your analysis, I would like to highlight the fact that the configuration of joint encoders offsets should be independent from the bus type CAN or ETHERNET. That would probably be the case also for other parameters. Maybe this would be a good opportunity to split common config parameters from those specific to the hardware interface type.

randaz81, 2016-11-25:
There are several points wrong in this discussion, and also in the title. 
There exist two different calibrator modules, parametricCalibrator and parametricCalibratorEth, that should be used respectively on CAN and ETH devices. Each of them use different sets of parameters: obviously ETH devices are newer and they support more features, controlled by different parameters. Introducing these features on legacy devices with different MCUs may be difficult, of even impossible, without revolutionizing completely the system. So, I'm sorry, but I cannot put nothing in common between the two systems.
Additionally, here something very weird happened:
You requested to execute parametricCalibratorEth, on a CAN device (see line 5 of head-calib.xml). The system works just by luck, no surprise that these parameters have no effect on a CAN device. And since some of these parameters are mandatory for parametricCalibratorEth, you cannot even remove them from the configuration file. In the end, machines are not responsible for human negligence...

traversaro, 2016-11-25:
Title updated. 
The issue seems to affect (among the others, I did not check all the possible robots affected, just an handful that I know use ETH motor control boards except for the head): 
 * [ ] `iCubGenova02` https://github.com/robotology/robots-configuration/blob/master/iCubGenova02/calibrators/head-calib.xml 
 * [ ] `iCubNancy01` https://github.com/robotology/robots-configuration/blob/master/iCubNancy01/calibrators/head-calib.xml 
 * [ ] `iCubTwente01` https://github.com/robotology/robots-configuration/blob/master/iCubTwente01/calibrators/head-calib.xml
 * [ ] `iCubDarmstad01` https://github.com/robotology/robots-configuration/blob/master/iCubDarmstadt01/calibrators/head-calib.xml

Note that extremely recent robots (such as `iCubGenova04`) are not affected by this, as they use ETH boards even in the neck.

traversaro, 2016-11-25:
A list of robots that could be affected is https://github.com/robotology/robots-configuration/search?utf8=%E2%9C%93&q=%3Cdevice+name%3D%22head_calibrator%22+type%3D%22parametricCalibratorEth%22%3E . The one not affected are the one that use ETH controlboards also in the neck (for sure `iCubGenova04` and `iCubLousanne02`. I am not sure, but I think that all others robots in that list are affected by this problem.

traversaro, 2016-12-01:
Complete robots that need a fix:  
* [ ] `iCubGenova02` https://github.com/robotology/robots-configuration/blob/master/iCubGenova02/calibrators/head-calib.xml 
 * [ ] `iCubNancy01` https://github.com/robotology/robots-configuration/blob/master/iCubNancy01/calibrators/head-calib.xml 
 * [ ] `iCubTwente01` https://github.com/robotology/robots-configuration/blob/master/iCubTwente01/calibrators/head-calib.xml
 * [ ] `iCubDarmstad01` https://github.com/robotology/robots-configuration/blob/master/iCubDarmstadt01/calibrators/head-calib.xml
* [ ] `iCubGenova06` https://github.com/robotology/robots-configuration/blob/master/iCubGenova06/calibrators/head-calib.xml
* [ ] `iCubSheffield01` https://github.com/robotology/robots-configuration/blob/master/iCubSheffield01/calibrators/head-calib.xml
* [ ] `iCubNottingham01` https://github.com/robotology/robots-configuration/blob/master/iCubNottingham01/calibrators/head-calib.xml

cc @randaz81 @davidetome

traversaro, 2017-07-01:
Issue moved to [robotology/robots-configuration #36](https://github.com/robotology/robots-configuration/issues/36) via [**ZenHub**](https://www.zenhub.com/)

traversaro, 2017-07-01:
Issue moved to robots-configuration because is configuration file related, and to use the appropriate tags for the affected robots.

closed 2017-07-01 by traversaro