robotology/icub-main #389: icub-main master branch compilation against yarp master branch is broken

traversaro, 2016-11-28:
Opening this issue to point anyone confused by this to some useful workaround: 
* If you want to keep using the last stable version, use the YARP tags `v2.3.66.2` and icub-main tag  `v1.4.0`
* If you want to switch to the latest version, that will be merged in master soon, checkout the `devel` branch.

traversaro, 2016-11-28:
Note that this should not happen at all YARP releases, it is just a problem related to https://github.com/robotology/yarp/issues/840 .

drdanz, 2016-11-29:
`icub-main` and all related repos tagged (except for `icub-firmware` and `icub-firmware-build` ones that will be tagged in the next hours):
* `v1.4.1`: "old" `master`
* `v1.6.0`: "new" `master` after devel has been merged.

closed 2016-11-29 by traversaro

drdanz, 2016-11-29:
`v1.4.1` tagged also on `icub-firmware` and `icub-firmware-build`.
`icub-firmware-build` now contains a firmware that is compatible with the new `master`, even though it is not the definitive one, as soon as the new one is ready we will tag `v1.6.0` in both repositories.