robotology/icub-main #397: Compilation error; V1.4.0; caused by an IPOPT file

misaki43, 2016-12-09:
Hello,

I tried to update all folders and I had some issues. Then I read the issue #389 and I installed the recommended version. 
But when I tried to make the icub-main repository, I had an error:
`[  8%] Building CXX object src/libraries/iKin/CMakeFiles/iKin.dir/src/iKinIpOpt.cpp.o
/home/user/Software/src/icub-main/src/libraries/iKin/src/iKinIpOpt.cpp:21:22: fatal error: IpTNLP.hpp: No such file or directory
 #include <IpTNLP.hpp>
`
I verified that: 
- I installed correctly ipopt; 
- the IpTNLP.hpp exist (/home/user/Software/src/Ipopt-3.11.7/Ipopt/src/Interfaces/IpTNLP.hpp); 
- the environment variables is correct:
```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/user/Software/src/Ipopt-3.11.7/build/lib
export IPOPT_DIR=/home/user/Software/src/Ipopt-3.11.7/build
export LD_RUN_PATH=$LD_RUN_PATH:/home/user/Software/src/Ipopt-3.11.7/build/lib
```
So have you an idea what I have this issue?

traversaro, 2016-12-09:
I don't know the problem in detail, but I strongly recommend against compiling your own Ipopt . 

Which operating system are you using? If you are using Ubuntu 16.04, you can download the ipopt package with:
~~~
sudo apt install coinor-libipopt-dev
~~~

Furthermore https://github.com/robotology/icub-main/issues/389 has been solved, so you should be able to compile the `master` branch of both `yarp` and `icub-main` with no problems.

misaki43, 2016-12-09:
thanks for this quick answer,
after having cleaning some things (create a new build repository in icub-main +  reinstalling ipopt)  it works... Sorry for that!

About the ipopt package, why do you recommend to compile it like that? (I close the issue after this answer).

traversaro, 2016-12-09:
Because `apt` does not compile `ipopt`, but instead download a package that contains `ipopt` in a binary format (a so-called [.deb package](https://en.wikipedia.org/wiki/Deb_(file_format))). This `.deb packages` are compiled and prepared by expert developers (called "Debian Developers") that make sure that that version of the library is compatible with all other packages that compose the [distribution](https://en.wikipedia.org/wiki/Linux_distribution). 
Furthermore, it simplified maintenance: even years from now, to check if ipopt is installed on your system by simply checking `apt`, instead of having to mantained all those enviromental variable and remember where you compiled ipopt.

pattacini, 2016-12-09:
@misaki43, did @traversaro answer your question?

misaki43, 2016-12-10:
Ok thanks @traversaro, I will transmit your suggestion  to the others!
@pattacini, sorry for the delayed answer.

closed 2016-12-10 by misaki43

traversaro, 2016-12-10:
@misaki43 From where did you get the information that you should IPOPT from source? If it is in some documentation page, it is in our best interest to fix also that documentation.

gsaponaro, 2016-12-10:
> From where did you get the information that you should IPOPT from source? If it is in some documentation page, it is in our best interest to fix also that documentation.

This wiki page: http://wiki.icub.org/wiki/Installing_IPOPT which pops up when googling ipopt icub, and is also linked from the [main Manual page](http://wiki.icub.org/wiki/Manual) and from these other wiki pages: http://wiki.icub.org/wiki/Special:WhatLinksHere/Installing_IPOPT

pattacini, 2016-12-10:
Hi @gsaponaro,

Those instructions do not suggest to compile `Ipopt` on your own, but only provide information about how to do that and I don't like to remove this part since it is still quite useful.

However, I can modify this small guide to enforce that **binaries is the recommended solution**.

gsaponaro, 2016-12-10:
Cool, thanks for making it clearer. Personally what I've been doing is compiling IPOPT manually on older systems, using the deb package on Ubuntu 16.04.

serena-ivaldi, 2017-04-19:
@pattacini I am re-installing now icub-main and there is no mention to install IPOPT with the apt package. Maybe it is included in the icub-common dependencies? But if you are installing from source, it is not clear. I was installing manually and found this issue by chance after googling...

pattacini, 2017-04-19:
@serena-ivaldi yes, Ipopt is provided from within the `icub-common` package, therefore there's no specific mention of it in the installation guide.