robotology/icub-main #401: Remotecontrolboard version needs updating

Tobias-Fischer, 2016-12-19:
Hi,
We just updated our pc104 today (`iCubLondon01`). We are not able to start `iKinGazeCtrl` / `iKinCartesianSolver` etc.
The error message is as follows (whole log attached
[log_icubunicorn_iKinCartesianSolver_12127_London.txt](https://github.com/robotology/icub-main/files/661736/log_icubunicorn_iKinCartesianSolver_12127_London.txt)

):
```
19345 4551.516991 <ERROR> expecting protocol 1 6 0, but the device we are connecting to has protocol version 1 5 1
19346 4551.517007 <ERROR>  please update YARP or the remotecontrolboard implementation
19347 4551.517022 <ERROR> checkProtocolVersion failed 
```
This is weird, as the iCub version is now 1.6.0. However, the `RemoteControlBoard.cpp` has still version 1.5.1 hardcoded: https://github.com/robotology/yarp/blob/master/src/libYARP_dev/src/devices/RemoteControlBoard/RemoteControlBoard.cpp
Therefore, I think this file needs updating.

Thanks,
Tobi and @maxime-petit 
[log_icubunicorn_iKinCartesianSolver_12127_London.txt](https://github.com/robotology/icub-main/files/661738/log_icubunicorn_iKinCartesianSolver_12127_London.txt)

Tobias-Fischer, 2016-12-19:
Actually, @maxime-petit found that https://github.com/robotology/yarp/blob/master/src/libYARP_dev/src/devices/ControlBoardWrapper/ControlBoardWrapper.h also needs updating in order to start the modules properly.

pattacini, 2016-12-20:
@drdanz @barbalberto can have a look into this.
@mbrunettini might be involved for the release.

barbalberto, 2016-12-20:
The master branch has version 1.5.1 while the devel version has been updated to 1.6.0.
The error means, in this case, that some PC in your network are set in devel branch while other in master branch. Since changes in the protocol have been introduced, make sure all PC are set in the same branch.

barbalberto, 2016-12-20:
By the way, the protocol number is about the protocol used by YARP remoteControlBoard and ControlBoardWrapper, it is not related to iCub

pattacini, 2016-12-20:
@barbalberto `icub-main` has been tagged at `1.6.0`, meaning that `master` must be `1.6.0`, if I'm not mistaken...

pattacini, 2016-12-20:
@barbalberto does it mean that `icub-main tag versioning` and `yarp protocol versioning` are not the same things? If so, we have a bit misleading versioning system...

By contrast, if they correspond, we might have a problem in `icub-main/master`.

cc @drdanz @Tobias-Fischer

barbalberto, 2016-12-20:
That version number does not have nothing to do with _**repository**_ versioning or software versioning.
It is simply a handshake between _that specific_ client/server devices pair. 
Each client/server shall have this handshake system to check if the server on one PC (icub) is updated with respect to the client on a different machine.
It is like the protocol check of the CAN protocol version between the robotInterface and icub's boards and so on.

pattacini, 2016-12-20:
Ok thanks.

I think the misunderstanding raised because of the overlap of numbers between `icub-main tag versioning` and `yarp protocol versioning`.

| icub-main branches | repository tag | yarp protocol |
| :---: | :---: | :---: |
| `master` | **1.6.0**👈 | 1.5.1 |
| `devel` | N.A. | 👉**1.6.0** |

Is this the present picture?

randaz81, 2016-12-20:
just edited the above table to underline that we are referring to the remoteControlBoard protocol. Other yarp components may have other protocols versions...

pattacini, 2016-12-20:
It's a pity we have this ambiguity in the numbers... 😕 
I think we can close it, @Tobias-Fischer

closed 2016-12-20 by Tobias-Fischer

barbalberto, 2016-12-21:
Yes, they are similar by change... but I'm gonna change it again soon so they will differ   :-D